dresden-elektronik/deconz-rest-plugin #4120: Switch Tuya _TZ3000_y9xprg33 - TS0011

bsfaxi, 2021-01-06:
## Device

- Product name: Tuya.
- Manufacturer: _TZ3000_y9xprg33.
- Model identifier: TS0011.
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

[image]

[image]

[image]

Unfortunately, the device does not appear in the Phoscon App as a light like the other Tuya switches, neither in Home Assistant via the integration.
Here is the On/Off cluster:
[image]

Thanks in advance for your great support as always.

Mimiix, 2021-01-06:
@Smanar

bsfaxi, 2021-01-06:
I have deselected and then re-selected the "read" buttons. deCONZ has recognized the device and affected him a new name!
It's now operational.
 
[image]

Thanks

closed 2021-01-06 by bsfaxi