dresden-elektronik/deconz-rest-plugin #4123: Device name

duffbeer2000, 2021-01-07:
## Device

- Product name: Q-Remote RGBW
- Manufacturer: Neuhaus Lightning Group
- Model identifier: NLG-Remote
- Device type : Remote
- Link: [Zigbee Alliance Product Page](https://zigbeealliance.org/de/zigbee_products/q%EF%BC%8Crgbw-remote-controller/)

## Screenshots
[image]
[image]
[image]
[image]
[image]
[image]

duffbeer2000, 2021-02-03:
Would it help, to send someone the remote?

Mimiix, 2021-02-04:
@SwoopX

duffbeer2000, 2021-02-26:
Push

duffbeer2000, 2021-03-29:
@Mimiix cloud you reopen the issue?
@SwoopX If I send you the remote would it be possible that you integrate it? (I pay for the return too)

WhistleMaster, 2021-07-04:
I also have a few of them, @SwoopX, I can send you one that you can keep.

SwoopX, 2021-07-06:
Somebody ping me on discord and I can see what I can do