dresden-elektronik/deconz-rest-plugin #4128: 2.07.1 FOH Switch by Senic doesn't work anymore

Fruchuxs, 2021-01-07:
## Describe the bug
Since 2.07.1 the FOH Switch by Senic stopped working. 
PAN ID and Channel unchanged.

## Steps to reproduce the behavior
Upgrade from 2.05.88 to 2.07.1, Signals from the Switch are ignored, or the Toggle doesn't work.
[image]

## Expected behavior
Toggle for Light on.

## Screenshots

[image]

## Environment
## Environment
 - Host system:Raspberry Pi 4B 4G Version
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Homeatic USB Stick
- New SD Card, thanks to BF - SanDisk Extrem Pro
- All Logs logged in tempfs, no high I/O
- BT and WiFi disabled

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
```
15:51:58:202 GW firmware version: 0x26660700
15:52:08:215 dev can't create file: /dev/ttyUSB0 via mknod (missing permissions)
15:52:08:215 dev /dev/ttyAMA0
15:52:08:215 GW firmware version: 0x26660700
15:52:18:230 dev can't create file: /dev/ttyUSB0 via mknod (missing permissions)
15:52:18:230 dev /dev/ttyAMA0
15:52:18:230 GW firmware version: 0x26660700
15:52:24:621 skip binding for attribute reporting of ep: 0x00 cluster 0x0001 (end-device might sleep)
15:52:28:211 dev can't create file: /dev/ttyUSB0 via mknod (missing permissions)
15:52:28:211 dev /dev/ttyAMA0
15:52:28:212 GW firmware version: 0x26660700
15:52:37:165 Current channel 25
15:52:37:187 Device TTL 6360 s flags: 0x7
15:52:38:208 dev can't create file: /dev/ttyUSB0 via mknod (missing permissions)
15:52:38:209 dev /dev/ttyAMA0
15:52:38:209 GW firmware version: 0x26660700
15:52:39:617 ZCL attribute report 0x0017880106F4A224 for cluster: 0x0406, ep: 0x02, frame control: 0x08, mfcode: 0x0000
15:52:40:597 Websocket disconnected 192.168.1.2:58260, state: 0, close-code: 1001, reason:
15:52:41:710 New websocket 192.168.1.2:58829 (state: 3)
15:52:47:266 Websocket disconnected 192.168.1.2:58829, state: 0, close-code: 1001, reason:
15:52:48:231 dev can't create file: /dev/ttyUSB0 via mknod (missing permissions)
15:52:48:231 dev /dev/ttyAMA0
15:52:48:232 GW firmware version: 0x26660700
15:52:48:308 New websocket 192.168.1.2:58892 (state: 3)
```

## Additional context

gvdhoven, 2021-01-28:
Same issue here; FOH switch assignment is not working at all.
- Version: 2.07.01 / 08/12/2020
- Firmware: 26660700

@manup is this a known issue?

Mimiix, 2021-01-28:
For both: Can you update to the newest version and see again?

gvdhoven, 2021-01-30:
@Mimiix do you mean the 2.09.01 beta? If you confirm i can test that.

Mimiix, 2021-01-30:
Yes.

Fruchuxs, 2021-02-01:
Sadly I need to admit, that I dropped my beloved RaspBee II. Since the latest FW Update I did, the performance of the RaspBee is just bad. Always the RaspBee is busy. So I decided to switch to Zigbee2MQTT with a CC2652RB dongle. This works like a charm (on the same ZLL Channel as RaspBee), but without delays and performance problems.

The spontaneous drop of the FoH compatibility was just one minor problem in the series of problems I had with deconz in the past. But I like the product and just hope that I'm just one people who hit strange Bugs and Problems.

However, I don't updated and tested, sorry.

closed 2021-03-05 by github-actions[bot]

JayJay68, 2021-09-19:
@Fruchuxs I have had strange problems, too (RP 3b, RaspBee) like permision problems, no connect to the RaspBee,....
So I make a backup with the deConz-app, download the latest stabile Image {2.16.x), restore the Backup and since then all things works like a charme.