dresden-elektronik/deconz-rest-plugin #4129: Alcantara 2 Heater/Thermostat

NicoB35, 2021-01-07:
## Device

- Product name: Acova Alcanrata 2
- Manufacturer:ZHENDER GROUP VAUX ANDIGNY
- Model identifier: ALCANTARA2 D1.00P1.01Z1.00
- Device type : 
   Thermostat
 

## Screenshots

SwoopX, 2021-01-17:
We'd need the remaining thermostat cluster attributes from attribute 0x0012 downwards. I'm afraid you'll probably need to manually read each of them via a double click on the value.

NicoB35, 2021-01-19:
Thank you for answering

When i Read all the thermostat cluster, only :

0x0000
0x0002
0x0003
0x0004
0x0010
0x0011
0x0012
0x0014
0X001B
0x001C
0x001E
0x0029
0x0030

remain black (not greyed)
I hope it will help.

[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2021-01-19:
Have you read all the attributes individually as requested? Doesn't look like it.

NicoB35, 2021-01-19:
Sorry for my bad english and misunderstanding
When i double click on (for example) 0x0012 (Occupied Heating Setpoint)
I get a little window : Attribute info
The read button gives me the value (example 1900 for 19,00°c)
The read config returns "not found"
[image]

I think i do something wrong and that you wait for something else
Could you explain me where is my mistake please, or give me the way to get the values you need.
Thank you for your time

NicoB35, 2021-01-20:
To add information,
I read the REST-API wiki yesterday and tried to find my heaters, nothing in lights / sensors
I also read various post in device request, I saw the same request you made : "We'd need the remaining thermostat cluster attributes from attribute 0x0012 downwards. I'm afraid you'll probably need to manually read each of them via a double click on the value."
Unfortunately, no answer so i couldn't use this request to get the values needed.I need a little walkthrough
I hope it will help

SwoopX, 2021-01-20:
> When i double click on (for example) 0x0012 (Occupied Heating Setpoint)
> I get a little window : Attribute info
> The read button gives me the value (example 1900 for 19,00°

This is exactly what needs to be done, press the `read` button there. `read config` is irrelevant, at least for now. After the value has been read, the value will be automatically updated in the attribute overview. If an attribute is not found on the device, the read will fail and be grayed out (which is fine and the info we need).

NicoB35, 2021-01-20:
I think I have the values needed :

0x0012   Occupied heating setpoint           s16          rw    1900

0x0014   Unoccupied Heating setpoint      S16          rw    1550

0x001B   Control Sequence of Operation   enum8    rw    Heating Only (Cooling only / Colling with Reheat / Colling and Heating 4- pipes / Colling and Heating 4-pipes with Reheat)

0x001C   System Mode                              enum8     rw     Auto     (Off / Cool / Heat / Emergency Heating / Precooling / Fan only / Dry / Sleep

0x001E   Thermostat Running Mode         enum8     r     Heat      (Cool / Off)

0x0029   Thermostat Running State          bmp16     r      0x0001   (see capture 0x0029.jpg)
[image]

0x0030   Setpoint Change Source             enum8     r      Zigbee    (Manual / Scedule)

I hope it will help

NicoB35, 2021-01-26:
Hello,
The request is still in "waiting for info" status.
I can provide more info if needed.
Don't hesitate to ask, will give them quickly.

Have a good day.

Mimiix, 2021-01-28:
@SwoopX

SwoopX, 2021-01-30:
I guess that is sufficient for now, thanks

closed 2021-03-09 by github-actions[bot]