dresden-elektronik/deconz-rest-plugin #4131: Enhance Battery Reporting Time (specifically for Samsung Smartthings Devices)

jameshirka, 2021-01-07:
## Feature request type
Enhance Battery Reporting Time to allow devices longer times before reporting and not set to default value, specifically for Samsung Smartthings Devices

## Description
Smartthings devices connected to Deconz have a very fast battery drain.

This appears to be because they are reporting too frequently and the minimum change interval is set to low compared to Samsungs documentation.
https://github.com/SmartThingsCommunity/SmartThingsPublic/blob/master/devicetypes/smartthings/smartsense-multi-sensor.src/smartsense-multi-sensor.groovy

In a previously closed issue, NickWateron appears to have found the issues and suggested changes:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2908#issuecomment-642698345

## Considered alternatives
No alternatives whilst using Deconz.

## Additional context
When using Deconz, Smartthings devices report battery too frequently causing drain on the battery and therefore reduced battery life.

The problem seems to be that if the configured reporting is more than 1.5 times the original value that reporting is reset back to default values.

This is to reopen issue:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2908

It seems the issue has been tracked down in comment:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2908#issuecomment-642698345

manup, 2021-01-07:
In future it will be a bit easier to configure these, see: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2908#issuecomment-756428713
Meanwhile for the next release I'll align the values with the groovy descriptions.

jameshirka, 2021-01-10:
> In future it will be a bit easier to configure these, see: [#2908 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2908#issuecomment-756428713)
> Meanwhile for the next release I'll align the values with the groovy descriptions.

Thanks, let me know when it's right to test.
It seems there might be some resetting of values back to default that might need to be changed as well.