dresden-elektronik/deconz-rest-plugin #4132: Tobular Motor Zemismart ZM25TQ Zigbee- new version _TZE200_fzo2pocs

jameshirka, 2021-01-08:
## Device
Product name: ZM25TQ
Manufacturer: _TZE200_fzo2pocs
Model identifier: TS0601
Device type : Roll-motor/cover

## Screenshots
[image]

This is the same device externally as:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3161
My device has the exact same markings on it as the one mentioned in the issue.

Smanar, 2021-01-08:
Arf bad luck "smart plug" like switches, not realy good.
You haven't 3 lights entry created when trying to integrate it ?

I need too, the manufacture number on the node info panel pls.

jameshirka, 2021-01-08:
> Arf bad luck "smart plug" like switches, not realy good.
> You haven't 3 lights entry created when trying to integrate it ?
> 
> I need too, the manufacture number on the node info panel pls.

No, it only creates a single entity which identifies as a smart plug but also a thermostat.
[image]

Node info screenshot:
[image]

Please let me know if there is anything else you need.

Smanar, 2021-01-09:
Do you know how to compile the code to make a try ?

```
sudo apt install deconz-dev
git clone --branch ZM25TQ https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

jameshirka, 2021-01-10:
> Do you know how to compile the code to make a try ?
> 
> ```
> sudo apt install deconz-dev
> git clone --branch ZM25TQ https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```

Thanks Smanar.

Sorry it took so long, I got another PI up and running to not have to play with the container that Deconz runs in for HA.

I gave it a shot, unfortunately, now they are seen as fairy lights/ smart plugs:
[image]

When I hit the API I get:
`    "2": {
        "etag": "77a694d6805b4fa35c415f070a57c6cf",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2021-01-10T08:20Z",
        "manufacturername": "_TZE200_fzo2pocs",
        "modelid": "TS0601",
        "name": "Smart plug 2",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Smart plug",
        "uniqueid": "5c:02:72:ff:fe:94:31:fe-01"
    },`

Any thoughts/ ideas?

Smanar, 2021-01-10:
This is more normal for me.
Can you try to make a second inclusion but without deleting the previous devices ?

If I m right you will have more field on the one with the uniqueid  finishing by "-01"

jameshirka, 2021-01-10:
You're 100% correct.

I now get (after moving it up and down by doing puts on the lift)

`    "2": {
        "etag": "41af9ec875f96039afa2b135038c1fe0",
        "hascolor": false,
        "lastannounced": "2021-01-10T10:20:47Z",
        "lastseen": "2021-01-10T10:30Z",
        "manufacturername": "_TZE200_fzo2pocs",
        "modelid": "TS0601",
        "name": "Smart plug 2",
        "state": {
            "alert": "none",
            "bri": 254,
            "lift": 100,
            "on": true,
            "open": false,
            "reachable": true
        },
        "swversion": null,
        "type": "Window covering device",
        "uniqueid": "5c:02:72:ff:fe:94:31:fe-01"
    },`

Is there anything else I should be checking?
Also is there a way of setting if the roller is reverse or not?
Ie this motor is installed on the left with the blind rolling forward which unfortunately means up on the remote puts the blind down and vice-e-versa.
Where as the rest of my motors are installed on the right and the remotes work the correct way.

Smanar, 2021-01-10:
Yep there is an API command to reserse side.

`PUT, PATCH /api/<apikey>/lights/<id> with {"reverse":true}`

I will check later if If I can find a way to prevent the 2 others device creation, but will be complicated because of the device type "smart plug".

jameshirka, 2021-01-11:
Thanks Smanar.

This is ready to go when you're ready to do the pull request.

Thanks for all your help.

closed 2021-01-11 by jameshirka

Smanar, 2021-01-11:
OK, so I have checked, I have no way to detect this device and differentiate from switches.
Will be possible in a future deconz version, but not for the moment.

For the moment I think you can delete the 2 other fake entries ?

It s like you want I can make the PR as it if you want ?

jameshirka, 2021-01-11:
Please make a PR as is.
I'm happy to have it working today and the new device JSON based DDF might be the right time to look at the multiple entities issue.

Thanks.

shlomki, 2021-11-12:
Hi @Smanar, I'm using the same device and need some help with setting the up & down limits. I saw that the Tuya app allows to save the up & down positions (as well as a middle position but there's only so much that I could hope for).
I was wondering if there's a chance that an API command could help with that, just like the reverse command?

How can I investigate if there are any other secret commands such as the "reverse" command you've found?
thanks!

Smanar, 2021-11-12:
Your device is the "_TZE200_fzo2pocs" ?
Because I m not sure this device use a tuya command to set max/min position ?

shlomki, 2021-11-12:
Yes, that's exactly the device I have.
I saw that `{"calibration": "true"}` is supported in the API but I couldn't figure out how to use it.
The manual says there are two options to calibrate - using a remote and using the app. I don't have a remote... so I'm trying to use deconz. 
You can see the screenshot from their app allows to set up, down, and middle.
[image]

Smanar, 2021-11-12:
This command is for device that have the covering cluster.
I have take a look again, and I don't find calibration command for covering that use the tuya cluster.

I m searching on other zigbee project but not more command.

shlomki, 2021-11-12:
Thanks for looking into it again, it's interesting that the OP doesn't have the window covering cluster - but I do.
For some reason though, it's read only.
[image]

Smanar, 2021-11-13:
Ok, so you have a new version ....

They are not read only, read only can be read, but disabled, it mean generaly the device don't support them.
But have you scroll down the attributes lists (the tab) there is more under, to be sure all are disabled.

Drapiece, 2022-08-29:
> 좋아, 그래서 당신은 새로운 버전이 있습니다 ....
> 
> 그들은 읽기 전용이 아니라 읽기 전용이 될 수 있지만 비활성화되어 있습니다. 일반적으로 장치가 지원하지 않는다는 것을 의미합니다. 그러나 속성 목록(탭)을 아래로 스크롤하여 모두 비활성화되었는지 확인하십시오.

hello, I'm getting a lot of help from you. (_TZ3000_okaz9tjs, _TZ3000_r6buo8ba)

I recently purchased a tublar motor.
Unfortunately this model. (TZE200_fzo2pocs)

I am using deconz built in hassos.

It seems that the code compile mentioned above is not applicable.
(SUDO APT Install Deconz-Dev
git clone --branch zm25tq https://github.com/smanar/deconz-trugin.git
CD Deconz-REST-PLUGIN
QMake & & Make
sudo cp ../libde_rest_plugin.so/usr/shaer/deconz/plugins)

Are there exists DDF in this environment?
Thank you.

Drapiece, 2022-08-29:
> 좋아, 당신은 새로운 버전을 출시했습니다 ....
> 
> 낭독이 낭독의 전당에 올랐습니다. 기술이 지원하지 않습니다. 이 목록(프론트)을 먼저 확인하십시오.

but, I find this.

motor STOP
[image]

motor DOWN
[image]

motor UP
[image]

I tried to create a DDF using the above method,
but I don't know how to insert it.
If there is only UP DOWN STOP, it would be possible to automate somehow in homeassistant.
Please help me again.
thank you!

Drapiece, 2022-08-29:
> hello, I'm getting a lot of help from you. (_TZ3000_okaz9tjs, _TZ3000_r6buo8ba)
>
> I recently purchased a tublar motor. Unfortunately this model. (TZE200_fzo2pocs)
>
> I am using deconz built in hassos. [...]

Smanar, 2022-08-29:
It s not possible yet using DDF for Tuya covering.
So to use a new device the only way is adding it using c++ code, and the PR will be probably refused, so you need to make that only for you.

Can mimic one of them
```

    {"_TYST11_wmcdj3aq", "mcdj3aq", "Zemismart", "Tuya_COVD ZM25TQ"},
    {"_TZE200_wmcdj3aq", "TS0601", "Zemismart", "Tuya_COVD ZM25TQ"},
    {"_TZE200_fzo2pocs", "TS0601", "Zemismart", "Tuya_COVD ZM25TQ"},
    {"_TYST11_xu1rkty3", "u1rkty3", "Smart Home", "Tuya_COVD DT82LEMA-1.2N"},
    {"_TZE200_xuzcvlku", "TS0601", "Zemismart", "Tuya_COVD M515EGB"},
    {"_TZE200_rddyvrci", "TS0601", "Moes", "Tuya_COVD AM43-0.45/40-ES-EZ(TY)"},
    {"_TZE200_zah67ekd", "TS0601", "MoesHouse / Livolo", "Tuya_COVD AM43-0.45-40"},
    {"_TZE200_nogaemzt", "TS0601", "Tuya", "Tuya_COVD YS-MT750"},
    {"_TZE200_zpzndjez", "TS0601", "Tuya", "Tuya_COVD DS82"},
    {"_TZE200_cowvfni3", "TS0601", "Zemismart", "Tuya_COVD ZM79E-DT"},
    {"_TZE200_5zbp6j0u", "TS0601", "Tuya/Zemismart", "Tuya_COVD DT82LEMA-1.2N"},
    {"_TZE200_fdtjuw7u", "TS0601", "Yushun", "Tuya_COVD YS-MT750"},
    {"_TZE200_bqcqqjpb", "TS0601", "Yushun", "Tuya_COVD YS-MT750"},
    {"_TZE200_nueqqe6k", "TS0601", "Zemismart", "Tuya_COVD M515EGB"},
    {"_TZE200_iossyxra", "TS0601", "Zemismart", "Tuya_COVD Roller Shade"},
```

For most device you just need to edit product_match.cpp file.

Drapiece, 2022-08-30:
> It s not possible yet using DDF for Tuya covering. So to use a new device the only way is adding it using c++ code, and the PR will be probably refused, so you need to make that only for you.
>
> Can [...]

thank you for your support!!
but, It can't be on builtin hassos.

suddenly??, tublar motor shown me normal status.
[image]
Temporarily available for use.

And, I ordered tuya zigbee gateway.
first, tuya zigbee gateway connect to tublar motor. and homeassistant connect tuya cloud.
I think it will be can so really normal status and move.

second, I will waiting stable update from deconz.
it is bestest action now. right?

thank you!!

Smanar, 2022-08-30:
Yes, because the future will be with DDF, there is a DDF for TRV using the Tuya cluster that wait for validation, one time this one will be included, will be the covering turn, but can take lot of time.