dresden-elektronik/deconz-rest-plugin #4134: Müller-Licht non color scene controller

helmutGitHub, 2021-01-08:
## Device

- Product name: mobile switch from Tint Starter Set
- Manufacturer: NA (Müller-Licht)
- Model identifier: NA 
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots
https://www.mueller-licht.de/produktinformationen/artikel/404015/
[image]
[image]
[image]

Mimiix, 2021-01-08:
Missing screenshots. Please provide all cluster screenshots.

helmutGitHub, 2021-01-08:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

duffbeer2000, 2021-01-08:
Hi Helmut, please read the wiki: [Request-Device-Support](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support)

The important thing: 

> **Please note that at least "Manufacturer Name" and "Model Identifier" must be populated with data (therefore, must not be empty), otherwise that information will not be usable.** For battery powered devices, after pressing read it is required to wake-up the device by pressing a button or any other means of interaction.

helmutGitHub, 2021-01-08:
Sorry for missing this information. I didn't know of the part with "read". Is there anything else missing?

[image]

1100101, 2021-02-12:
I too have this device and would like to use the remote with deCONZ.
@Mimiix Is there still information missing that I may provide?

Mimiix, 2021-02-12:
> I too have this device and would like to use the remote with deCONZ.
> @Mimiix Is there still information missing that I may provide?

I totally missed the added screens..

My apologies!

1100101, 2021-02-18:
@Mimiix I don’t want to nit-pick, but shouldn’t you remove the “incomplete issue” label here?

Mimiix, 2021-02-18:
Wait what.

I removed it 😕

Mimiix, 2021-02-18:
It better be gone now.

1100101, 2021-02-18:
Thanks, all fine now :)

helmutGitHub, 2021-03-12:
@Mimiix, can you  make a rough estimate how long it takes to include this switch?

Mimiix, 2021-03-12:
Whenever a devs feels like it. @helmutGitHub . Switches are hard to implement and devs often want a hands on device to make sure to implement it correctly.

helmutGitHub, 2021-03-12:
@Mimiix, if you wish, i can provide you with a switch.

Mimiix, 2021-03-12:
> @Mimiix, if you wish, i can provide you with a switch.

I'm not a dev. Perhaps @SwoopX or @ebaauw is interested

helmutGitHub, 2021-03-12:
@SwoopX, @ebaauw, are you interested?

ebaauw, 2021-04-04:
Sorry, missed this one.  I’m happy to have a look at the switch, please PM me on Discord.

ebaauw, 2021-04-10:
Received the remote, thanks, @helmutGitHub.  Apparently, already quite a few issues for this device; linking them to here.

Device seems quite standard: client _On/Off_, _Level Control_, and _Scenes_clusters.  It doesn't send anything, until these are bound.  _OTAU_ cluster seems to be implemented; no _Power Configuration_ and the _Power Descriptor_ doesn't seem to report the battery level (value 0b1100).

The _On_ button alternates between sending _On_ and _Off_ on release.  No distinction between short vs long press.  Will be exposed as 1002.

The _DimUp_ and DimDown_ buttons send _Step_ on short press/release; _Move_ on hold; and _Stop_ on release.  Will be exposed as 2002, 3002, 2001/2003 and 3001/3003.

The _Favorite_ button (manual uses American spelling) sends _Recall Scene_ on press/release, and _Store Scene_ when held for 5 seconds.  The commands are sent to the bound group.  The scene seems to be hard-coded to 4003/01.  Probably best supported as 4002 and 4003.

The server _ZLL Commissioning_ cluster reports four groups, 0x05ae to 0x05b1, but they don't seem to be used for anything (maybe when touchlinking?).

helmutGitHub, 2021-04-10:
You are really fast and so i'm looking forward for the implementation. Until then, thank you very much.

ebaauw, 2021-04-18:
Above commit should add support.  The device is a bit wacky, it might take some time after pairing before all the bindings are setup.  Also when holding the DimUp or DimDown button, it sends x002 on press, x001 on hold and x003 on release.  Basically it will always register a short press before a long press.  The Scene button sends 4002 on press/release and 4003 on hold.

closed 2021-04-18 by manup

helmutGitHub, 2021-04-18:
Thank you very much, Erik.

gforums, 2021-04-20:
Thanks also from my side for the fast implementation, appreciate!

gforums, 2021-04-23:
Switched to newest Beta some minutes ago.
Remote is recognized and visible for me in deconz openhab.
Trigger events work as described for me.

Thanks for the implementation again, being an even more customer now :)