dresden-elektronik/deconz-rest-plugin #4135: No button handler for: RWL021 after upgrading software

Tooa, 2021-01-08:
## Describe the bug

I upgraded from `marthoc/deconz:armhf-2.05.84` to `marthoc/deconz:armhf-2.07.01` and my Philips hue dimmer stopped working with the error message: "No button handler for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 001". Downgrading to `marthoc/deconz:armhf-2.05.84` resolves the issue. 

## Steps to reproduce the behavior

1. Working setup with `marthoc/deconz:armhf-2.05.84`
2. Replace image with `marthoc/deconz:armhf-2.07.01` (latest stable)
3. Click on dimmer to switch on the light (worked on `2.05.84`)

## Expected behaviour

The light bulb turns on

## Environment

 - Host system: Raspberrypi 4, 4GB
 - Running method: Marthoc Docker container `marthoc/deconz:armhf-2.07.01` (latest stable)
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

Let me know if you need more of the logs as DEBUG level 2 logs a lot. The snippet below should contain two button presses. 

```
deconz    | 12:36:36:283 Force binding of attribute reporting for sensor switch_living_room
deconz    | 12:36:36:283 [INFO] - No button handler for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 004
deconz    | 12:36:36:284 discard sensor state push for 2: state/buttonevent (already pushed)
deconz    | 12:36:36:285 discard sensor state push for 2: state/eventduration (already pushed)
deconz    | 12:36:36:286 discard sensor state push for 2: state/lastupdated (already pushed)
deconz    | 12:36:36:825 poll node 00:17:88:01:03:54:85:0d-0b
deconz    | 12:36:36:825 Poll light node ceiling_living_room
deconz    | 12:36:36:878 read attributes of 0x001788010354850D cluster: 0x0006: [ 12:36:36:878 0x0000 12:36:36:878 ]
deconz    | 12:36:36:878 add task 278 type 19 to 0x001788010354850D cluster 0x0006 req.id 4
deconz    | 12:36:36:878 Poll APS request 4 to 0x001788010354850D cluster: 0x0006
deconz    | 12:36:36:901 Poll APS confirm 4 status: 0x00
deconz    | 12:36:36:901 Erase task req-id: 4, type: 19 zcl seqno: 127 send time 0, profileId: 0x0104, clusterId: 0x0006
deconz    | 12:36:36:927 Node data 0x001788010354850d profileId: 0x0104, clusterId: 0x0006
deconz    | 12:36:36:927 0x001788010354850D: update ZCL value 0x0B/0x0006/0x0000 after 0 s
deconz    | 12:36:36:927 enqueue event event/checkgroupanyon for /groups/65520
deconz    | 12:36:36:983 enqueue event config/localtime for /config/
deconz    | 12:36:37:079 enqueue event state/buttonevent for /sensors/2
deconz    | 12:36:37:079 enqueue event state/eventduration for /sensors/2
deconz    | 12:36:37:079 enqueue event state/lastupdated for /sensors/2
deconz    | 12:36:37:079 Force binding of attribute reporting for sensor switch_living_room
deconz    | 12:36:37:079 [INFO] - No button handler for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 004
deconz    | 12:36:37:080 discard sensor state push for 2: state/buttonevent (already pushed)
deconz    | 12:36:37:081 discard sensor state push for 2: state/eventduration (already pushed)
deconz    | 12:36:37:082 discard sensor state push for 2: state/lastupdated (already pushed)
deconz    | 12:36:37:200 enqueue event state/buttonevent for /sensors/2
deconz    | 12:36:37:200 enqueue event state/eventduration for /sensors/2
deconz    | 12:36:37:200 enqueue event state/lastupdated for /sensors/2
deconz    | 12:36:37:200 Force binding of attribute reporting for sensor switch_living_room
deconz    | 12:36:37:200 [INFO] - No button handler for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 004
deconz    | 12:36:37:201 discard sensor state push for 2: state/buttonevent (already pushed)
deconz    | 12:36:37:202 discard sensor state push for 2: state/eventduration (already pushed)
deconz    | 12:36:37:203 discard sensor state push for 2: state/lastupdated (already pushed)
deconz    | 12:36:37:909 poll node 00:17:88:01:02:7a:71:a0-0b
deconz    | 12:36:37:909 Poll light node floor_3_living_room
deconz    | 12:36:37:962 read attributes of 0x00178801027A71A0 cluster: 0x0006: [ 12:36:37:962 0x0000 12:36:37:962 ]
deconz    | 12:36:37:962 add task 283 type 19 to 0x00178801027A71A0 cluster 0x0006 req.id 14
deconz    | 12:36:37:962 Poll APS request 14 to 0x00178801027A71A0 cluster: 0x0006
deconz    | 12:36:37:983 enqueue event config/localtime for /config/
deconz    | 12:36:37:983 Wait 4s till query finished
deconz    | 12:36:38:006 Poll APS confirm 14 status: 0x00
deconz    | 12:36:38:006 Erase task req-id: 14, type: 19 zcl seqno: 128 send time 0, profileId: 0x0104, clusterId: 0x0006
deconz    | 12:36:38:027 Node data 0x00178801027a71a0 profileId: 0x0104, clusterId: 0x0006
deconz    | 12:36:38:028 0x00178801027A71A0: update ZCL value 0x0B/0x0006/0x0000 after 0 s
deconz    | 12:36:38:028 enqueue event event/checkgroupanyon for /groups/65520
deconz    | 12:36:38:197 read attributes of 0x00178801027A71A0 cluster: 0x0008: [ 12:36:38:198 0x0000 12:36:38:198 ]
deconz    | 12:36:38:198 add task 285 type 19 to 0x00178801027A71A0 cluster 0x0008 req.id 19
deconz    | 12:36:38:198 Poll APS request 19 to 0x00178801027A71A0 cluster: 0x0008
deconz    | 12:36:38:302 Poll APS confirm 19 status: 0x00
deconz    | 12:36:38:302 Erase task req-id: 19, type: 19 zcl seqno: 129 send time 0, profileId: 0x0104, clusterId: 0x0008
deconz    | 12:36:38:304 read attributes of 0x00178801027A71A0 cluster: 0x0300: [ 12:36:38:304 0x0001 12:36:38:304 0x0003 12:36:38:304 0x0004 12:36:38:304 0x0007 12:36:38:304 0x0008 12:36:38:304 0x4000 12:36:38:304 0x4001 12:36:38:304 0x4002 12:36:38:304 ]
deconz    | 12:36:38:304 add task 287 type 19 to 0x00178801027A71A0 cluster 0x0300 req.id 21
deconz    | 12:36:38:304 Poll APS request 21 to 0x00178801027A71A0 cluster: 0x0300
deconz    | 12:36:38:324 Node data 0x00178801027a71a0 profileId: 0x0104, clusterId: 0x0008
deconz    | 12:36:38:324 0x00178801027A71A0: update ZCL value 0x0B/0x0008/0x0000 after 0 s
deconz    | 12:36:38:410 Poll APS confirm 21 status: 0x00
deconz    | 12:36:38:410 Erase task req-id: 21, type: 19 zcl seqno: 130 send time 0, profileId: 0x0104, clusterId: 0x0300
deconz    | 12:36:38:444 Node data 0x00178801027a71a0 profileId: 0x0104, clusterId: 0x0300
```

In contrast this is how it looks like when it is working in `image: marthoc/deconz:armhf-2.05.84`:

```
deconz    | 12:14:38:959 [INFO] - No button map for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 001
deconz    | 12:14:38:984 delay sending request 20 dt 1 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0008 onAir: 1
deconz    | 12:14:38:984 delay sending request 21 dt 1 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:010        0x0017880103548895 force poll (2)
deconz    | 12:14:39:055 [INFO] - No button map for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 001
deconz    | 12:14:39:084 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:184 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:284 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:384 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:484 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:584 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:604 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:684 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:784 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:884 delay sending request 21 dt 0 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:39:984 delay sending request 21 dt 1 ms to 0x0017880103548895, ep: 0x0B cluster: 0x0300 onAir: 1
deconz    | 12:14:40:027        0x0017880103548895 force poll (2)
deconz    | 12:14:40:060        0x0017880103548895 force poll (2)
```

## additional details

```
version: "2"
services:
  deconz:
    image: marthoc/deconz:armhf-2.07.01
    #image: marthoc/deconz:armhf-2.05.84
    container_name: deconz
    restart: unless-stopped
    ports:
      - 8085:8080
      - 4043:445
    volumes:
      - /opt/deconz/config:/root/.local/share/dresden-elektronik/deCONZ
      - /etc/localtime:/etc/localtime:ro
    devices:
      - /dev/ttyACM0
    environment:
      - DECONZ_WEB_PORT=8080
      - DECONZ_WS_PORT=445
      - DECONZ_DEVICE=/dev/ttyACM0
      - DEBUG_INFO=2
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
```

Tooa, 2021-01-08:
@ebaauw  @manup Can you guys please reopen? For me the bug report looks according to the guidelines.

ebaauw, 2021-01-08:
I could, but then the bot closes it again.  @Mimiix, what's wrong here?

SwoopX, 2021-01-08:
The button handler is not the problem since Philips devices weren't and still are not covered by the button maps.

Mimiix, 2021-01-08:
> I could, but then the bot closes it again. @Mimiix, what's wrong here?

User didn't comply. Expected Behavior (as per template), user had Expected Behavio**u**r.

Tooa, 2021-01-08:
> User didn't comply. Expected Behavior (as per template), user had Expected Behaviour.

Thanks for helping out here. 

> The button handler is not the problem since Philips devices weren't and still are not covered by the button maps.

So, I guess the message `No button handler for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 001` was added in the newer release and is expected behavior.

Do you need other information from me to find the cause then?

Tooa, 2021-01-08:
The lower part showing version 2.05.84 wasn't executed with a debug flag value of 2. I only did this for the non-Wirkung version. 

Do you want me to run the working version 2.05.84 with the debug flag enabled? Might this help you?

SwoopX, 2021-01-09:
Version 2.09.00 in my production network:

```
21:44:04:037 enqueue event state/buttonevent for /sensors/3
21:44:04:037 enqueue event state/eventduration for /sensors/3
21:44:04:037 enqueue event state/lastupdated for /sensors/3
21:44:04:038 [INFO] - No button map for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 002
21:44:04:040 read attributes of 0x00158D0001FE575E cluster: 0x0702: [ 21:44:04:040 0x0000 21:44:04:041 0x0400 21:44:04:041 ]
21:44:04:041 add task 212 type 19 to 0x00158D0001FE575E cluster 0x0702 req.id 126
21:44:04:041 Poll APS request 126 to 0x00158D0001FE575E cluster: 0x0702
21:44:04:047 Websocket 127.0.0.1:59596 send message: {"e":"changed","id":"3","r":"sensors","state":{"buttonevent":2000,"eventduration":0,"lastupdated":"2021-01-09T20:44:04.037"},"t":"event","uniqueid":"00:17:88:01:04:f4:2f:61-02-fc00"} (ret = -1098597144)
21:44:04:059 discard sensor state push for 3: state/eventduration (already pushed)
21:44:04:063 discard sensor state push for 3: state/lastupdated (already pushed)
21:44:04:135 enqueue event config/localtime for /config/
21:44:04:137 rule event /config/localtime: 21:44:03.215 -> 21:44:04.134 (919ms)
21:44:04:261 Poll APS confirm 126 status: 0xE9
21:44:04:261 0x00158D0001FE575E error APSDE-DATA.confirm: 0xE9 on task
21:44:04:261 Erase task req-id: 126, type: 19 zcl seqno: 142 send time 0, profileId: 0x0104, clusterId: 0x0702
21:44:04:284 enqueue event state/buttonevent for /sensors/3
21:44:04:285 enqueue event state/eventduration for /sensors/3
21:44:04:285 enqueue event state/lastupdated for /sensors/3
21:44:04:285 [INFO] - No button map for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 002
21:44:04:289 Websocket 127.0.0.1:59596 send message: {"e":"changed","id":"3","r":"sensors","state":{"buttonevent":2002,"eventduration":2,"lastupdated":"2021-01-09T20:44:04.284"},"t":"event","uniqueid":"00:17:88:01:04:f4:2f:61-02-fc00"} (ret = -1098597144)
21:44:04:301 discard sensor state push for 3: state/eventduration (already pushed)
21:44:04:304 discard sensor state push for 3: state/lastupdated (already pushed)
```

Tooa, 2021-01-09:
For me with `2.09.00` it looks like the following below. Again no `websocket` event. How does our setup differ? Have you also migrated from a previous version (including mine) or is this a fresh install? 

```
deconz    | 21:33:07:302 Poll APS confirm 3 status: 0x00
deconz    | 21:33:07:302 Erase task req-id: 3, type: 19 zcl seqno: 66 send time 0, profileId: 0x0104, clusterId: 0x0006
deconz    | 21:33:07:313 Node data 0x001788010354850d profileId: 0x0104, clusterId: 0x0006
deconz    | 21:33:07:313 0x001788010354850D: update ZCL value 0x0B/0x0006/0x0000 after 0 s
deconz    | 21:33:07:313 enqueue event event/checkgroupanyon for /groups/65520
deconz    | 21:33:07:461 enqueue event state/buttonevent for /sensors/2
deconz    | 21:33:07:461 enqueue event state/eventduration for /sensors/2
deconz    | 21:33:07:461 enqueue event state/lastupdated for /sensors/2
deconz    | 21:33:07:461 [INFO] - No button map for: RWL021 endpoint: 0x02 cluster: 0xFC00 command: 0x00 payload[0]: 001
deconz    | 21:33:07:462 discard sensor state push for 2: state/buttonevent (already pushed)
deconz    | 21:33:07:463 discard sensor state push for 2: state/eventduration (already pushed)
deconz    | 21:33:07:465 discard sensor state push for 2: state/lastupdated (already pushed)
```

SwoopX, 2021-01-10:
The difference is for sure me running bare Raspbian and I don't use HA, I've set up the system freshly like 2 weeks ago, but it also worked before. Got an update history starting somewhere early 2.05.xx.

Do you have more sensors/switches in your network? If yes, are they doing alright?

schmosby, 2021-01-10:
I'm seeing the same issue with my Lutron Connected Bulb remotes. I have a really long history of updates since whatever version came out in December 2019. This seems similar but if you suggest a new issue, I can make a new one.

v2.0.9 shows that same error message and also does NOT pass the button presses to Home Assistant or Node-RED when listening for deconz-events on port 40850. 

I also noticed that I was not able to correctly pair the remote under v.2.0.9, the system generated group for the switch when adding via phoscon never was generated, I had to spin up a version of v84 in order to add it correctly (and then add assign lights to the system generated group via the old interface). The group # (26567 below) in the api/sensors call in config was always null for 2.0.9, this is the remote I did the v84 and 2.0.9 comparisons with.

```
   "9": {
    "config": {
      "group": "26567",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "1902d2cdb9d256624398f9ec3d35481d",
    "lastseen": "2021-01-10T01:28Z",
    "manufacturername": "Lutron",
    "mode": 4,
    "modelid": "LZL4BWHL01 Remote",
    "name": "Bedroom Remote",
    "state": {
      "buttonevent": 2003,
      "lastupdated": "2021-01-10T00:07:19.828"
    },
    "swversion": "1.1",
    "type": "ZHASwitch",
    "uniqueid": "ff:ff:00:0f:e7:ff:61:aa-01-1000"
  }
}
```

For both logs I hit all four buttons starting from the bottom.

**v.2.0.5.84 - Prior version working fine and sending out events**
```
20:21:53:632 Mgmt_Lqi_req zdpSeq: 201 to 0x841826000009302F start index 0
20:21:53:632 APS-DATA.request id: 222, addrmode: 0x03, addr: 0x841826000009302f, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
20:21:53:659 APS-DATA.confirm id: 222, status: 0x00 SUCCESS
20:21:53:659 APS-DATA.confirm request id: 222 -> confirmed, timeout 1610241713
20:21:53:674 APS-DATA.indication srcAddr: 0x676e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 135, rssi: -77
20:21:53:674 APS-DATA.indication request id: 222 -> finished
20:21:53:674 APS-DATA.request id: 222 erase from queue
20:21:53:674 ZDP status = 0x00 -> SUCCESS
20:21:53:674 ZDP Mgmt_Lqi_rsp zdpSeq: 201 from 0x841826000009302F total: 22, startIndex: 0, listCount: 3
20:21:53:674     * neighbor: 0xCCCCCCFFFE94B85D (0x904A), LQI: 56, relation: 0x02 rxOnWHenIdle: 1
20:21:53:674     * neighbor: 0x00212EFFFF04FB61 (0x0000), LQI: 46, relation: 0x02 rxOnWHenIdle: 1
20:21:53:674     * neighbor: 0x84182600000934FB (0xA446), LQI: 66, relation: 0x02 rxOnWHenIdle: 1
20:21:53:816 APS-DATA.indication srcAddr: 0xefd8, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -51
20:21:53:816 [INFO] - Button 4002 off
20:21:53:819 Websocket 127.0.0.1:43358 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2021-01-10T01:21:53.816"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
20:21:53:820 Websocket 192.168.29.231:60214 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2021-01-10T01:21:53.816"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
20:21:53:822 discard sensor state push for 9: state/lastupdated (already pushed)
20:21:53:872 Node 0xCCCCCCFFFE9C8DA3 is known by 19 neighbors, last seen 10 s
20:21:54:057 poll node cc:cc:cc:ff:fe:97:4e:a1-01
20:21:54:057 Poll light node Bedroom Nightstand
20:21:54:109 read attributes of 0xCCCCCCFFFE974EA1 cluster: 0x0006: [ 20:21:54:110 0x0000 20:21:54:110 ]
20:21:54:110 add task 4019 type 19 to 0xCCCCCCFFFE974EA1 cluster 0x0006 req.id 229
20:21:54:110 Poll APS request 229 to 0xCCCCCCFFFE974EA1 cluster: 0x0006
20:21:54:112 APS-DATA.request id: 229, addrmode: 0x03, addr: 0xccccccfffe974ea1, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x01 queue: 0 len: 5 tx.options 0x00
20:21:54:146 Poll APS confirm 229 status: 0x00
20:21:54:146 Erase task req-id: 229, type: 19 zcl seqno: 180 send time 0, profileId: 0x0104, clusterId: 0x0006
20:21:54:146 APS-DATA.confirm id: 229, status: 0x00 SUCCESS
20:21:54:146 APS-DATA.confirm request id: 229 -> erase from queue
20:21:54:152 APS-DATA.indication srcAddr: 0xeedc, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -51
20:21:54:152 APS-DATA.request id: 229 erase from queue
20:21:54:152 ZCL got data for node=0xEEDC, cl=0x0006, at=0x0000, status=0x00, type=0x10
20:21:54:152 Node data 0xccccccfffe974ea1 profileId: 0x0104, clusterId: 0x0006
20:21:54:153 0xCCCCCCFFFE974EA1: update ZCL value 0x01/0x0006/0x0000 after 0 s
20:21:54:154 Websocket 127.0.0.1:43358 send message: {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"id":"35","lastannounced":null,"lastseen":"2021-01-10T01:21Z","manufacturername":"The Home Depot","modelid":"Ecosmart-ZBT-A19-CCT-Bulb","name":"Bedroom Nightstand","swversion":"2.3","type":"Color temperature light","uniqueid":"cc:cc:cc:ff:fe:97:4e:a1-01"},"e":"changed","id":"35","r":"lights","t":"event","uniqueid":"cc:cc:cc:ff:fe:97:4e:a1-01"} (ret = 400)
20:21:54:155 Websocket 192.168.29.231:60214 send message: {"attr":{"colorcapabilities":0,"ctmax":65279,"ctmin":0,"id":"35","lastannounced":null,"lastseen":"2021-01-10T01:21Z","manufacturername":"The Home Depot","modelid":"Ecosmart-ZBT-A19-CCT-Bulb","name":"Bedroom Nightstand","swversion":"2.3","type":"Color temperature light","uniqueid":"cc:cc:cc:ff:fe:97:4e:a1-01"},"e":"changed","id":"35","r":"lights","t":"event","uniqueid":" [... cut ...]
20:34:03:067 [INFO] - No button map for: LZL4BWHL01 Remote endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 0FE
20:34:03:080 Node 0xCCCCCCFFFE94D92B is known by 7 neighbors, last seen 4 s
20:34:03:559 Node 0xCCCCCCFFFE974EA1 is known by 7 neighbors, last seen 2 s
20:34:03:712 enqueue event config/localtime for /config/
20:34:03:800 Mgmt_Lqi_req zdpSeq: 201 to 0xCCCCCCFFFE94B85D start index 6
```

Events not received, they should have been sent in-between these two light entries.
[image]

SwoopX, 2021-01-10:
@schmosby Please open a new issue for this. Stay brief and recycle the info you've already given here. Your case is different and the fix is easy. Will reply on that issue then, thanks.

Tooa, 2021-01-10:
> The difference is for sure me running bare Raspbian and I don't use HA, I've set up the system freshly like 2 weeks ago, but it also worked before. Got an update history starting somewhere early 2.05.xx.

The home assistant automation for the RWL021 acts on the deconz events it receives. I think we can rule out the home assistant part here, because the web socket events are not being by send deconz as we can see in the logs. 

```bash
- alias: 'Dimmer - Turn on Living Room light when 1 click'
  trigger:
    platform: event
    event_type: deconz_event
    event_data:
      id: switch_living_room
      event: 1002
  action:
```

> Do you have more sensors/switches in your network? If yes, are they doing alright?

I have two Philips hue motion sensors that work as before i.e. reporting all values and state changes correct. My other Philips hue light bulbs also work as expected. I finally own two `RWL021` and both do not work in my automation any more. The battery level of them is reported correct though. 

The question boils down to what could break the web socket events in my case in the change set of `marthoc/deconz:armhf-2.05.87`?

SwoopX, 2021-01-15:
To be honest, I got no clue what to do here since I cannot reproduce this. As shown above, everything's working as expected in my setup. Have you already tried re-pairing one of those?

@ebaauw Any further ideas maybe?

manup, 2021-05-06:
Is this still not working or can this be closed?

I've tested the Hue Dimmer RWL021 with date code 20160302 in deCONZ v2.11.2 no problems here, all button events are generated.

c-t-d, 2021-05-21:
@Tooa: I had exact the same issue and could resolve it by: 
- delete the dimmer switch in phoscon
- assign/connect the dimmer switch in phoscon webapp
--> the dimmer switch comes as a new device (and group) in the websocket-events and events are generated again.
Sorry if this hint has already been posted elsewhere.

lettore, 2022-07-08:
Not sure if this is related but for at least 3 months I have had a lot of pain with this model of remote in Deconz, I have 10 remotes RWL021 and at least 4 of them are constantly not working, when you press the button it blinks red light instead of green. I can reset them and they work immediately for less than one day. Then again stop working and usually they freeze the entire network for something like 10 seconds.
I also have 2 ZYCT-202 that are not supported in UI? (I see them in light) and they work great as I look for events in Node-red.
Ikea remotes also work perfectly.
I'm not sure what's is broken with Philips because I used them for years with Deconz and I didn't add/change anything in the last few months.

closed 2025-05-02 by manup