dresden-elektronik/deconz-rest-plugin #4139: ZNCZ04LM (lumi.plug.mmeu01) fail after deconz reboot

net1957, 2021-01-08:
## Describe the bug
There is a bit of a problem with my ZNCZ04LM.
It works fine but after raspberry pi reboot I cannot control smart plug until I press button on it.
## Steps to reproduce the behavior
 - Reboot Pi
 - no action work on the smart plugs
 - push the button on it
 - all actions are working

## Expected behavior
It should work

## Screenshots

## Environment
 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian)
 - Firmware version: (26680700)
 - deCONZ version: (2.07.01)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: RFLINK and SSD
 - Main usage with Jeedom 4.0.62

## deCONZ Logs
## Additional context
The plugs are new, So I don't know if this was working in previous releases.

information for the plugs (all are on same sw version and have the same problem):
{
"8": {
"etag": "0cc603de166f4750f23e32d2adf62130",
"hascolor": false,
"lastannounced": "2020-12-20T11:57:27Z",
"lastseen": "2021-01-08T19:45Z",
"manufacturername": "LUMI",
"modelid": "lumi.plug.mmeu01",
"name": "Prise_4",
"state": {
"alert": "none",
"on": true,
"reachable": true
},
"swversion": "09-06-2019",
"type": "Smart plug",
"uniqueid": "04:cf:8c:df:3c:86:77:ce-01"
}, .

Issue #2583 reference some users with the same problem

SwoopX, 2021-01-15:
I got the device here and cannot reproduce your issue.

archiespb, 2021-01-16:
Sad to hear. I have two similar devices and both have this bug. At the same time the LLKZMK11LM relay works fine.
What information do you need to find and repair this bug?

SwoopX, 2021-01-22:
@archiespb Can you please read the attributes from the lumi specific cluster in deconz GUI and share a screenshot? I might have stumpled upons something.

archiespb, 2021-01-23:
Sure.

[image]

SwoopX, 2021-01-23:
Hm, too bad. Was after attribute 0x0009, which seems to be related to reporting as well. It should be set to 1, as it already is...

apantsiop, 2021-01-25:
I have the same issue with part GMR4014GL (which is identified lumi.plug.mmeu01, that is why I added this comment). 

2 plugs, same behavior.

kgeree, 2021-01-26:
I have the same issue with Hassio + deconz + conbee I. and the ZNCZ04LM (Xiaomi Mi Smart Plug - Firmware: 09-06-2019)
Deconz version: 6.6.2
Conbee version: 2.07.01 / 12/8/2020
Conbee firmware: 26390500

HassIO versions: core-2021.1.5, supervisor-2021.01.7
Its very annoying.. let me know if I can help providing any info to resolve this issue.

SwoopX, 2021-02-07:
I can now confirm this behaviour. Upon restart, it can obviously happen that deconz considers the node as a zombie and drops all requests to the device. Consequently, nothing is send on zigbee level.

`´´
23:56:45:541 add task 542 type 14 to 0x04CF8CDF3C764934 cluster 0x0006 req.id 153
23:56:45:541 drop request to zombie (rx = 0)
´´`

closed 2021-02-08 by manup

Zeddy-, 2021-02-08:
I happened to stumble upon this same issue, using zigbee2mqtt with Slaesh on a raspberry Pi. So it sounds like the issue is with the hardware

archiespb, 2021-02-13:
Yes, it works! With some delay - about couple of minutes after host rebooting, but it works. Thanks, guys!

bcutter, 2021-02-13:
I updated deCONZ (HA docker container) and 30 minutes later the device is still not reachable. Does NOT work (yet) for me.

@archiespb how long did you need to wait for the plug to become responsive?

archiespb, 2021-02-13:
@bcutter about one or two minutes.

bcutter, 2021-02-17:
> @bcutter about one or two minutes.

No way, not for me. No luck with this piece of ***** (self-censored...) hardware.

Tested it three times meanwhile. Latest test: even after one hour the plug is not reacting to switch on/off commands. Would either need to wait like forever (not sure how many hours) or push the local button. Both options are inacceptable. I don't understand why the integration in deCONZ has this issue. Other ZigBee gateways, other software - no single issue at all. Need to blame deCONZ I guess.

This plug and deCONZ is a pure headache creator, so frustrating.

bcutter, 2021-02-23:
Final discovery:

I need to *manually read simple descriptor(s)* to avoid needing the local push button. After that it immediately responds to remote actions (switching on/off).

Why can't deCONZ handle this automatically during startup/when building the mesh?