dresden-elektronik/deconz-rest-plugin #4140: IKEA TRÅDFRI TRÅDFRI bulb E27 CWS opal 600lm RGBW Bulb Color Temperature not working

Davst, 2021-01-09:
## Describe the bug
Color temperature now works for other IKEA bulbs, but this IKEA bulb (an older version of the RGBW bulb listed as 0x2801 in the  still doesn't report support for mireds in Home Assistant and in Phoscon the color temp doesn't work. Oddly enough you can get it in the old deCONZ gui.

Think this hardware was overlooked with the recent fixes. 

## Steps to reproduce the behavior

ebaauw, 2021-01-09:
This is not a bug.  The light doesn't support colour temperature in hardware nor firmware.

Davst, 2021-01-09:
> This is not a bug. The light doesn't support colour temperature in hardware nor firmware.

This is incorrect.

From the product URL: "600-lumen-wireless-dimmable-colour-and-white-spectrum"

From the product description @ https://www.ikea.com/gb/en/p/tradfri-led-bulb-e27-600-lumen-wireless-dimmable-colour-and-white-spectrum-globe-opal-white-00408612/

> With a smart LED bulb you can dim and switch the colour of your lighting, and adapt your lighting to different activities, **for example a warmer light for dinner and a brighter, colder light for working.**
Use TRÅDFRI remote to control up to 10 LED bulbs that will all behave in the same way. Turn on and off, dim and change between a total of 9 different colours/white tones.
When adding TRÅDFRI gateway (sold separately) and IKEA Home smart app, you can change between a total of 20 different colours/white tones. You can also create several groups of light sources and control them in different ways.
The LED light source consumes up to 85% less energy and lasts 20 times longer than incandescent bulbs.
Helps lower your electric bill because dimming the lights saves energy.
Can only be used with IKEA Smart lighting products.
Add the gateway and app to also **switch between Cold sky (6000 Kelvin), Cool daylight (5000 Kelvin), Sunrise (3000 Kelvin)**, Peach, Dark red, Light pink, Dark pink, Dark purple, Blue, Lime and Yellow.
You need one of the steering devices in the TRÅDFRI series (remote control, wireless dimmer or wireless motion sensor) in order to connect your light sources with the gateway and the app.
The light from this LED bulb feels as strong as the light from a traditional 48W incandescent bulb.
LED life time approx. 25.000 hours.
Use an opal light bulb if you have an ordinary lamp shade or lamp and want an even, diffused distribution of light.
Standby power consumption: 0.5W.
The light bulb can be used in temperatures from -20°C till +40°C.
This product bears the CE mark.
Download IKEA Home smart app for free via Google Play or App Store, depending on which mobile device you have.
Works with IKEA Home smart.
With the remote control you can switch between **Cool white (4000 Kelvin), Warm white (2700 Kelvin), Warm glow (2200 Kelvin), Candlelight (1780 Kelvin)**, Warm amber, Dark peach, Pink, Light purple, and Light blue.
This product allows wireless dimming. With IKEA Smart lighting, you can dim your lights without a wired-in installation.
The bulb is not suitable to use with wired dimmers.
TRÅDFRI remote control and gateway are sold separately.

Example from the old deCONZ web interface where you can, in fact change color temperature:
Same bulb
Color spectrum: https://i.imgur.com/MMc1uVL.png
Color Temperature: https://i.imgur.com/6475XMh.png

ebaauw, 2021-01-09:
I stand corrected.  Can you please post a screenshot of the _Color Control_ cluster from the deCONZ GUI, making sure to read the attributes.  Especially interest in the bottom attributes, 0x400A, 0x400B, and 0x400C.

Can you also please list the `/lights` resource for this bulb, as exposed by the REST API?

Davst, 2021-01-09:
Move to Color Temperature: returns UNSUP_CLUSTER_COMMAND 

Color Control Cluster: 
[image]

```
/lights response

"25": {
    "colorcapabilities": 8,
    "etag": "0c9b9d04b9bfb7b3b6afda80629bc21f",
    "hascolor": true,
    "lastannounced": "2021-01-09T11:31:19Z",
    "lastseen": "2021-01-09T15:55Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 CWS opal 600lm",
    "name": "Adrian Tak",
    "state": {
      "alert": "none",
      "bri": 254,
      "colormode": "xy",
      "effect": "0",
      "hue": 0,
      "on": true,
      "reachable": true,
      "sat": 0,
      "xy": [
        0.5,
        0.389
      ]
    },
    "swversion": "1.3.013",
    "type": "Color light",
    "uniqueid": "90:fd:9f:ff:fe:20:66:68-01"
  },
```

Davst, 2021-01-09:
The question is what the old webui sends this bulb when changing color temp.. since that clearly works

Davst, 2021-01-09:
Response from one currently using white light

```
"30": {
    "colorcapabilities": 8,
    "etag": "0c9b9d04b9bfb7b3b6afda80629bc21f",
    "hascolor": true,
    "lastannounced": "2021-01-08T23:52:13Z",
    "lastseen": "2021-01-09T15:55Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 CWS opal 600lm",
    "name": "Hall",
    "state": {
      "alert": "none",
      "bri": 3,
      "colormode": "xy",
      "effect": null,
      "hue": 0,
      "on": true,
      "reachable": true,
      "sat": 0,
      "xy": [
        0.458,
        0.41
      ]
    },
    "swversion": "1.3.013",
    "type": "Color light",
    "uniqueid": "d0:cf:5e:ff:fe:28:a7:91-01"
  }
```

ebaauw, 2021-01-09:
> Move to Color Temperature: returns UNSUP_CLUSTER_COMMAND
>    ` "colorcapabilities": 8,`

So I stand un-corrected.  The light does **not** support colour temperature.  Color Capabilities is a bitmap, and 0x08 means only _Current X_ and _Current Y_.  This is consistent with the `Color Light` device type.  The REST API plugin rightly doesn't expose `state.ct`.

> The question is what the old webui sends this bulb when changing color temp.. since that clearly works

It translates the colour temperature into corresponding `xy` values.  It also does that for _Extended Color Light_ bulbs, resulting in ugly whites because the light will use the RGB instead of the CCT channels.  It only seems to send `ct` to _Color Temperature Light_ bulbs.  Of course, when the web app was created, there weren't yet any RGB+CCT lights on the market.

Davst, 2021-01-10:
In this case there is a MASSIVE difference if i try to set the “white” as color instead of via the white balance both in quality of light color which blends perfectly with other color temp bulbs and in brightness which goes much brighter than the rgb combinations allow. 

This is being discussed in https://github.com/Koenkk/zigbee2mqtt/issues/4845 as well with similar conclusions. 

The weird thing is that the old web UI seems to be able to set it to the white balance mode.

Davst, 2021-01-10:
After some further experimentation... I might have been able to achieve close to this result in the RGB settings of the old web interface... and after some more experimentation a somewhat similar result in the new pwa interface.... but this is waaay harder to achieve.

It seems.. that the W led triggers when saturation is less than half for certain color ranges.. it does seem that the far cold spectrum dips into blue rgb mode with a lot less lumens available and possibly the same at the very furthest warm side, but when you get into a certain white balance range you get a very noticeable brightness boost from when the W led kicks in.

This is also the case if I use a remote to reset the light.. where I get a great warm white color and brightness that can't quite be achieved again with the RGB controls but that I can get close to with the color balance control of the old webUI.

In the other thread that I linked some users had tried making a scene from current settings in the light, and then had some success beefing that scene back to the light to achieve the same brightness level. Also reported that it was hard to manually send those values as they got a slightly different result.

My theory is that the IKEA bulb triggers the W led to handle certain color temperatures in the RGB range, I believe that the old webUI is hitting close to these and that I can get the W led with some blending of the RGB leds... even though I'm not hitting that exact sweet spot. This might be further motivated by that the IKEA trådfri app has a certain number of colors and whites that you can shift between (Bulb might be designed to perform very well at these specific points)

The fact that the Trådfri gateway, and resetting the bulb can produce a clear superior result leads me to believe that there is some additional "special" either order or way to set the bulb to fully go into the color spectrum mode only. Would be interesting to listen to the IKEA gateway send the white shades to the lamp to see how it achieves this.

Davst, 2021-01-10:
Also in the FCC teardown photos of this bulb you can clearly see the actual leds present https://fccid.io/FHO-LED1624G9/Internal-Photos/Internal-Photos-3631433.html

ebaauw, 2021-01-10:
> The fact that the Trådfri gateway, and resetting the bulb can produce a clear superior result leads me to believe that there is some additional "special" either order or way to set the bulb to fully [...]

The bulbs have some special configuration on factory reset, that allows the Trådfri 5-button remote to cycles through some preset states, using manufacturer-specific commands in the _Scenes_ cluster.  As soon as deCONZ configures `/groups/0` on the lights, the configuration is broken, and the special commands no longer work.  I've never been able to figure out the details.

> The weird thing is that the old web UI seems to be able to set it to the white balance mode.

You need to check what `xy` values the web UI sends.

> Also in the FCC teardown photos of this bulb you can clearly see the actual leds present

I see three LEDs for the E27, but hard to tell what colours.  Looks like three different types of LEDs to me, presumably RGB.  You would have to power the light and check what LEDs light up under what conditions.  For "real" colour temperature lights use two (additional) channels for colour temperatures: warm white and a cool white channel (WWCW or RGB+CCT); RGBW produces bright white, but only of a fixed temperature.

Davst, 2021-01-10:
Sorry. This picture shows the Outputs for the leds as well https://fccid.io/FHO-LED1624G9/Internal-Photos/Internal-Photos-3631436

yeah I’ll try to figure out the current settings for the old UI. At this point I’m feeling more inclined to either get a tradfri gateway or replace the bulbs as controlling them via phoscon or HA via deCONZ will be through “hacks”

TheAppService, 2021-01-21:
+1 For this, I'm having the same problem at the moment! Did you figure out some solution already @Davst ? How could I possibly help?

Davst, 2021-01-26:
@TheAppService Not really.. I don't have the possibility to sniff out the method it sets the bulb... currently considering if I have to get a Trådfri gateway to get this thing to work or if i should avoid Ikeas color bulbs alltogether

kvanisterdael, 2021-02-19:
I am trying to control this via the REST api. Setting the "xy" gives me the correct color, but I can't seem to change the brightness. Setting the "bri" via the API doesn't have any impact on the brightness of the light bulb. 
Anyone with the same experience, that the brightness is not working?

GSzabados, 2021-03-14:
>  "xy": [
>         0.5,
>         0.389

I came across this issue meanwhile I was searching the aforementioned x and y values. I have exactly the same bulb, but I see the values from a different point.

I use Zigbee2MQTT with a CC2531 dongle. The bulbs gives me the aforementioned x, y values, when I ask Google to set the light to Warm White, what I assume it should be 2700K or 3000K. But when I do a conversion from 2700 or 3000 K to x, y the result is this (using this converter: https://www.luxalight.eu/en/cie-convertor)

2700K X:0.459867518 Y:0.410600974
3000K X:0.436939256 Y:0.404076823

None of them even close to that value...

I guess there is somewhere a conversion issue regarding these x, y values calculated from K.

GSzabados, 2021-03-14:
@ebaauw, here are the 3 scenes what the remote can trigger:

https://github.com/Koenkk/zigbee2mqtt/issues/1232#issuecomment-768567712

And in that discussion some reference to the presets to use from the 5 button remote.

johanneswilkens, 2021-12-10:
Is there already an update about this issue? The lights worked perfectly, but after some update the last month on my lights the color temperature option suddenly disappeared as well...
It's quite sad I can throw all my lights in the trash now because they are basically uncontrollable...
Even an option to get a "fake" temperature control would help a lot because this clearly are cct bulbs (with some really shitty rgb as a bonus)

closed 2022-01-10 by github-actions[bot]

Floyer007, 2022-03-05:
Anything new on this?
Color temperature is gone for me as well.

Einstein2150, 2022-03-15:
Same issue here ...

Einstein2150, 2022-03-15:
> Is there already an update about this issue? The lights worked perfectly, but after some update the last month on my lights the color temperature option suddenly disappeared as well...
> It's quite [...]

Seem that when I send the "Mired color temperature" in Node-Red (contrib-deconz-plugin) it switches the white ones on ... 🤔