dresden-elektronik/deconz-rest-plugin #4142: Phoscon app not finding gateway

sabotagedninja, 2021-01-09:
## Describe the bug
Hi,

I have a problem with the Phoscon app not being able to find the conbee 2 gateway. It currently runs on a Windows 10 machine with docker installed (running a container on port 80) and Virtualbox (running a VM which advocates itself on a completely different IP address). The deConz api is running on port 80, so I shut docker down, and the VM as well, just to be sure. But it cannot find the gateway.

Funny thing is, on a different Windows 10 computer, everything works perfectly without any problems!

I've tried:
- Running as clean as possible with just deConz - no luck
- Using a usb extension cable for the conbee 2 stick - no luck
- Giving the first computer the IP address of the second computer (since it was somehow searching on that ip and not on the ip of the machine it was on) - no luck
- Throwing the pc out of the window - no luck

I haven't tried the latest deConz version 2.09 (I'm on 2,07). Will try that. 

I couldn't find anyone with the same issues online. I don't know what to do anymore. Does anyone have any idea's? 

Thanx!!

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
 - Host system: PC Windows 10
 - Running method: Windows
 - Firmware version: 26580700
 - deCONZ version: (2.07.00)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? yes, mouse + keyboard

## deCONZ Logs

## Additional context

Smanar, 2021-01-09:
Hello, 
>Running as clean as possible with just deConz - no luck

Mean without Docker/VM ?

SwoopX, 2021-01-09:
You need to upgrade to version 2.09.00.

Mimiix, 2021-01-12:
This was fixed in 2.09.00. @sabotagedninja as Swoop said. Can you check if that fixes it for you?

wanze2001, 2021-01-12:
Hallo,
ich habe auch ein Problem mit meinem ConBeeII
Er läuft auf einen Intel Server unter Proxmox, er hat lange gut funktioniert nun nach einen Update auf 2.09.00 und auf Firmware 0x26680700.bin.GCF geht der Stick nun nicht mehr. Ich kann keine Backups mehr machen und er zeigt auch an das er kein Netzwerk findet. Ich habe mittlerweile die Firmware und auch die Phoscon gedowngradet mehrfach neugestartet und alles hat nichts gebracht. Was kann ich noch machen? Ein Reset wäre die letzte Lösung da ich kein Backup gemacht habe

sabotagedninja, 2021-01-18:
Hi, sorry for not answering in a timely fashion! I upgraded the deConz application to the latest version 2.09.00. The upgrade did not fix my problem, but i've figured out what caused the problem; the Home Assistent OS running in Virtual Box. The moment I start the VM, the COM3 device (conbee2) disappears from device manager. I have no idea why, and sometimes it does appear. But when it does, the phoscon gateway is not detected. 

I've got hassos_ova-4.15.vmdk running in Virtual Box 6.1.16

I'm thinking of running Home Assistent in Docker anyway, but if you'd like I can do some investigating if that helps develop conbee. But you'll have to guide me, for I have no idea what I can do from this point forward.

Hope this helps! Cheers and thanx!

closed 2021-03-05 by github-actions[bot]