dresden-elektronik/deconz-rest-plugin #4144: ConBee II does not report connected end devices neighbor correctly

DrCoolzic, 2021-01-09:
## Describe the bug

I have a ZigBee network that uses the ConBee II as the Coordinator. The in the Zigbee network the ConBee coordinator is connected to three end devices (Xiaomi device) and one router (Philips Hue device). The router device is connected to several devices (but this is not part of the problem). 
Unfortunately it seems that the Conbee II is only reporting the Router as aneighbor but it is not reporting the end devices neighbors? I am trying to find out if this is a known bug or if my ConBee stick is broken.

## Steps to reproduce the behavior

To show the problem the ConBee II has been tested in two environments:

1) the deCONZ application running on Windows 10. With this application I get the following map
[image]
As you can see we cannot see the 3 end devices directly connected to the ConBee coordinator

2) Home assistant running ZHA integration
When zigpy request the neighbor information, only the router node is returned but not the end devices
```
2021-01-09 11:12:38 DEBUG (MainThread) [zigpy.neighbor] [0x0000] request status: Status.SUCCESS. response: Neighbors(entries=1, start_index=0, neighbor_table_list=[Neighbor(extended_pan_id=00:21:2e:ff:ff:06:19:bc, ieee=00:17:88:01:06:7b:cf:38, nwk=0xABDC, packed=37, permit_joining=<PermitJoins.Accepting: 1>, depth=1, lqi=252)])
2021-01-09 11:12:38 DEBUG (MainThread) [zigpy.neighbor] [0x0000] Done scanning. Total 1 neighbours
```
Reporting is done incorrectly even though the end devices are correctly connected and working.

## Expected behavior

ConBee should returns all neighbors (routers AND end devices).
Note that sometimes the ConBee returns the correct number of neighbors but the description's information is incorrect and therefore it is rejected.

## Screenshots

[image]

## Environment

FYI: I bought my ConBee II stick recently but found that nevertheless the firmware needed to be updated.
I have done the firmware update on Windows 10 following the procedure from here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-windows. 
The firmware is now updated to 26680700 (30 nov 2020)  but unfortunately this update did not fix the problem.
### First environment
- Host system: PC
 - Running method: Windows 10 latest
 - Firmware version: 26680700 
 - deCONZ version: 2.09.00
 - Device: ConBee II
 - Do you use an USB extension cable: no
### Second environment
 - Host system: Raspberry Pi 4
 - Running method: ZHA integration with Home assistant
 - Firmware version: 26680700 
 - Device: ConBee II
 - Do you use an USB extension cable: (no) 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

[home-assistant.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5791186/home-assistant.log)

## Additional context

None

SwoopX, 2021-01-09:
What makes you think there's anything wrong?

DrCoolzic, 2021-01-10:
I am relatively new to ZigBee and therefore I will probably use incorrect terms to describe the problem but hopefully I should be able to give you all the necessary information.

My ConBee II stick is connected to a USB port of a RPi4 system running Home Assistant. To build/use my ZigBee network I am using the ZigBee Home Automation (**ZHA**) https://www.home-assistant.io/integrations/zha/ based on the **Zigpy** https://github.com/zigpy/zigpy stack.

I have started to build the ZigBee network by adding two Xiaomi Aqara End Devices (Weather and motion devices). I had no problem to add these two devices to the network and they were working correctly. The ZHA integration that has a Network visualization capability showing all the ZigBee device nodes and the links between these nodes. In my case the three devices (one ZC + 2 ZED) where shown correctly but WITHOUT any link. Later on, I have added a Philips Hue light bulb router and two more end devices connected to this router. The network was displayed like this:
[image]
As you can see the links between the ConBee ZC and the Philips ZR is displayed correctly as well as the links between the ZR and the connected ZED, but the links between the ZC and the two ZED connected to the ConBee ZC do not show. For information many people have been reporting this problem and therefore I am not the only one.
•	https://community.home-assistant.io/t/zha-network-visualization-missing-links-lines/255388
•	https://community.home-assistant.io/t/problem-with-zha-network-visualization/261324
•	https://community.home-assistant.io/t/zigzag-panel-showing-no-graph/263918/2
•	https://community.home-assistant.io/t/zigbee-mesh-map-problem/226134

I have open a bug report in HA core (now responsible for the ZHA integration) and I have conducted some tests as requested by them
https://github.com/home-assistant/core/issues/44954#issuecomment-757376045
From what I understand, a Coordinator like the ConBee is keeping track of all the devices in the network and there must be some sort of tables to keep the routes to reach these devices. In the case of the ConBee it is possible to reach all the nodes correctly. It exist a network command that allow querying a ZC coordinator and/or a ZR router to get a list of all its neighbors. Apparently, when the ConBee ZC device receive this specific command it does NOT reports correctly all the neighbors (only reports some of them).
To illustrate the problem I have attached the home-assistant.log file that contains all ZigBee debug information (I had to set several debug flags). In this file if you search for ‘zigpy.network’ you will see the requests for findings neighbors for the ConBee ZC [0x0000] and for the Philips ZR [0xabdc]. As you can see the ZC only report one sibling connection the the ZR but not reports the connections to the Aqara ZED. On the other hand the Philips ZR does report correctly the sibling connection to the ZC and to all its ZED children.
[attached home-assistant.log file](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5792583/home-assistant.log)

The bug is about to be closed in HA core as the graph produced by the ZHA visualization tools reflects correctly the **incorrect** information returned by the ConBee ZC. 

Even though it is possible to communicate correctly to the devices connected to the ConBee it is irritating that it does not correctly return neighbors information (thus not permitting to show them on network visualization).

As mentioned above, per request of people looking at the problem, I have updated the ConBee II to the latest firmware but it did not solve the problem. As I did this on a Windows machine, I took the opportunity to test the behavior of the ConBee device using the deCONZ Windows application. I have published the graphs in my previous post and this graph DO NOT show the two devices directly connected to the ConBee (not only we do not see the links but with deCONZ we even do not see the ZED devices).
Let me know if you want me to provide more information or if you want me to do more tests

SwoopX, 2021-01-10:
I'm changing this one to a user question until further notice or justification.

Now, as for the topic raised, I still cannot follow your argumentation to be honest and bringing Xiaomi devices into play, which do not really care about zigbee standard, makes it even harder. If I'm not mistaken, ZHA doesn't use deconz at all, but the hardware from DE, so we're discussing something here. So in that sense, you're in the wrong place here

Secondly, there's different requests with different types of payload to get some entries/information from the neighbor tables. I'd assume you don't know which are used and so don't do we. It would require some traffic sniffing and analysis to underline your theory in that case. And here, the software (ZHA) determines what is send in which way. Any debug output from HA will not really be helpful here, as it only shows, what has been parsed/processes rather then transmitted.

Also, sharing a screenshot from deconz which looks totally fine does not indicate any bug. This also raises the question from where you would know which is the true parent of the Xiaomi end device other than the graphical representation in the GUI/picture.

As a side node, deconz GUI has every now and then a small glitch in drawing the connection lines, resulting in nodes having apparently no connection. This can typically be resolved by moving the node in question just a little bit, It can also take a significant amount of time for Xiaomi devices for those lines to be drawn, as they are deep sleepers (let's say up to 1h).

Lastly, there's various screenshots in various issues around here showing the exact opposite of what you describe: Connection lines from various end devices to the coordinator.

Still open there might be something odd, but it doesn't look like it.

DrCoolzic, 2021-01-11:
First of all thanks to have taken time to reply to my long post. I would like to provide you with some comments/questions about your reply.

> Now, as for the topic raised, I still cannot follow your argumentation to be honest and bringing Xiaomi devices into play, which do not really care about zigbee standard, makes it even harder. 

I may not have been clear in my description. I do not enough knowledge in ZigBee products to so I will not comment your remark concerning Xiaomi products. However, the Xiaomi sensors works perfectly with the ConBee. The only problem here is that the ConBee device is not reporting the fact that it is connected to the Xiaomi device even though the Xiaomi devices are working perfectly with the ConBee.

> If I'm not mistaken, ZHA doesn't use deconz at all, but the hardware from DE, so we're discussing something here. So in that sense, you're in the wrong place here

Indeed, you raise an interesting point. I tried to find a suitable location to report a potential problem with the ConBee, and the only place I found was this one. However, as you point out ZHA does not use deCONZ. From the quick test, I was able to do with deCONZ on Windows (during the firmware update); it seems that my problem is also present in deCONZ. Therefore, if it helps I am ready to redo all the tests on deCONZ. I have the choice to do the tests either directly on Windows or in Home Assistant using the deCONZ plug-in (which runs the deCONZ application directly on the RPi4 in the HA environment).

> Secondly, there's different requests with different types of payload to get some entries/information from the neighbor tables. I'd assume you don't know which are used and so don't do we. It would [...]

I am a little confused about your remarks. When you refer to the possibility of sniffing traffic, I guess you mean RF traffic? However, the problem I'm describing here does not require RF traffic. There must be some API call in the 'application layer' that query the 'hardware layer' of the ConBee to retrieve information stored in the ConBee tables. 
Assuming I redo the tests with the deCONZ application, is it possible to set debug flags in order to visualize the content of calls between the applicative level and the HW of the deCONZ? This would allow to find why some connected and working devices are potentially not reported as neighbors.

> Also, sharing a screenshot from deconz which looks totally fine does not indicate any bug. This also raises the question from where you would know which is the true parent of the Xiaomi end device other than the graphical representation in the GUI/picture.

Indeed, it is justified to ask this question. When I started my tests I have only put in the network the ConBee and the two Xiaomi sensors. Obviously, in this kind of network the question does not arise. Later, in order to further explore the problem, I have added a router that was correctly reported as a neighbor of the ConBee while the two Xiaomi sensors were still not shown. I then added several sensors directly to the router that have been reported correctly as a neighbors of the router. 
If I start a new test using the deCONZ I can limit the network to only the deCONZ coordinator and two/three end devices.

> It can also take a significant amount of time for Xiaomi devices for those lines to be drawn, as they are deep sleepers (let's say up to 1h).

I don't know enough about how the ZigBee protocol works but I think that the adoption of device (even a Xiaomi ) is done very quickly, even if it is true that it can take up to an hour to further send temperature information in a stable environment. It seems (pure speculation) that the coordinator keeps track of routing information (to know how to reach a child and if buffering is necessary) in some tables and keeps track neighborhood information in different tables. This would explain why the end devices works correctly and why the graph (here I am referring to ZHA using the neighbors tables) is incorrect.

> Lastly, there's various screenshots in various issues around here showing the exact opposite of what you describe: Connection lines from various end devices to the coordinator.

You are probably not use to the ZHA graph and representation. A coordinator with NWK=0 is shown as a rectangle, a router as an oval, and an end device as a circle. Therefore no the screen shot does show that the ConBee coordinator is only link to one router. The router shows four links: one to the coordinator + three to end devices. Two end devices seems not connected to anything: the that in fact are connected to the ConBee.

One last question: If I start to test the ConBee with deCONZ is there a way to reset the ConBee device so that all internal tables/information are reset. This might be as simple as starting a new network?

Let me know what you want me to do

DrCoolzic, 2021-01-11:
For more technical details please follow the link
https://github.com/home-assistant/core/issues/44954#issuecomment-757987910
it provides more technical details

SwoopX, 2021-01-11:
My friend, I appreciate your passion and time you out into this, but you're hunting ghosts!

I'll just reply to a couple of statements here and please forgive me if I'm rather explicit here and there, nothing personal at all.

> The only problem here is that the ConBee device is not reporting the fact that it is connected to the Xiaomi device even though the Xiaomi devices are working perfectly with the ConBee.

This is you "assumed problem" with ZHA. Deconz visualizes direct connections from the coordinator to Xiaomi and other enddevices. Screenshots on that are scattered here all over the place.

> it seems that my problem is also present in deCONZ

Prove that! The screenshots you've provided from deconz tell the exact opposite.

> I guess you mean RF traffic? However, the problem I'm describing here does not require RF traffic. There must be some API call in the 'application layer' that query the 'hardware layer' of the ConBee to retrieve information stored in the ConBee tables.

Yes, zigbee traffic was meant. From my perspective, it's brave to speak of a bug and during the attempt to evidence it, referring to the actual implementation which could be the culprit. The packages show you what's actually being requested and received, bare to the bone. What makes you so sure processing to the neighbors from the routers is not faulty? How do you determine who's the end devices parent?

> You are probably not use to the ZHA graph and representation. A coordinator with NWK=0 is shown as a rectangle, a router as an oval, and an end device as a circle. Therefore no the screen shot does [...]

No, I don't use it and I don't want to use it. This is the deconz repository and this is related to ZHA, different software. I'm sure they have good people over there who can figure out any potential fix in case it is needed.

You also mentioned multiple times that you have just some basic zigbee knowledge, are speculating and assuming. That's not a shame at all any not many people would like to have or require deeper zigbee knowledge. But you have to admit it's a bit of a bulldozer style to come up with a "bug" and reference a different software for that. Get that knowledge if you're interested, but this is not the adequate place here. Join the various Discord channels, build it up and search the web. You might also want to read the zigbee spec https://zigbeealliance.org/wp-content/uploads/2019/11/docs-05-3474-21-0csg-zigbee-specification.pdf to gain a better understanding and determine, if it's really a bug.

Based on solid evidence and argumentation, we can look into it in case anything's odd.

DrCoolzic, 2021-01-11:
I guess you're right. I definitely need to gain a better understanding of the ZigBee protocol rather than abuse your patience.
I already partially read the ZigBee specification (and other more concise doc) but this is a lot to digest :)

FYI Meanwhile I restarted a new ZigBee network from scratch using deCONZ that, with the help of the debug flags, provides a lot of useful information that hopefully will make me me look a little bit less ignorant.

Again thanks for your feedback. I will now close the user question because this is not the right place to discuss about the ConBee HW. Unfortunately I have not been able to find a place where to discuss technical details about the ConBee implementation of the ZigBee 3.0 stack but I will continue to search.

closed 2021-01-11 by DrCoolzic

DrCoolzic, 2021-01-12:
FYI I think I have finally found the right place to report my communication problems between ZHA and ConBee II :)
https://github.com/dresden-elektronik/deconz-serial-protocol

fennsen64, 2021-04-09:
> 
> 
> FYI I think I have finally found the right place to report my communication problems between ZHA and ConBee II :)
> https://github.com/dresden-elektronik/deconz-serial-protocol

Your 2 cents are totally correct from my experiences: If you have a Xiaoamo device (in my case Xiaomi mii smart plugs EU) as Router connected to a conbee under Deconz, the links to the neighbours (e.g. sensors) from the router are NOT shown. As well as the mesh connections between two Xiaomi routers vice versa. Only connections from the Coordinator to the router is shown in the GUI.

As soon as you add an 'good working' router (let's say a Hue Bulb), all connections fro there are coccertly shown.

Regardless: the network is working fluent, so this is only a 'graphic issue' but also for myself very irritating..

matkoo72, 2021-05-05:
> Your 2 cents are totally correct from my experiences: If you have a Xiaoamo device (in my case Xiaomi mii smart plugs EU) as Router connected to a conbee under Deconz, the links to the neighbours (e.g. sensors) from the router are NOT shown. As well as the mesh connections between two Xiaomi routers vice versa. Only connections from the Coordinator to the router is shown in the GUI.
> Regardless: the network is working fluent, so this is only a 'graphic issue' but also for myself very irritating..

I can confirm - Raspbee II + Xiaomi smart plug + xiaomi temperature sensors - working fine (okay, installed two days ago). The issue, trully irritating, is lack of plug-sensor visualisation in deCONZ. I would be happy if this could be fixed.