dresden-elektronik/deconz-rest-plugin #4147: Tuya TS0601 thermostat: “lastupdate” is empty

TheNON75, 2021-01-09:
## Describe the bug
Whenever a config or state update occurs, the "lastupdate" information is always empty.

## Steps to reproduce the behavior
Check rest api

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-01-09:
Yep , my fault.
I think there is this bug in all tuya device.

Someone can compile a modification to make a test ?

Smanar, 2021-01-10:
Ok , have made some tries with @machielw, the problem is found, will be corrected when I will add his device for next beta I think.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3183

TheNON75, 2021-01-10:
Hi @Smanar,

Thanks for the good news, great work!

Smanar, 2021-01-10:
And from his test, it solve too a "not reachable" bug.
It s enabled only for him ATM, but will add it for other command too.

TheNON75, 2021-01-10:
Excellent, @Smanar!
What was the not reachable bug in details?

Smanar, 2021-01-11:
I think it was the same problem, the "lastupdate " was never updated, and after some time the device was marked as "unreachable".

[image]

easybeat, 2021-01-19:
> I think it was the same problem, the "lastupdate " was never updated, and after some time the device was marked as "unreachable".
> 
> [image]

Oh this sounds like a similar problem on TS0601 Radiator TRV. I lose connection with this all the time...is this possible?

Smanar, 2021-01-19:
Depend of you realy loose the connxion or just have it marked as unreacheable ?
If you still receive information like temperature, yep I think it s same problem, we have soon finished the siren, I will put a link for tries here.

easybeat, 2021-02-09:
@Smanar I've just updated to 2.09.02 and it looks like on my TS0601 TRV the last updated is still null:

[image]

This is a screenshot from ioBroker. Is there a way to check this in Deconz App?

Thanks

Smanar, 2021-02-09:
The last update issue is normaly corrected here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4379 (not yet validated)

But I have made a mistake for some TRV on a previous PR too, what is your manufacture name ?

easybeat, 2021-02-09:
> The last update issue is normaly corrected here #4379 (not yet validated)
> 
> But I have made a mistake for some TRV on a previous PR too, what is your manufacture name ?

[image]

Smanar, 2021-02-09:
Arf bad luck :(, I m asking to manup if I can correct it, but ATM you will probably have problem with preset/mode request if you use them.
In worst situation the corrective will be in the beta, just after the stable, the code is already in PR list.

easybeat, 2021-02-09:
Hey, no problem!! It can wait.

Thanks for your work.

closed 2021-03-12 by github-actions[bot]

TheNON75, 2021-06-30:
Unstale

Smanar, 2021-06-30:
This bug is still present ?

TheNON75, 2021-06-30:
I had to revert from the latest version as there was at least one other issue with it, so I cannot tell right now. 
In which stable is it fixed? Sorry it is unclear from the above posts.

Smanar, 2021-07-01:
The last one is enought, long time this issue is "supposed" to be resolved.

TheNON75, 2021-07-10:
I confirm it works in the latest. 
(I might need to open a new issue for the "lastseen")

Thank you very much