dresden-elektronik/deconz-rest-plugin #4148: Drayton Wiser Radiator Thermostat

fireblade70, 2021-01-09:
## Device

- Product name: Drayton Wiser Radiator Thermostat.
- Manufacturer: Schneider Electric
- Model identifier: iTRV
- Device type : Thermostat

## Screenshots

[image]
[image]
[image]

SwoopX, 2021-01-17:
Required screenshots as per request template are missing.

fireblade70, 2021-02-25:
Can you please clarify which screenshot is missing? I followed the instructions as per [Instructions](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support). I see them clearly in my original post?
PS apologies for the delay in responding.

fireblade70, 2021-02-25:
> Can you please clarify which screenshot is missing? I followed the instructions as per [Instructions](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support). I see them clearly in my original post?
> PS apologies for the delay in responding.

Mimiix, 2021-03-03:
The cluster screenshots @fireblade70 . https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

fireblade70, 2021-03-29:
[Drayton TRV Clusters.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6222111/Drayton.TRV.Clusters.zip)
Sorry for delay. I have screenshot of all the clusters attached

Mimiix, 2021-03-30:
@fireblade70 Can you add them seperately?

closed 2021-04-25 by manup

sbuelow, 2022-02-08:
Hello
I bought a Drayton Wiser Thermostat Valve, fooled by the fact that was reported as working and supported.
After some tests I was only able to pair it with the coordinator (Conbee II), but there's no way to manage it.
Battery level and temperature are reported.
Does anybody know if it has to be reconfigured, in the Thermostat Cluster perhaps ?
Otherwise I have no choice but to return it and look for other working products.
Thanks in advance

sbuelow, 2022-02-08:
Thanks for your feedback @fireblade70
It looks like this product is on the way home :).
FYI best results up to now obtained with Eurotronic Spirit.

Feldi1606, 2024-04-16:
Does anyone use this TRV? I would really like to buy the thermostats, but based on the comments I'm not sure whether they work with deconz or not.

Feldi1606, 2024-05-29:
I have a problem with the iTRV. I connected the thermostat to Deconz and I also see some values. But the battery is always displayed to me at 10 percent. Is there an error there? And can someone help me program the weekly programs and connect them to a room thermostat from the series?

SwoopX, 2024-05-29:
Hi,

can you please share the DDF automatically generated by deconz? I'd recommend to start deconz and wait somewhere between 30 and 60 minutes before doing it, this will collect as much info as possible from the device based on the legacy code. Thanks!

Feldi1606, 2024-05-30:
I hope you can do something with it and that I did it right. Otherwise please give instructions on what exactly I have to do.

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "Schneider Electric",
  "modelid": "iTRV",
  "product": "iTRV",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/Schneider/itrv.json",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0201",
          "0x0204",
          "0x0B05"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/mode",
          "description": "Operational mode.",
          "default": 1
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "description": "The current device battery level in 0&ndash;100&thinsp;%.",
          "default": 0
        },
        {
          "name": "config/coolsetpoint",
          "description": "Target cooling temperature of a thermostat.",
          "default": 0
        },
        {
          "name": "config/heatsetpoint",
          "description": "Target temperature of a thermostat.",
          "read": {
            "at": "0x0012",
            "cl": "0x0201",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0012",
            "cl": "0x0201",
            "dt": "0x29",
            "ep": 0,
            "eval": "Item.val;",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0012",
            "cl": "0x0201",
            "ep": 0,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/locked",
          "description": "Locks or unlocks the device. This item has to be understood as child-lock.",
          "read": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0001",
            "cl": "0x0204",
            "dt": "0x30",
            "ep": 0,
            "eval": "Item.val",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0204",
            "ep": 0,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "config/mode",
          "description": "Mode of the device."
        },
        {
          "name": "config/offset",
          "description": "Relative offset to the main measured value.",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "description": "True when device is on; false when off.",
          "refresh.interval": 5,
          "read": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0006",
            "ep": 0,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/temperature",
          "description": "The current temperature in °C &times; 100.",
          "parse": {
            "at": "0x0000",
            "cl": "0x0402",
            "ep": 0,
            "eval": "Item.val = Attr.val + R.item('config/offset').val",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "state/valve",
          "description": "The open percentage of the thermostat valve.",
          "read": {
            "at": "0x0008",
            "cl": "0x0201",
            "ep": 0,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0008",
            "cl": "0x0201",
            "ep": 0,
            "eval": "Item.val = Attr.val;",
            "fn": "zcl:attr"
          },
          "default": 0
        }
      ]
    }
  ]
}`

SwoopX, 2024-05-30:
That should do fine for a start, thanks. Needs some amendments, will share something once ready.

Accodrong to the screenshots here in this issue, the device does not support any schedules/weekly programs.

Feldi1606, 2024-05-31:
I recreated all screenshots. I think the device supports weekly programs. Or am I wrong?

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2024-06-02:
> I think the device supports weekly programs. Or am I wrong?

No, it doesn't. What makes you think it does? That is on-device schedules to be more precise allowing the radiators to be steered when any home automation system or deconz is down. With an home automation system, you can of course implement whatever you want.

Could you please give the DDF here a try? #7801

Feldi1606, 2024-06-02:
That´s what I mean, I would store a Programm on the Thermostat. But I found no manual how to do this.

SwoopX, 2024-06-02:
> That´s what I mean, I would store a Programm on the Thermostat. But I found no manual how to do this.

Well, as previously said, the device doesn't support it and hence the lack of any supporting documentation.

Feldi1606, 2024-06-13:
Hello, the battery is always displayed to me at 10%. The Battery ist new so I think it musst shown with near 100%. 
FHEM shows me 2 readings "battery" and "batteryPercent" both say 10%. In Phoscon is can see Batterie with the same value. 
Is ther a Problem in the DDF?

PR #7801 (2024-06-02): DDF for Drayton Wiser Radiator Thermostat iTRV
DDF is supposed to replace the legacy implementation. Although it is expected the device will be fully functional, somebody owning the thermostat should test in deconz GUI if there are any usable system mode settings. For now, it is statically set to auto, but can be changed if necessary.

Legacy code can be removed in a subsequent PR then.

Closes #4148