dresden-elektronik/deconz-rest-plugin #4151: Lutron Connected Bulb remotes button presses are not handled and events are not broadcasted

schmosby, 2021-01-10:
## Describe the bug
Lutron Connected Bulb remote button presses are not being handled in 2.0.9 and events are not sent out when listening via Home Assistant or Node-Red @ port 40850. These events and button presses are handled in prior versions (example of v84 shown in this issue.

## Steps to reproduce the behavior

On version, 2.0.9
1. Open either Home Assistant or Node-Red and open a listener on deconz_events that are sent out on port 40850
2. Press any of the remote buttons

Repeat on version v84

## Expected behavior

1. Button presses are handled and events should make it to the listener on both versions

Actual result is that only v84 is handling the button press and sending the events

## Screenshots

## Environment

 - Host system: ( PC / Ubuntu)
 - Running method: ( Marthoc Docker container )
 - Firmware version: (2660700)
 - deCONZ version: (2.0.9)
 - Device: ( ConBee II )
 - Do you use an USB extension cable: (yes, 5 footer in the air ) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? Bluetooth adapter on opposite side of PC

## deCONZ Logs

Tests were done with the remote below, added via v84.
```
   "9": {
    "config": {
      "group": "26567",
      "on": true,
      "reachable": true
    },
    "ep": 1,
    "etag": "1902d2cdb9d256624398f9ec3d35481d",
    "lastseen": "2021-01-10T01:28Z",
    "manufacturername": "Lutron",
    "mode": 4,
    "modelid": "LZL4BWHL01 Remote",
    "name": "Bedroom Remote",
    "state": {
      "buttonevent": 2003,
      "lastupdated": "2021-01-10T00:07:19.828"
    },
    "swversion": "1.1",
    "type": "ZHASwitch",
    "uniqueid": "ff:ff:00:0f:e7:ff:61:aa-01-1000"
  }
}
```

For both logs I hit all four buttons on the remote starting from the bottom.

**v.2.0.5.84 - Prior version working fine and sending out events**
```
20:21:53:632 Mgmt_Lqi_req zdpSeq: 201 to 0x841826000009302F start index 0
20:21:53:632 APS-DATA.request id: 222, addrmode: 0x03, addr: 0x841826000009302f, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
20:21:53:659 APS-DATA.confirm id: 222, status: 0x00 SUCCESS
20:21:53:659 APS-DATA.confirm request id: 222 -> confirmed, timeout 1610241713
20:21:53:674 APS-DATA.indication srcAddr: 0x676e, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 135, rssi: -77
20:21:53:674 APS-DATA.indication request id: 222 -> finished
20:21:53:674 APS-DATA.request id: 222 erase from queue
20:21:53:674 ZDP status = 0x00 -> SUCCESS
20:21:53:674 ZDP Mgmt_Lqi_rsp zdpSeq: 201 from 0x841826000009302F total: 22, startIndex: 0, listCount: 3
[... 142 lines cut ...]
20:34:01:846 [INFO] - No button map for: LZL4BWHL01 Remote endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP (0x03) payload[0]: 000
20:34:02:119 Node 0xCCCCCCFFFE893B25 is known by 7 neighbors, last seen 5 s
20:34:02:373 APS-DATA.indication srcAddr: 0xefd8, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
20:34:02:374 [INFO] - No button map for: LZL4BWHL01 Remote endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP_WITH_ON_OFF (0x06) payload[0]: 000
20:34:02:570 APS-DATA.indication srcAddr: 0xefd8, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
20:34:02:571 [INFO] - No button map for: LZL4BWHL01 Remote endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STOP (0x03) payload[0]: 000
20:34:02:600 Node 0xCCCCCCFFFE2BE738 is known by 6 neighbors, last seen 4 s
20:34:02:711 enqueue event config/localtime for /config/
20:34:02:788 poll node 7c:b0:3e:aa:00:a4:02:3a-03
20:34:02:788 Poll light node Family Room Door
20:34:02:841 Poll APS request to 0x7CB03EAA00A4023A cluster: 0x0006 dropped, values are fresh enough
20:34:03:066 APS-DATA.indication srcAddr: 0xefd8, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -52
20:34:03:067 [INFO] - No button map for: LZL4BWHL01 Remote endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: MOVE_TO_LEVEL_WITH_ON_OFF (0x04) payload[0]: 0FE
20:34:03:080 Node 0xCCCCCCFFFE94D92B is known by 7 neighbors, last seen 4 s
20:34:03:559 Node 0xCCCCCCFFFE974EA1 is known by 7 neighbors, last seen 2 s
20:34:03:712 enqueue event config/localtime for /config/
20:34:03:800 Mgmt_Lqi_req zdpSeq: 201 to 0xCCCCCCFFFE94B85D start index 6
```

Events not received, they should have been sent in-between these two light entries.
[image]

## Additional context

I had to add/pair this remote using v84, I am unsure if it was just user error on my part but I did not have luck when paring the remote under 2.0.9 as the system generated group for the switch when adding via phoscon never was generated,

SwoopX, 2021-01-10:
To fix this instantly, you must find the file `button_maps.json` on your system and search for the following line:

`"modelids": ["LZL4BWHL"],`

Then change `LZL4BWHL` to `LZL4BWHL01 Remote` and save. Lastly, restart deconz

schmosby, 2021-01-10:
@SwoopX confirmed as working.. Thank you!!

[image]

```
21:40:49:041 [INFO] - Button 4002 off LZL4BWHL01 Remote
21:40:49:041 enqueue event state/buttonevent for /sensors/9
21:40:49:044 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2021-01-10T02:40:49.041"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:044 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2021-01-10T02:40:49.041"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:044 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":4002,"lastupdated":"2021-01-10T02:40:49.041"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:679 [INFO] - Button 3001 dimm down LZL4BWHL01 Remote
21:40:49:679 enqueue event state/buttonevent for /sensors/9
21:40:49:681 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3001,"lastupdated":"2021-01-10T02:40:49.679"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:681 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3001,"lastupdated":"2021-01-10T02:40:49.679"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:682 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3001,"lastupdated":"2021-01-10T02:40:49.679"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:905 [INFO] - Button 3003 dimm down release LZL4BWHL01 Remote
21:40:49:905 enqueue event state/buttonevent for /sensors/9
21:40:49:908 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3003,"lastupdated":"2021-01-10T02:40:49.905"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:908 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3003,"lastupdated":"2021-01-10T02:40:49.905"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:49:908 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":3003,"lastupdated":"2021-01-10T02:40:49.905"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:424 [INFO] - Button 2001 dimm up LZL4BWHL01 Remote
21:40:50:424 enqueue event state/buttonevent for /sensors/9
21:40:50:426 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2001,"lastupdated":"2021-01-10T02:40:50.424"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:426 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2001,"lastupdated":"2021-01-10T02:40:50.424"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:426 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2001,"lastupdated":"2021-01-10T02:40:50.424"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:634 [INFO] - Button 2003 dimm up release LZL4BWHL01 Remote
21:40:50:634 enqueue event state/buttonevent for /sensors/9
21:40:50:636 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2003,"lastupdated":"2021-01-10T02:40:50.634"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:636 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2003,"lastupdated":"2021-01-10T02:40:50.634"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:50:637 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":2003,"lastupdated":"2021-01-10T02:40:50.634"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:51:080 [INFO] - Button 1002 on LZL4BWHL01 Remote
21:40:51:080 enqueue event state/buttonevent for /sensors/9
21:40:51:082 Websocket 127.0.0.1:60886 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-01-10T02:40:51.080"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:51:082 Websocket 192.168.29.223:19010 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-01-10T02:40:51.080"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
21:40:51:083 Websocket 192.168.29.231:49516 send message: {"e":"changed","id":"9","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-01-10T02:40:51.080"},"t":"event","uniqueid":"ff:ff:00:0f:e7:ff:61:aa-01-1000"} (ret = 164)
```

closed 2021-01-10 by schmosby