dresden-elektronik/deconz-rest-plugin #4152: SR-ZG2833PAC-FOH

NePheus, 2021-01-10:
## Device

- Product name: [ZigBee Unterputz Modul für Friends of Hue](https://www.led-trading.de/ZigBee-Unterputz-Modul-fuer-Friends-of-Hue)
- Manufacturer: Sunricher
- Model identifier: SR-ZG2833PAC-FOH
- Device type: Switch

## Screenshots

I tried to pair this device in Phoscon (deCONZ tells me inside the 'Permit join' dialog to open the network for switches with the webapp). Phoscon shows me a device:
[image]

But I cant add all keys, I just can add one and thats not working reliable. In deCONZ I dont see this device (because its an simulated ZGP device, am I right?). I opened the Debug view inside the deCONZ GUI and get events like:
[image]

But when I press the key on my device, an event is not listed each time, more like each 5 times. Tell me if I can give your more information somehow.

I waited for a device like this so long and would be really happy about deCONZ supporting it. Thanks in advance.

NePheus, 2021-01-14:
I could have read the phoscon manual more carefully...

1. Wire the buttons of the device as described in this manual: https://www.led-trading.de/mediafiles/daten/9126.pdf [image]
2. Follow this guide: https://phoscon.de/en/support#pairing-friends-of-hue-switch
Important: The 1-4 keys are not the 'wired' numbered keys as in the first guide, but the real keys when you watch it from the front.

Now I can use each key to control my lights in the Phoscon App, everything is working fine!

closed 2021-01-14 by NePheus

Tscherno, 2021-02-11:
Hi,

how did you manage to connect it at all? I have one which was connected to Hue Bridge and switched everything to Deconz. I didn't manage to get that switch to do any connection to Deconz.

NePheus, 2021-02-11:
I just followed the steps described above and now I can assign each button in the phoscon app.

mikkihugo, 2021-05-30:
> Hi,
> 
> how did you manage to connect it at all? I have one which was connected to Hue Bridge and switched everything to Deconz. I didn't manage to get that switch to do any connection to Deconz.

Im on channel 15 so pressing button connected to K1. Is there any way to reset the SR-ZG2833PAC-C4?

kvanisterdael, 2021-06-04:
I have the SR-ZG2833PAC-FOH and I can connect it to my Phoscon gateway. However after the initial connection, the buttons never react to anything (not even a buttonevent log in the REST API).

[image]

```
"105": {
        "config": {
            "on": true
        },
        "ep": 242,
        "etag": "1729b648aa5bd18a0f5bfd73b84c10b0",
        "lastseen": "2021-06-04T11:47Z",
        "manufacturername": "PhilipsFoH",
        "modelid": "FOHSWITCH",
        "name": "Friends of Hue Switch",
        "state": {
            "buttonevent": 7002,
            "lastupdated": "2021-06-04T11:47:43.680"
        },
        "type": "ZGPSwitch",
        "uniqueid": "00:00:00:00:01:71:95:7b-f2"
    },
```

misse, 2021-09-11:
@kvanisterdael I'm running into the same issue as you - have you made any progress?