dresden-elektronik/deconz-rest-plugin #4155: Upgrade to to Deconz2.09.00 causes connection issue (not connected)

bschenzle, 2021-01-10:
## Describe the bug

As soon as I upgraded to deconz to deconz-2.09.00-qt5.deb the Phoscon App was not showing any information of the Vendor as well as Firmware is telling Not Connected. 
[image]

## Steps to reproduce the behavior

Tried a lot of things around and I dont know If my stick got smashed during an upgrade but the last trial (Step 4) with  a preinstalled Phoscon Raspian OS was working

1. The ConBeeStick is also not appearing withing the Deconz - GUI anymore
Till this point I read a couple of comments that some people went to a previous version deconz-2.07.01-qt5.deb for example and downgraded their firmware version. I follow this steps and downgraded my firmware Verion as well es went to the latest stable version (deconz-2.07.01-qt5.deb). As soon as I startet Phoscon I receive the error message network was not found.... and I am not able to do anything. 

2. I thought I got crashed something on my Pi and I completely reinstalled Raspian OS as well as Deconz but it is not changing anything in the behaviour

Tried a couple of different Firmare Versions. Flashing of the Stick works without error but it is not solving my issue so far.

3. I connected my stick to a Win 10 PC and the ConBee-Stick does not even get recognized anymore within Deconz as well as the GCFFlasher.
So I stopped on Win and did not investigate even further.

4. Last but not least I installed a preinstalled Phoscon Raspian OS on a other SD Card and just executed Phoscon.
On the preinstalled OS even the latest Firmare is runing on 2.07.00 and looks fine.

## Expected behavior

Phoscon or Deconz should be connected to the ConBee-Stick

## Screenshots

1. 
[image]
2. 
[image]
[image]
[image]
3.
[image]
[image]
[image]
4.
[image]
[image]
[image]

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian)
 - Firmware version: (26390500)
 - deCONZ version: (2.09.00)
 - Device: (ConBee I)
 - Do you use an USB extension cable: (no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

10:20:50:580 HTTP API GET /api/config?_=1610274050543 - 192.168.178.40
10:20:50:581 {"apiversion":"1.16.0","bridgeid":"0000000000000000","datastoreversion":"93","factorynew":false,"mac":"dc:a6:32:29:88:4e","modelid":"deCONZ","name":"MS-GW","replacesbridgeid":null,"starterkitid":"","swversion":"2.9.0"}
10:21:34:937 failed to reconnect to network try=4
10:21:34:939 wait reconnect 10 seconds
10:21:35:937 wait reconnect 9 seconds
10:21:36:938 wait reconnect 8 seconds
10:21:37:937 wait reconnect 7 seconds
10:21:38:937 wait reconnect 6 seconds
10:21:39:937 failed to reconnect to network try=5
10:21:39:939 wait reconnect 5 seconds
10:21:40:077 dev /dev/ttyS0 (/dev/ttyS0)
10:21:40:937 wait reconnect 4 seconds
10:21:41:938 wait reconnect 3 seconds
10:21:42:937 wait reconnect 2 seconds
10:21:43:937 wait reconnect 1 seconds
10:21:44:054 dev /dev/ttyS0 (/dev/ttyS0)
10:21:44:061 COM check bootloader
10:21:44:937 failed to reconnect to network try=6
10:21:46:063 COM timout query bootloader, assume application
10:21:47:500 device state timeout ignored in state 2
10:21:49:937 try to reconnect to network try=7
10:21:50:108 dev /dev/ttyS0 (/dev/ttyS0)
10:21:50:437 device state timeout ignored in state 2
10:21:53:437 device state timeout ignored in state 2
10:21:54:938 try to reconnect to network try=8
10:21:59:459 device state timeout ignored in state 2
10:21:59:939 try to reconnect to network try=9
10:22:00:078 dev /dev/ttyS0 (/dev/ttyS0)

## Additional context

Any thoughts on that? To be honest I dont know why I can not run it on a pure Raspian OS. Would appreciate your comments.

Mimiix, 2021-01-12:
Did you click the "update firmware" button in phoscon?

That probably broke the firmware and causes the loop. However, flashing in windows fixes it. Just ignore the disconnect loop.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

bschenzle, 2021-01-12:
Hi Mimiix,

I really appreciate your fast reponse. Somehow the manually firmware-flash which I proceeded on the weekend on a unix system did not bring the connection back. Now I followed your instruction and I upgraded the Firmare on a windows device and this brought the ConBee-Stick back to life. Now I can connect to 2.09.00 and it brings the connection back ;).

[image]

Thanks a lot and I appreciate that support

closed 2021-01-12 by bschenzle