dresden-elektronik/deconz-rest-plugin #4158: deCONZ cannot connect to ConBee II if deconz-gui service is started (gateway used/locked by web app?)

sebres, 2021-01-11:
### Summary:

#### Note: 
*Although installed from Phoscon repository, like described [here](https://phoscon.de/de/conbee2/install#raspbian), the issue affecting GUI (not a web/app) and because I found several similar issues here (e. g. #1859, #1963) I am opening it here.
Since github allows to move the issues between repo's, feel free to move it to other better matching repository (I did not found that).*

#### Issue:
GUI won't start over menu itrem, if started from shell using `deCONZ` it is opening, but shows "Not Connected" and an attempt to connect to gateway (with Connect button) does nothing, it remains disconnected.

As already said, I found some similar issues #1859, #1963, etc. But the answers provided there are only partial helpful and did not solved that in my case (see Solution section below).
I think the issue should be revisited, at least for the latest version (for linux/pi) in GUI (not in web/app) + ConBee II.

### Bug description:

If one trying to start GUI (e. g. like described [here](https://phoscon.de/de/conbee2/install#raspbian), from Menu > Programming > deCONZ), **and service deconz-gui (or even deconz) is already started**, GUI won't start at all (nothing happens).
One is able to start it from shell using `deCONZ` command, but then the status is "Not Connected" (and no reconnect possible).

I don't know whether the gateway device is accessible if it is "busy" in other process (can be shared across two processes), 
but it looks like it is unusable if one of services no matter deconz-gui or deconz running.

Start commands of services do look so:
```
$ systemctl cat deconz-gui | grep Exec
ExecStart=/usr/bin/deCONZ --http-port=80

$ systemctl cat deconz | grep Exec
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80
```
Probably because both service units starting a web-app (see `--http-port=80` parameter), device is used/locked by this http server and it is impossible to connect from GUI.
Anyway also if deconz-gui is running, web-app is accessible via http and gateway is available there (not from GUI) without any issue.

### Solution:

stop deconz-gui (and surely deconz too) and if they are down, start it using `deCONZ` command (e. g. from shell):
```bash
sudo service deconz-gui stop; 
sudo service deconz stop; # to ensure both are down
deCONZ
```
The tips like https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1859#issuecomment-531320283 are misleading, because `systemctl enable deconz-gui` will enable service at boot (and **after reboot, it is running and connection to gateway is blocked**). Anyway `enable` is for totally different purposes and not needed here at all - it is enough and necessary to stop the services.

### Steps to reproduce the behavior

As example from point of view of user within [install process](https://phoscon.de/de/conbee2/install#raspbian):

  1. add repo, and install it with apt like described
  2. try to start GUI (with menu item or direct `/usr/bin/deCONZ-autostart.sh`)
  3. no GUI starts
  4. start it with `deCONZ`
  5. GUI started, but shows "Not connected" and no possibility to reconnect.

Simple example:

  1. start service `deconz-gui` and then GUI:
```bash
sudo service deconz stop;
sudo service deconz-gui start; 
deCONZ
```
  2. GUI started, but shows "Not connected" and no possibility to reconnect.

### Expected behavior

- either GUI starts and gateway is connectable 
- or GUI starts and if services are running (blocking), there is some note or warning about the connectivity issues in that case, an attempt to connect could also try to stop both (after a confirm dialog "This causes stop of deconz services. Continue?").
- GUI not starts but an alert shows an error - cannot start GUI as long as deconz services are running.

### Environment

 - Host system: linux (Debians) or rpi (Raspbian, buster)
 - Running method: Debian / Raspbian
 - deCONZ version: 2.07.01
 - Commit: 97b0c1fa379d882d4e0089211f2430de6156827c
 - Gateway-Device: ConBee II
 - Firmware version: 0x26660700

closed 2021-01-11 by close-issue-app[bot]

retrography, 2021-02-14:
This bot response is the exact reason why such a significant issue remains unaddressed. Everyone is struggling with this since day one, specially when using the official image is not an option. They suffer until they switch to another solution or otherwise find a half-broken solution. This is the clearest description of the issue I have seen to date, and it would be foolish of deCONZ to ignore it, or to assume this specific user is misinformed. Please open this issue, fix it, and add proper documentation about it. It is a pity that such a great device is so poorly documented.

sebres, 2021-02-14:
Glad to hear it helped someone. This is probably the primary reason why I did opened the issue here.
As for documentation, I think since [wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki) is open to edit for everyone, one could also add it there, just I have sooo many things to do (persistently no time in last few months).

retrography, 2021-02-15:
I have personally solved it in a different manner, given my specific circumstance:

I needed a remote X connection and not VNC, so I had to edit the deconz-gui service and change the monitor to 10. Like this when I connect with ssh, stop deconz service and start deconz-gui, the user interface automatically launches on my XQuartz (MacOS). This only works, though, if I am logged in as the user that normally runs the daemon (e.g. hoobs, if you run hoobs). Otherwise, running deCONZ with any other user results in the normal config not being loaded.

It is a layered issue that can not be solved in one straightforward manner under different circumstances. To really solved the issue, the process needs to be simplified to reduce dependencies on the right config.

JayNaire, 2022-04-27:
My thanks to @sebres as this is still a problem 28Apr22. I've wasted hours on it. Not a good intro to the world of Zigbee!

karlquint, 2022-11-27:
Same here, encountering this issue Nov 2022.