dresden-elektronik/deconz-rest-plugin #4159: Sonoff SNZB-03 detected and added but is always "not reachable"

noisemaker00, 2021-01-11:
## Describe the question or issue you are having
I have a Sonoff SNZB-03 motion sensor, I tried to connect to deCONZ using RaspBee II and succesfully added the sensor in the Phoscon Web App, but the sensor seems not working, the web app keep saying "Not reachable"

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

jamiewhitham, 2021-01-16:
I have same issue. I have one that I got in December works fine. Another one I got on 16th Jan was found and is shown but says unreachable.

andrekosak, 2021-02-04:
same here. just got two SNZB-03 devices, both paired with "unreachable" state

jamiewhitham, 2021-02-04:
> same here. just got two SNZB-03 devices, both paired with "unreachable" state

I deleted mine and re-paired and now it works same as the older one.

closed 2021-03-05 by github-actions[bot]

kquinsland, 2021-07-09:
I have the same issue.

A `SNZB-03` PIR module with Zigbee 2 Mqtt by way of a ConBee 2.

On startup, this is what Zigbee2MQTT logs:

```
Jul 09 17:52:59 z2m docker[20826]: Zigbee2MQTT:info  2021-07-09 09:52:59: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fd)
Jul 09 17:52:59 z2m docker[20826]: Zigbee2MQTT:info  2021-07-09 09:52:59: Starting zigbee-herdsman (0.13.117)
Jul 09 17:52:59 z2m docker[20826]: Zigbee2MQTT:debug 2021-07-09 09:52:59: Using zigbee-herdsman with settings: '{"adapter":{..."serialPort":{"adapter":"deconz","path":"/dev/ttyACM0"}}'
Jul 09 17:52:59 z2m docker[20826]: CREATED DECONZ ADAPTER
Jul 09 17:52:59 z2m docker[20826]: Zigbee2MQTT:info  2021-07-09 09:52:59: zigbee-herdsman started (resumed)
Jul 09 17:52:59 z2m docker[20826]: Zigbee2MQTT:info  2021-07-09 09:52:59: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":107,"product":0,"revision":"0x266b0700","transportrev":0},"type":"ConBee2/RaspBee2"}'
```

And my PIR sensor has been unresponsive for the last ... 11 weeks! I've been busy, but am now getting a little bit of time to start looking into why the `SNZB-03` won't stay connected to the radio.

I have a little over 20 devices attached to the conbee2 and they're all within about 5m distance. The USB stick is mounted out in the open away from any source of interference so i'm reasonably confident that it's not a layer 1 issue.

Since @noisemaker00 didn't post any useful logs, i've got some  that I hope will be useful:

After attempting to reconfigure it w/ Z2m:

```
Jul 09 18:29:31 z2m docker[20826]: Zigbee2MQTT:debug 2021-07-09 10:29:31: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'My Motion'
Jul 09 18:29:31 z2m docker[20826]: Zigbee2MQTT:info  2021-07-09 10:29:31: Configuring 'My Motion'

Jul 09 18:30:32 z2m docker[20826]: Zigbee2MQTT:error 2021-07-09 10:30:32: Failed to configure 'My Motion', attempt 1 (Error: Bind 0x00124b00deadbeef/1 genPowerCfg from '0x00212efff00dbabe/1' failed (waiting for response TIMEOUT)
Jul 09 18:30:32 z2m docker[20826]:     at DeconzAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:712:23)
Jul 09 18:30:32 z2m docker[20826]:     at Generator.throw (<anonymous>)
Jul 09 18:30:32 z2m docker[20826]:     at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/deconz/adapter/deconzAdapter.js:25:65))

```

After that, I removed the device from z2m and restarted z2m and attempted to re-add the device:

```
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: Device '0x00124b00deadbeef' joined
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00deadbeef","ieee_address":"0x00124b00deadbeef"},"type":"device_joined"}'
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: Starting interview of '0x00124b00deadbeef'
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b00deadbeef","ieee_address":"0x00124b00deadbeef","status":"started"},"type":"device_interview"}'
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b00deadbeef"},"type":"device_connected"}'
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b00deadbeef"},"type":"pairing"}'
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:04: Received Zigbee message from '0x00124b00deadbeef', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID null
Jul 09 18:33:04 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:04: Skipping message, definition is undefined and still interviewing
Jul 09 18:33:06 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:06: Received Zigbee message from '0x00124b00deadbeef', type 'readResponse', cluster 'genBasic', data '{"modelId":"66666"}' from endpoint 1 with groupID null
Jul 09 18:33:06 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:06: Skipping message, definition is undefined and still interviewing
Jul 09 18:33:07 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:07: Received Zigbee message from '0x00124b00deadbeef', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"eWeLink"}' from endpoint 1 with groupID null
Jul 09 18:33:07 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:07: Received Zigbee message from '0x00124b00deadbeef', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID null
Jul 09 18:33:08 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:08: Received Zigbee message from '0x00124b00deadbeef', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID null
Jul 09 18:33:08 z2m docker[21131]: Zigbee2MQTT:debug 2021-07-09 10:33:08: Received Zigbee message from '0x00124b00deadbeef', type 'readResponse', cluster 'genBasic', data '{"appVersion":3}' from endpoint 1 with groupID null
[... cut ...]
Jul 09 18:33:15 z2m docker[21131]: Zigbee2MQTT:info  2021-07-09 10:33:15: Successfully configured '0x00124b00deadbeef'

```

Even after this successful pairing, motion events don't seem to actually trigger. I can waive my had directly in front of the sensor and I see the red LED blink twice... but the z2m logs don't show anything akin to an event in the same way they would if I had triggered a contact sensor.

Is there something that I can do w/ the ConBee to get some more useful debugging output?

graphicsxp, 2022-01-17:
I have two new Sonoff motion sensors SNZB3, and although both were successfully paired with deConz, they are both in Unreachable state. No motion is detected when waving my hand in front.
Has anyone managed to use these sensors with deConz ?

andrekosak, 2022-01-17:
> I have two new Sonoff motion sensors SNZB3, and although both were successfully paired with deConz, they are both in Unreachable state. No motion is detected when waving my hand in front. Has anyone managed to use these sensors with deConz ?

i gave up using Sonoff motion sensors. I had 4 of them. Put them together for test. All of them had reacted on a motion in a different way. Some of them just stopped working in a few days. Thrown them all into trash as i am not satisfied with my tests.

what next? got a few IKEA TRÅDFRI motion sensors - they great since 1 month. connected directly over Zigbee to my Conbee.

graphicsxp, 2022-01-17:
ok, I guess I'll give up on them too. The IKEA sensors is a good option, although I don't like how big they are. Thanks for the tip.