dresden-elektronik/deconz-rest-plugin #4162: Inspect Zigbee Mesh/ Correct latency

boehser-enkel, 2021-01-11:
## Describe the question or issue you are having
Hi,

i have some bulbs and Aqara Motion Sensor in my Bathroom.
My Motion Sensor sometimes reacts with some delay (10-30 Seconds).
I restarted, reconfigured and switched to a new motion sensor but it showed the same behaviour.
Only a restart of the Phoscon App worked but i don't want to restart everything every couple of days.

Is there a way to look into the mesh network and maybe correct the connection? With Deconz App?

greetz,
boehser enkel

## Screenshots

## Environment
2.07.01 / 8.12.2020
26390500

 - Host system:RPI
 - Running method: Raspbian
 - Firmware version:26390500
 - deCONZ version: 2.07.01
 - Device: Raspbee
## deCONZ Logs

## Additional context

Smanar, 2021-01-11:
Yep you have a better control with the desktop version but you can start with the log, available in headless setup https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

You can catch some usefull error message.

boehser-enkel, 2021-01-11:
> qt.qpa.screen: QXcbConnection: Could not connect to display
> Could not connect to any X display.

So i have no display in headless.. How do i run this?

Smanar, 2021-01-11:
oups sorry try with adding `-platform minimal`

boehser-enkel, 2021-01-12:
I couldn't reproduce the error today but maybe you can see general errors?

> 12:01:26:866 HTTP Server listen on address 0.0.0.0, port: 8090, root: /usr/share/deCONZ/webapp/
12:01:26:881 CTRL. 3.27.212:01:27:046 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
12:01:27:046 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:01:27:046 dev /dev/ttyS0 (/dev/ttyS0)
12:01:27:047 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
12:01:27:246 parent process -bash
12:01:27:246 gw run mode: normal
12:01:27:246 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
12:01:27:246 sd-card cid: 035344534333324780ffffffff013700
12:01:27:247 DB sqlite version 3.27.2
12:01:27:248 DB PRAGMA page_count: 53
12:01:27:248 DB PRAGMA page_size: 4096
12:01:27:248 DB PRAGMA freelist_count: 6
12:01:27:248 DB file size 217088 bytes, free pages 6
12:01:27:248 DB PRAGMA user_version: 7
12:01:27:248 DB cleanup
12:01:27:249 DB create temporary views
12:01:27:249 DB view [0] created
12:01:27:250 DB view [1] created
12:01:27:250 DB view [2] created
12:01:27:250 DB view [3] created
12:01:27:250 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
12:01:27:278 sql exec SELECT key FROM config2
12:01:27:278 sql exec SELECT key,value FROM config2
12:01:27:278 Load config UTC: 2020-11-27T06:04:36 from db.
12:01:27:278 Load config announceinterval: 45 from db.
12:01:27:278 Load config announceurl: https://phoscon.de/discover from db.
12:01:27:278 Load config apiversion: 2.05.88 from db.
12:01:27:278 Load config bridgeid: 00212EFFFF040CDB from db.
12:01:27:278 Load config datastoreversion: 93 from db.
12:01:27:278 Load config dhcp: true from db.
12:01:27:278 Load config discovery: false from db.
12:01:27:278 Load config factorynew: false from db.
12:01:27:278 Load config fwneedupdate: false from db.
12:01:27:278 Load config fwupdatestate: idle from db.
12:01:27:278 Load config fwversion: 0x26390500 from db.
12:01:27:278 Load config gateway: 127.0.0.1 from db.
12:01:27:278 Load config group0: 65520 from db.
12:01:27:278 Load config groupdelay: 50 from db.
12:01:27:278 Load config gwpassword: $1$8282jdkm$5Q5qTp4NwBv.Nzk9ROM1d0 from db.
12:01:27:278 Load config gwusername: delight from db.
12:01:27:278 Load config homebridge: not-managed from db.
12:01:27:278 Load config homebridge-pin:  from db.
12:01:27:278 Load config homebridgepin:  from db.
12:01:27:279 Load config homebridgeupdate: false from db.
12:01:27:279 Load config homebridgeupdateversion:  from db.
12:01:27:279 Load config homebridgeversion:  from db.
12:01:27:279 Load config ipaddress: 192.168.88.200 from db.
12:01:27:279 Load config linkbutton: false from db.
12:01:27:279 Load config localtime: 2020-11-27T07:04:36 from db.
12:01:27:279 Load config mac: 38:60:77:7c:53:18 from db.
12:01:27:279 Load config modelid: deCONZ from db.
12:01:27:279 Load config name: Zigbee-GW from db.
12:01:27:279 Load config netmask: 255.0.0.0 from db.
12:01:27:279 Load config networkopenduration: 180 from db.
12:01:27:279 Load config otauactive: false from db.
12:01:27:279 Load config otaustate: off from db.
12:01:27:279 Load config panid: 0 from db.
12:01:27:279 Load config permitjoin: 0 from db.
12:01:27:279 Load config permitjoinfull: 0 from db.
12:01:27:279 Load config port: 81 from db.
12:01:27:279 Load config portalservices: false from db.
12:01:27:279 Load config proxyaddress: none from db.
12:01:27:279 Load config proxyport: 0 from db.
12:01:27:279 Load config rfconnect: 1 from db.
12:01:27:279 Load config rfconnected: false from db.
12:01:27:279 Load config rgbwdisplay: 1 from db.
12:01:27:279 Load config runmode: systemd/headless from db.
[... 959 lines cut ...]
12:02:25:238 enqueue event config/localtime for /config/
12:02:25:239 wait reconnect 4 seconds
12:02:26:239 enqueue event config/localtime for /config/
12:02:26:239 wait reconnect 3 seconds
12:02:27:239 enqueue event config/localtime for /config/
12:02:27:239 wait reconnect 2 seconds
12:02:27:738 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
12:02:27:738 failed to reconnect to network try=8
12:02:28:238 enqueue event config/localtime for /config/
12:02:28:238 wait reconnect 1 seconds
12:02:28:279 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
12:02:28:279 COM: /dev/ttyS0 :  (0x0000/0x0000)
12:02:28:279 dev /dev/ttyS0 (/dev/ttyS0)
12:02:28:279 auto connect com /dev/ttyS0
12:02:28:289 Serial com disconnected, reason: 4
12:02:28:290 device disconnected reason: 4, index: 0
12:02:29:173 shutdown after signal(2)
12:02:29:273 saved node state in 0 ms
12:02:29:302 sync() in 29 ms
12:02:29:302 REST API plugin shutting down
12:02:29:302 DB save zll database items 0x000000C1
12:02:29:302 DB sql exec REPLACE INTO rules (rid, name, created, etag, lasttriggered, owner, status, timestriggered, actions, conditions, periodic) VALUES ('5','Rule ON','2020-11-04T09:43:22','"88b14031afced396b366933390740eb9"','none','87E4621F1A','enabled','0','[{"address":"/groups/4/action","body":{"on":true},"method":"PUT"}]','[{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/buttonevent","operator":"eq","value":1002},{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/lastupdated","operator":"dx"}]','0')
12:02:29:303 DB sql exec REPLACE INTO rules (rid, name, created, etag, lasttriggered, owner, status, timestriggered, actions, conditions, periodic) VALUES ('6','Rule OFF','2020-11-04T09:43:32','"88b14031afced396b366933390740eb9"','none','87E4621F1A','enabled','0','[{"address":"/groups/4/action","body":{"on":false},"method":"PUT"}]','[{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/buttonevent","operator":"eq","value":2002},{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/lastupdated","operator":"dx"}]','0')
12:02:29:314 DB saved in 11 ms
12:02:29:326 sync() in 11 ms

Mimiix, 2021-01-12:
Nothing there.

boehser-enkel, 2021-01-12:
Okay i‘ll try later with the error reproduced

boehser-enkel, 2021-01-12:
So there was a 6 sec delay of "Bewegungsmelder". Are you able to see anything?
What other alternative do i have?

>15:36:31:219 HTTP Server listen on address 0.0.0.0, port: 8090, root: /usr/share/deCONZ/webapp/
15:36:31:237 CTRL. 3.27.215:36:31:382 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
15:36:31:382 COM: /dev/ttyS0 :  (0x0000/0x0000)
15:36:31:383 dev /dev/ttyS0 (/dev/ttyS0)
15:36:31:383 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
15:36:31:597 parent process -bash
15:36:31:597 gw run mode: normal
15:36:31:597 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
15:36:31:597 sd-card cid: 035344534333324780ffffffff013700
15:36:31:598 DB sqlite version 3.27.2
15:36:31:599 DB PRAGMA page_count: 53
15:36:31:599 DB PRAGMA page_size: 4096
15:36:31:599 DB PRAGMA freelist_count: 5
15:36:31:599 DB file size 217088 bytes, free pages 5
15:36:31:599 DB PRAGMA user_version: 7
15:36:31:599 DB cleanup
15:36:31:600 DB create temporary views
15:36:31:600 DB view [0] created
15:36:31:601 DB view [1] created
15:36:31:601 DB view [2] created
15:36:31:601 DB view [3] created
15:36:31:601 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
15:36:31:628 sql exec SELECT key FROM config2
15:36:31:628 sql exec SELECT key,value FROM config2
15:36:31:628 Load config UTC: 2020-11-27T06:04:36 from db.
15:36:31:628 Load config announceinterval: 45 from db.
15:36:31:628 Load config announceurl: https://phoscon.de/discover from db.
15:36:31:628 Load config apiversion: 2.05.88 from db.
15:36:31:628 Load config bridgeid: 00212EFFFF040CDB from db.
15:36:31:628 Load config datastoreversion: 93 from db.
15:36:31:628 Load config dhcp: true from db.
15:36:31:628 Load config discovery: false from db.
15:36:31:628 Load config factorynew: false from db.
15:36:31:628 Load config fwneedupdate: false from db.
15:36:31:628 Load config fwupdatestate: idle from db.
15:36:31:628 Load config fwversion: 0x26390500 from db.
15:36:31:628 Load config gateway: 127.0.0.1 from db.
15:36:31:628 Load config group0: 65520 from db.
15:36:31:628 Load config groupdelay: 50 from db.
15:36:31:628 Load config gwpassword: $1$8282jdkm$5Q5qTp4NwBv.Nzk9ROM1d0 from db.
15:36:31:628 Load config gwusername: delight from db.
15:36:31:628 Load config homebridge: not-managed from db.
15:36:31:628 Load config homebridge-pin:  from db.
15:36:31:628 Load config homebridgepin:  from db.
15:36:31:628 Load config homebridgeupdate: false from db.
15:36:31:628 Load config homebridgeupdateversion:  from db.
15:36:31:628 Load config homebridgeversion:  from db.
15:36:31:628 Load config ipaddress: 192.168.88.200 from db.
15:36:31:628 Load config linkbutton: false from db.
15:36:31:628 Load config localtime: 2020-11-27T07:04:36 from db.
15:36:31:628 Load config mac: 38:60:77:7c:53:18 from db.
15:36:31:628 Load config modelid: deCONZ from db.
15:36:31:629 Load config name: Zigbee-GW from db.
15:36:31:629 Load config netmask: 255.0.0.0 from db.
15:36:31:629 Load config networkopenduration: 180 from db.
15:36:31:629 Load config otauactive: false from db.
15:36:31:629 Load config otaustate: off from db.
15:36:31:629 Load config panid: 0 from db.
15:36:31:629 Load config permitjoin: 0 from db.
15:36:31:629 Load config permitjoinfull: 0 from db.
15:36:31:629 Load config port: 81 from db.
15:36:31:629 Load config portalservices: false from db.
15:36:31:629 Load config proxyaddress: none from db.
15:36:31:629 Load config proxyport: 0 from db.
15:36:31:629 Load config rfconnect: 1 from db.
15:36:31:629 Load config rfconnected: false from db.
15:36:31:629 Load config rgbwdisplay: 1 from db.
[... 868 lines cut ...]
15:37:04:739 enqueue event config/localtime for /config/
15:37:04:739 wait reconnect 14 seconds
15:37:05:738 enqueue event config/localtime for /config/
15:37:05:738 wait reconnect 13 seconds
15:37:06:738 failed to reconnect to network try=3
15:37:06:738 enqueue event config/localtime for /config/
15:37:06:738 wait reconnect 12 seconds
15:37:06:780 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
15:37:06:780 COM: /dev/ttyS0 :  (0x0000/0x0000)
15:37:06:781 dev /dev/ttyS0 (/dev/ttyS0)
15:37:06:781 GW firmware update select /dev/ttyS0 device
15:37:07:738 enqueue event config/localtime for /config/
15:37:07:738 wait reconnect 11 seconds
15:37:08:739 enqueue event config/localtime for /config/
15:37:08:739 wait reconnect 10 seconds
15:37:09:000 shutdown after signal(2)
15:37:09:146 saved node state in 0 ms
15:37:09:169 sync() in 22 ms
15:37:09:169 REST API plugin shutting down
15:37:09:169 DB save zll database items 0x000000C1
15:37:09:170 DB sql exec REPLACE INTO rules (rid, name, created, etag, lasttriggered, owner, status, timestriggered, actions, conditions, periodic) VALUES ('5','Rule ON','2020-11-04T09:43:22','"7d42ec07d3c21561fcce0b0d32a626bb"','none','87E4621F1A','enabled','0','[{"address":"/groups/4/action","body":{"on":true},"method":"PUT"}]','[{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/buttonevent","operator":"eq","value":1002},{"address":"/sensors/ec:1b:bd:ff:fe:c4:54:76-01-1000/state/lastupdated","operator":"dx"}]','0')
15:37:09:170 DB sql exec REPLACE INTO rules (rid, name, created, etag, lasttriggered, owner, status, timestriggered, actions, conditions, periodic) VALUES ('6','Rule OFF','2020-11-04T09:43:32','"7d42ec07d3c21561fcce0b0d32a626bb"','none','87E4621F1A','enabled','0','[{"address":"/groups/4/action","body":{"on":false},"method":"PUT"}]','[{"address":"/sensors/ [... cut ...]
15:37:09:171 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:21:2e:ff:ff:04:0c:db-01', '1.0', '{"dark":false,"daylight":true,"lastupdated":"2021-01-12T14:36:31.680","status":180,"sunrise":"2021-01-12T07:23:29","sunset":"2021-01-12T15:29:01"}', '{"configured":true,"lat":"52.42265","long":"10.786546","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
15:37:09:186 DB saved in 16 ms
15:37:09:201 sync() in 15 ms

Smanar, 2021-01-12:
Hu, yes, so much connexion issue.
I think if you try to use it on windows, you will have the "usb disconnection sound" ?

Try with updating the firmware, even the version you already have. Remember don't use USB 3.0.

boehser-enkel, 2021-01-12:
I have a Conbee so no USB.
I just ran apt-get upgrade

Smanar, 2021-01-12:
But the conbee is a USB Key ?

boehser-enkel, 2021-01-12:
No not usb. GPIO

Smanar, 2021-01-12:
Arf, bad luck.
But your problem look realy same the firmware issue.

If you have SSD drie for exemple, try to use extension cable to put them away.

And try tu upload again the firmware (even the same you already have) https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

boehser-enkel, 2021-01-12:
Okay i just reflased firmware. I will see if it helps.
I have no SSD. Better use Conbee with USB cable?
But everything just worked fine for almost 2 years

SwoopX, 2021-01-12:
Can you please double check if you got this here in `/boot/config`?

```
enable_uart=1
dtoverlay=miniuart-bt
core_freq=250
```

boehser-enkel, 2021-01-12:
Thats what i got in /boot/config.txt:

>[all]
#dtoverlay=vc4-fkms-v3d
enable_uart=1
dtoverlay=pi3-miniuart-bt-overlay

SwoopX, 2021-01-12:
Can you give it a try configuring it as previously suggested? UART configuration on the Pi's is bitchy.

boehser-enkel, 2021-01-12:
I just did and rebooted. I will look into it. What does this config changes?
Do you need another Log?

SwoopX, 2021-01-12:
It should ensure the Raspbee gets the (full featured) primary UART, while the (stripped) secondary UART goes to Bluetooth. it's important that the Raspbee gets the full bandwidth available.

boehser-enkel, 2021-01-13:
Sadly i did not et better. The latency issue persists.
Here another Log (Do i have to reproduce the problem within the logging?)
>22:41:27:037 HTTP Server listen on address 0.0.0.0, port: 8090, root: /usr/share/deCONZ/webapp/
22:41:27:051 CTRL. 3.27.222:41:27:173 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:41:27:173 dev /dev/ttyAMA0 (/dev/ttyAMA0)
22:41:27:173 COM: /dev/ttyS0 :  (0x0000/0x0000)
22:41:27:173 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
22:41:27:425 parent process -bash
22:41:27:426 gw run mode: normal
22:41:27:426 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
22:41:27:426 sd-card cid: 035344534333324780ffffffff013700
22:41:27:427 DB sqlite version 3.27.2
22:41:27:428 DB PRAGMA page_count: 53
22:41:27:428 DB PRAGMA page_size: 4096
22:41:27:428 DB PRAGMA freelist_count: 6
22:41:27:428 DB file size 217088 bytes, free pages 6
22:41:27:428 DB PRAGMA user_version: 7
22:41:27:428 DB cleanup
22:41:27:429 DB create temporary views
22:41:27:430 DB view [0] created
22:41:27:430 DB view [1] created
22:41:27:430 DB view [2] created
22:41:27:430 DB view [3] created
22:41:27:430 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
22:41:27:457 sql exec SELECT key FROM config2
22:41:27:457 sql exec SELECT key,value FROM config2
22:41:27:457 Load config UTC: 2020-11-27T06:04:36 from db.
22:41:27:457 Load config announceinterval: 45 from db.
22:41:27:457 Load config announceurl: https://phoscon.de/discover from db.
22:41:27:457 Load config apiversion: 2.05.88 from db.
22:41:27:457 Load config bridgeid: 00212EFFFF040CDB from db.
22:41:27:457 Load config datastoreversion: 93 from db.
22:41:27:457 Load config dhcp: true from db.
22:41:27:457 Load config discovery: false from db.
22:41:27:457 Load config factorynew: false from db.
22:41:27:457 Load config fwneedupdate: false from db.
22:41:27:458 Load config fwupdatestate: idle from db.
22:41:27:458 Load config fwversion: 0x26390500 from db.
22:41:27:458 Load config gateway: 127.0.0.1 from db.
22:41:27:458 Load config group0: 65520 from db.
22:41:27:458 Load config groupdelay: 50 from db.
22:41:27:458 Load config gwpassword: $1$8282jdkm$5Q5qTp4NwBv.Nzk9ROM1d0 from db.
22:41:27:458 Load config gwusername: delight from db.
22:41:27:458 Load config homebridge: not-managed from db.
22:41:27:458 Load config homebridge-pin:  from db.
22:41:27:458 Load config homebridgepin:  from db.
22:41:27:458 Load config homebridgeupdate: false from db.
22:41:27:458 Load config homebridgeupdateversion:  from db.
22:41:27:458 Load config homebridgeversion:  from db.
22:41:27:458 Load config ipaddress: 192.168.88.200 from db.
22:41:27:458 Load config linkbutton: false from db.
22:41:27:458 Load config localtime: 2020-11-27T07:04:36 from db.
22:41:27:458 Load config mac: 38:60:77:7c:53:18 from db.
22:41:27:458 Load config modelid: deCONZ from db.
22:41:27:458 Load config name: Zigbee-GW from db.
22:41:27:458 Load config netmask: 255.0.0.0 from db.
22:41:27:458 Load config networkopenduration: 180 from db.
22:41:27:458 Load config otauactive: false from db.
22:41:27:458 Load config otaustate: off from db.
22:41:27:458 Load config panid: 0 from db.
22:41:27:458 Load config permitjoin: 0 from db.
22:41:27:458 Load config permitjoinfull: 0 from db.
22:41:27:458 Load config port: 81 from db.
22:41:27:458 Load config portalservices: false from db.
22:41:27:458 Load config proxyaddress: none from db.
22:41:27:458 Load config proxyport: 0 from db.
22:41:27:458 Load config rfconnect: 1 from db.
22:41:27:458 Load config rfconnected: false from db.
22:41:27:458 Load config rgbwdisplay: 1 from db.
22:41:27:458 Load config runmode: systemd/headless from db.
[... 797 lines cut ...]
22:41:51:610 enqueue event config/localtime for /config/
22:41:51:611 wait reconnect 8 seconds
22:41:52:611 failed to reconnect to network try=1
22:41:52:611 enqueue event config/localtime for /config/
22:41:52:611 wait reconnect 7 seconds
22:41:52:704 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:41:52:704 dev /dev/ttyAMA0 (/dev/ttyAMA0)
22:41:52:704 COM: /dev/ttyS0 :  (0x0000/0x0000)
22:41:52:704 GW firmware update select /dev/ttyAMA0 device
22:41:53:610 enqueue event config/localtime for /config/
22:41:53:610 wait reconnect 6 seconds
22:41:54:610 enqueue event config/localtime for /config/
22:41:54:610 wait reconnect 5 seconds
22:41:55:611 enqueue event config/localtime for /config/
22:41:55:611 wait reconnect 4 seconds
22:41:56:610 enqueue event config/localtime for /config/
22:41:56:610 wait reconnect 3 seconds
22:41:57:533 shutdown after signal(2)
22:41:57:618 saved node state in 0 ms
22:41:57:661 sync() in 42 ms
22:41:57:661 REST API plugin shutting down
22:41:57:661 DB save zll database items 0x000000C1
22:41:57:662 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:21:2e:ff:ff:04:0c:db-01', '1.0', '{"dark":true,"daylight":false,"lastupdated":"2021-01-13T21:41:27.507","status":230,"sunrise":"2021-01-13T07:22:43","sunset":"2021-01-13T15:30:33"}', '{"configured":true,"lat":"52.42265","long":"10.786546","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
22:41:57:690 DB saved in 28 ms
22:41:57:708 sync() in 17 ms

Smanar, 2021-01-14:
You have not changed something rcently on wifi / bluetooth / power supply ?
The device is correctly plugged ?

Mimiix, 2021-01-14:
The only thing i see is that it doesn't connect it properly.

Also, post these logs on pastebin instead of here. the sql lite logs are horrible...

Would be nice to see APS logs.

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

boehser-enkel, 2021-01-15:
Anoer Log with a delay: https://pastebin.com/mwjUviA9
No changes to wifi, blueoth, power. Yes it is correctly plugged.
I am running headless but i will look into vlc for APS logs.

Is that something i could possbly correct by reinstalling? I maybe want to go with homebridge image with integrated deconz in future. But it is much work because of homebridge/ homekit config and PI-hole

boehser-enkel, 2021-01-16:
Another Log with a delay: https://pastebin.com/mwjUviA9
No changes to wifi, blueoth, power. Yes it is correctly plugged.

Is that something i could possbly correct by reinstalling? I maybe want to go with homebridge image with integrated deconz in future. But it is much work because of homebridge/ homekit config and PI-hole
Better go with Conbee and USB cable? Is there a way to migrate the actual config?

Here are APS Logs: https://pastebin.com/z73aB5iZ

Smanar, 2021-01-16:
Honestly All seem fine for me for software side, no reason for something broken if you haven't changed something on the OS.
To be sure you can try a preinstalled SD card https://phoscon.de/en/conbee/sdcard.

If the new SD is working > problem on the OS.

And yes, you can migrate your config from raspbee to conbee.

boehser-enkel, 2021-01-16:
But there is a problem with a device or room where motion sensor reacts with a delay. I want to correct this

closed 2021-03-05 by github-actions[bot]