dresden-elektronik/deconz-rest-plugin #4166: FEIBIT FNB56-GAS05FB1.4 Gas Sensor without carbonmonoxide state

chlctt, 2021-01-11:
Hello,

I have bought a 3rd party gas sensor from china, which has the same case as the HEIMAN Gas Sensor from here: https://www.amazon.de/HEIMAN-Smarthome-Combustible-Gas-warner-kompatibel/dp/B0793RKQWW

This is the device I bought:

I was able to pair the device with deconz, can see device in Phoscon, can access the device in deConz GUI showing the same values as initially presented in #

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

SwoopX, 2021-01-17:
It's a blemish you can ignore. That doesn't impact functionality.

chlctt, 2021-01-18:
Sorry, I do not understand your point. Do you mean the ZHAFire state shown above will not detect fire, but carbonmonoxide? But there can be carbonmonoxide without fire?

My expactation would be, that the device is publishing the state for carbonmonoxide "ZHACarbonmonoxide", but the only state which is available is "ZHAFire".

closed 2021-03-05 by github-actions[bot]