dresden-elektronik/deconz-rest-plugin #4167: Aquara vibration sensor seems to be triggered on every connect?

michis0806, 2021-01-11:
## Describe the bug

After the update to 2.09.00 I get a vibration notification on a more or less regular basis. It seems that vibration is triggered on some updates of the temperature. Sometimes every 10 minutes. 

## Steps to reproduce the behavior

Connect deconz to ioBroker or Home Assistant, create a automation/script that triggers some notification service when vibration is detected. It's triggered regular even when no vibration was present. The vibration strength keeps the same of the last "real" detection. 

## Expected behavior

deconz notifies only when a real vibration is present.

## Screenshots
This is a screenshot from the log visualized with grafana.

[image]

## Environment

 - Host system: NAS
 - Running method: marthoc/docker 
 - Firmware version: (26680700)
 - deCONZ version: (2.09.00)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

I put the sensor in my garage on a concrete floor without any vibration. It still get's triggered falsely. I also tried a second vibration sensor with a newer/different firmware. The problem stays apparent.
[image]

ebaauw, 2021-01-11:
Did you try a fresh battery?

michis0806, 2021-01-12:
Good point. But yes ;) I also tried another sensor. Same behavior...

evan-shaw, 2021-01-12:
I also have exactly the same behaviour with my vibration sensor.

Mimiix, 2021-01-14:
Ofcourse we have a 15:31:31:684 0x00158D00033DFF73 error APSDE-DATA.confirm: 0xE9 on task (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log)

But let's ignore that for now.

Either way, i see reports here with different strenghts. So i am wondering if it is actually deconz or the device being funny. Would be easy to spot if anyone could sniff iit.

michis0806, 2021-01-16:
@Mimiix I tried two sensors, same behavior with both of them. Worked for a year or so without any problems...  How can I sniff it?

Thanks
Michael

closed 2021-03-05 by github-actions[bot]