dresden-elektronik/deconz-rest-plugin #4168: Constant disconnections

SHxKM, 2021-01-12:
## Describe the bug
After failing to get the Docker VM to see the USB port, I'm running deconz inside an Ubuntu VirtualBox with the Host computer being a MacOS machine. My Mac only has USB-C ports, so I'm using a dongle as an adapter ([see pic](https://www.ivory.co.il/files/catalog/org/1569492203p03KL.jpg)).

After mounting both the Adapter and the Conbee II, the Deconz app shows the device, and I can inspect the network. I even managed to pair two devices. However, a very short time after that, the Deconz app goes back to the initial screen, and sees no devices (Conbee II) connected.

## Steps to reproduce the behavior
1. VirtualBox Ubuntu
2. Mount USB device
3. Inspect Deconz
4. Conbee II disconnects after being available

## Expected behavior
The Conbee II stick isn't disconnected after a short time.

## Screenshots

## Environment

 - Host system: MacOS
 - Running method: VirtualBox Ubuntu 20.4
 - Firmware version: 26580700
 - deCONZ version: 2.09.00
 - Device: ConBee II
 - Do you use an USB extension cable: not sure (see text above)
 - Is there any other USB or serial devices connected to the host system? If so: The adapter that helps connect the Conbee II.

## deCONZ Logs

Since I haven't figured out how to share clipboard between host and guest yet, Here's a screenshot of the logs:

[image]

Up until EVENT_TIMEOUT the device is working properly.

## Describe the bug
After failing to get the Docker VM to see the USB port, I'm running deconz inside an Ubuntu VirtualBox with the Host computer being a MacOS machine. My Mac only has USB-C ports, so I'm using a dongle as an adapter ([see pic](https://www.ivory.co.il/files/catalog/org/1569492203p03KL.jpg)).

After mounting both the Adapter and the Conbee II, the Deconz app shows the device, and I can inspect the network. I even managed to pair two devices. However, a very short time after that, the Deconz app goes back to the initial screen, and sees no devices (Conbee II) connected.

## Steps to reproduce the behavior

## Expected behavior
The Conbee II stick isn't disconnected after a short time.

## Screenshots

## Environment

 - Host system: MacOS
 - Running method: VirtualBox Ubuntu 20.4
 - Firmware version: 26580700
 - deCONZ version: 2.09.00
 - Device: ConBee II
 - Do you use an USB extension cable: not sure (see text above)
 - Is there any other USB or serial devices connected to the host system? If so: The adapter that helps connect the Conbee II.

## Additional context

The VM still shows the Conbee II as shared with it when the disconnection occurs.

Mimiix, 2021-01-12:
I already have 3 issues here and each 3 could be the issue.

VirtualBox is known for this behavior. And we have no clue what the translation does on the USB-c. Then we have MacOS that probably does something too.

Would be intresting to see if you have any logs on APS and APS_L2

SHxKM, 2021-01-12:
> Would be intresting to see if you have any logs on APS and APS_L2

@Mimiix can you please elaborate on what these two mean? 

If VitualBox is known for these issues, and we can’t rely on Docker for Mac because of USB-passthrough issues, then I have no reliable way to run Deconz on my MacOS?

Mimiix, 2021-01-12:
In your screen you have thsoe checkboxes (logs)

SHxKM, 2021-01-12:
Enabling USB-3 and choosing that as the interface solved the problem, or so it seems.

closed 2021-01-12 by SHxKM