dresden-elektronik/deconz-rest-plugin #4170: Ikea Panel bug on brightness

slurpey, 2021-01-12:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi )
 - Running method: (Home Assistent deCONZ Add-on )
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84 / 9/14/2020)
 - Device: (ConBee III)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II

## deCONZ Logs

## Additional context

manup, 2021-01-14:
@ebaauw Do I remember correctly that in Homebridge-hue you use some serialization to prevent this from happening?

Not sure how to do this in the plugin without creating a mess of code. Later on the device implementation has some better answers for such problems by providing a queue which optionally can be executed in order if needed.

ebaauw, 2021-01-14:
I simply add a `transitiontime` of 0 before issuing a PUT with more than one attribute in the body.

closed 2021-03-05 by github-actions[bot]