dresden-elektronik/deconz-rest-plugin #4173: [Alarms] Execute custom script when alarm starts or ends (with w/o fading)

Commifreak, 2021-01-12:
## Feature request type

## Description
It would be nice to set a script path (sh file) which should be executed on the raspberry when an alarm finishes. Invl. support for fading events (start script at start or end of fading for scenes in alarms)

I want to start mplayer when the light ends fading in the morning.

## Considered alternatives

## Additional context

acid115, 2021-01-12:
Just use the integrated websocket server.
https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/websocket/

With that and a small python server script you can observe every state change or button press. And run commands ...

Commifreak, 2021-01-23:
I tried that now - but - there are messages for nearly every event - except for alarms.

https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/websocket/#message-fields

parameter r does not have any item for alarms. I tested it also by adding an alarm and let it trigger. I only see, that the brightness is raising/lowering but not why.

Commifreak, 2021-02-22:
Any chance that the websocket announces that as well?

Commifreak, 2021-02-22:
BTW: I wrote simple scrips for all of my wanted features:

* PHP: Gets alarms from deconz via API
* Bash scripts: connect bluetooth speaker and manages playing webradio when the alarm starts