dresden-elektronik/deconz-rest-plugin #4175: [Device Support Request] Tuya Zigbee Repeater - Ts0207Device name

TheAlphaLaw, 2021-01-12:
## Device

Product name: Tuya Zigbee Repeater
Manufacturer: _TZ3000_m0vaazab
Model identifier: Ts0207
Device type : Repeater

## Screenshots

[image]
[image]
[image]
[image]

nukleuz80, 2021-01-13:
Hi there,

is the repeater the one with USB?
Seems like a USB-Stick?

Thanks.

TheAlphaLaw, 2021-01-13:
> Hi there,
> 
> is the repeater the one with USB?
> Seems like a USB-Stick?
> 
> Thanks.

Correct, it's a mini USB stick with a 10 dBm range. https://zigbee.blakadder.com/Tuya_TS0207.html

nukleuz80, 2021-01-13:
> > Hi there,
> > is the repeater the one with USB?
> > Seems like a USB-Stick?
> > Thanks.
> 
> Correct, it's a mini USB stick with a 10 dBm range. https://zigbee.blakadder.com/Tuya_TS0207.html

Ok, thanks for the device request!
I have the same...

Smanar, 2021-01-13:
And it don't create device in API ?
You realy need it ? I can try to make like for the ikea one, But on my side I relay don't have utility having it in API ?

TheAlphaLaw, 2021-01-13:
> And it don't create device in API ?
> You realy need it ? I can try to make like for the ikea one, But on my side I realy don't have utility having it in API ?

Yes, I need these to repeat the signal around my house. I have 6400 sqft to cover and there are some deadzones where the zigbee devices are not routers. The IKEA repeaters are larger and cover 2-3 ports on my USB/120v outlets. The Tuya repeaters take a single USB port on each outlet and do not interfere with other connectors.

Smanar, 2021-01-13:
Yep, but you can use it even it s not visible in tha API, do you realy need it visible in the API ?

TheAlphaLaw, 2021-01-13:
> Yep, but you can use it even it s not visible in tha API, do you realy need it visible in the API ?

I see, well they don't seem to be repeating on the device map. They seem to lose connection to device while the Ikea repeaters do not.  Aside from that, it would be nice to add it into the API. That way I can pull it into Home Assistant and my device maps.

TheAlphaLaw, 2021-01-25:
@Smanar Just to confirm. The Conbee II can find the device, but I'm not seeing any mesh or signal extension.

Smanar, 2021-02-19:
@TheAlphaLaw Sorry I have totaly forget you.
BTW do you have a linux machine to test code modification ?

TheAlphaLaw, 2021-02-19:
@Smanar 

No worries! No I only have Windows machines, Home Assistant is running on a VM.

closed 2021-03-20 by github-actions[bot]

Hedda, 2021-05-11:
Any news on compatibility with Tuya Zigbee 3.0 USB Range Extender (a.k.a. Zigbee Signal Repeater) as a Zigbee Router device?

Tuya SKU: RP280

It is however not clear if it really works as a Zigbee Router with third-party gateways or if need Tuya gateway as their description:

https://community.home-assistant.io/t/tuya-zigbee-signal-repeater-working-with-zha-and-conbee-2/249329

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4175

https://github.com/Koenkk/zigbee2mqtt/issues/4104

https://zigbee.blakadder.com/Tuya_TS0207.html

Signal Repeater Zigbee 3.0 Model TS0207 manufactured by Tuya

Zigbee ID: "TS0207"; "_TZ3000_m0vaazab"; "FNB54-WTS08ML1.0"

https://www.aliexpress.com/i/1005001459823340.html?spm=2114.12057483.top.4.55932e050QWoD7

https://www.aliexpress.com/i/1005001274714963.html

https://www.aliexpress.com/item/1005002441359324.html

https://www.aliexpress.com/i/1005002526017479.html

https://www.amazon.com/Extender-Protable-Amplifier-Gateway-Repeater/dp/B08MWBKK4P

https://www.amazon.com/Repeaters%EF%BC%8CKKmoon-Amplification-Transmitters-Enhancement-Compatible/dp/B08R3QC9X6/

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Mimiix, 2021-05-11:
@Hedda Could only be checked if you have the actual device. Can't tell from this unfortunately.

TheAlphaLaw, 2022-03-24:
@nukleuz80 Is it possible to relay device states of any kind to Home Assistant?

Smanar, 2022-03-25:
I m not sure to understand, this device is just a "repeter", can't be used for detection ...

Smanar, 2022-03-25:
Ha yes I undersand, you want to use it on your car, and check for the presence on the zigbee network ?

State/reachable is not presice enought ?

TheAlphaLaw, 2022-03-25:
@Smanar Phosconz reports it as a light/water sensor. No states though, even if added to a group. Reports "Unknown" to home assistant no matter the connectivity status.

Smanar, 2022-03-26:
Phoscon is not deconz. There is lot of thing supported by the API, but not in phoscon.
Can you check the device json direclty in the API, using for exemple phoscon / help / api Information / light

As this device don't need reporting, it s possible it was too much "quiet".

TheAlphaLaw, 2022-03-29:
@Smanar 

{
    "etag": "42c20039760315979817baaaa04070e7",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2022-03-24T17:45Z",
    "manufacturername": "_TZ3000_m0vaazab",
    "modelid": "TS0207",
    "name": "Q50",
    "state": {
        "alert": "none",
        "reachable": false
    },
    "swversion": "0xFFFFFFFF",
    "type": "Range extender",
    "uniqueid": "08:6b:d7:ff:fe:ea:bc:3c-01"
}

Smanar, 2022-03-29:
So you can use the field "reachable", no ? to check presence.

TheAlphaLaw, 2022-03-29:
Yes, that would work. However, the Phoscon integration does not pull that data into Home Assistant for some reason. Deconz is running on a Windows server, instead of on my home assistant instance. So I need to use the Phoscon integration to pull in data.

Kane610, 2022-03-30:
I guess it will not work because it is expected to be a light but doesn't actually report any data expected from a light.

If it would be added to HA it would need to be exposed as a binary connectivity sensor.

Kane610, 2022-03-30:
> Whatever I can do to help, please let me know. There are no other devices I know of that can be easily used this way, because of size and power requirements.

You can start by evaluating your hypothesis for how relevant it is to use with your car.

TheAlphaLaw, 2022-03-30:
> > Whatever I can do to help, please let me know. There are no other devices I know of that can be easily used this way, because of size and power requirements.
> 
> You can start by evaluating your hypothesis for how relevant it is to use with your car.

Very relevant. WiFi won't connect fast enough. I have 4 Unifi AC HD inside and even tested an outside AP. I can't get any wifi devices (including my vehicle) to connect fast enough. ZigBee is instant when powered on and even connects farther out.

Might be the devices, protocol, or cross-talk; unknown. What I do know is this tiny ZigBee repeater destroys them all.

Goal is to open and close garage with vehicle presence.

Kane610, 2022-03-30:
Can you share events of this as it would be great to keep real life data to use with tests

TheAlphaLaw, 2022-03-30:
Can you provide an example of what you'd like to see?

Kane610, 2022-03-30:
enabling debug logging of Deconz integration in hass shows all events sent from Deconz over web socket, showing the events for connecting / disconnecting / reconnecting would be great

TheAlphaLaw, 2022-03-30:
Sure, I can show for WiFi device. Obviously not for ZigBee until it's implemented.

Kane610, 2022-03-30:
> Sure, I can show for WiFi device. Obviously not for ZigBee device until it's implemented.

This is only zigbee so no need to share anything but what is implemented in this issue

TheAlphaLaw, 2022-03-30:
I'm not sure anything will show in deconz integration for this device but I will check.

Smanar, 2022-03-30:
Ha ?
It mean "state/available" is used only on light device ? not used by defaut if present ?

Kane610, 2022-03-30:
I added basic support to pydeconz, that is not the same as adding it to the integration though.

TheAlphaLaw, 2022-03-31:
> I added basic support to pydeconz, that is not the same as adding it to the integration though.

Ok thank you! 👍 

Do I need to submit a request to those that manage the HA integration? I have Deconz installed on a windows server instead of directly in HA. Unfamiliar with next steps but happy to take them.

Kane610, 2022-03-31:
> those that manage the HA integration

Thats me!

I won't give any promise when its implemented and ready. I fully prioritise completing my current rewrite and then I have stuff I have postponed in favour of doing the rewrite.

TheAlphaLaw, 2022-03-31:
> > those that manage the HA integration
> 
> Thats me!
> 
> I won't give any promise when its implemented and ready. I fully prioritise completing my current rewrite and then I have stuff I have postponed in favour of doing the rewrite.
> 
> If its not part of the next HA release 2022.5 ping me, then I might have forgotten it

Sounds like a plan, thank you!

TheAlphaLaw, 2022-05-04:
> Thats me!
>
> I won't give any promise when its implemented and ready. I fully prioritise completing my current rewrite and then I have stuff I have postponed in favour of doing the rewrite.
>
> If its [...]

@Kane610 

It doesn't look to be on 2022.5. Any update? Thank you!

TheAlphaLaw, 2022-05-09:
@Kane610 

Checking back in, confirmed not implemented yet :)

Kane610, 2022-05-09:
There are some other things I will need to attend prior to revisiting this device. I can't promise anything about when I can get to it

TheAlphaLaw, 2022-05-27:
> There are some other things I will need to attend prior to revisiting this device. I can't promise anything about when I can get to it

Any chance on this going in for June? Happy to make a donation if you point me in the right direction. 

Thank you!

Kane610, 2022-06-03:
Before I can fix this I need to improve the unique id handling of the integration because it is not sustainable approach right now.

My profile on GitHub should have a GitHub sponsors reference.

TheAlphaLaw, 2022-09-05:
> Before I can fix this I need to improve the unique id handling of the integration because it is not sustainable approach right now.
> 
> My profile on GitHub should have a GitHub sponsors reference.

Checking back in as a reminder :) Thanks!

Kane610, 2022-09-06:
> > Before I can fix this I need to improve the unique id handling of the integration because it is not sustainable approach right now.
> > My profile on GitHub should have a GitHub sponsors reference.
> 
> Checking back in as a reminder :) Thanks!

Most architectural changes are in place, so closer than ever :)

TheAlphaLaw, 2022-11-02:
@Kane610 How are we looking? Eager to test :). Out of curiosity, will lights that lose connectivity to the Conbee II also show as unavailable in HA?

Kane610, 2022-11-02:
> @Kane610 How are we looking? Eager to test :). Out of curiosity, will lights that lose connectivity to the Conbee II also show as unavailable in HA?

If deconz marks the device unavailable so will home assistant. 

I started looking at how to do this one just a few weeks back. Shouldn't be too long sorry for the slow progress. So much to do and so little time