dresden-elektronik/deconz-rest-plugin #4176: deCONZ keeps on loosing Friends of Hue switch

SeraphimSerapis, 2021-01-13:
## Describe the bug

I'm using several Friends of Hue switches with deCONZ and most of them seem to be working just fine. The problem is that one of my switches (surrounded by several lights) seems to drop connection after a couple of days and then has to be [re-added](https://phoscon.de/en/support#pairing-friends-of-hue-switch) to ensure it works again.

I followed the official guidance to add a switch by pressing the bottom right button for 10 seconds to switch to channel 25 and then confirm by pressing the bottom left and top right button briefly.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

Version: 2.09.00 / 23/12/2020
Firmware: 26390500

 - Host system: PC
 - Running method: Marthoc Docker container
 - Firmware version: 26390500
 - deCONZ version: 2.09.00 / 23/12/2020
 - Device: ConBee I
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Aeotec Z-Stick Gen 5

## deCONZ Logs

## Additional context

Mimiix, 2021-01-13:
this is a ZGP device right?

I'd say, use that issue to talk about it. 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436

SeraphimSerapis, 2021-01-13:
Yes, that's correct. Thanks for pointing me towards #436 - leaving to to @manup to decide if he wants to keep this issue or not.

manup, 2021-01-14:
I think we can close this issue and track the progress in the: linked https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436

The upcoming changes summarized in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436#issuecomment-757308236 should solve the problem.

SeraphimSerapis, 2021-01-14:
Thanks for the feedback!

closed 2021-01-14 by SeraphimSerapis