dresden-elektronik/deconz-rest-plugin #4178: Cannot change time zone, only ect/GMT-2 available

Maddinthebrain, 2021-01-13:
## Describe the question or issue you are having

Hi, I found out that the timestamps of the REST messages to my fhem instance differs 1 hour from fhem instance, while the fhem instance has the correct time. As I wanted to change the timezone of deconz, I saw that only Etc/GMT-2 is available in the dropdown menu.

deconz version is 2.07.01 8.12.2020

How can I solve this issue?

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian 
 - Firmware version: (26390500)
 - deCONZ version: (2.07.01)
 - Device: ConBee I 
 - Do you use an USB extension cable: 0,5m yes, because of the USB3.0 SSD issue, putting the conbee away from the SSD
 - Is there any other USB or serial devices connected to the host system? If so: Weather station, USB-Stick

## deCONZ Logs

## Additional context

Maddinthebrain, 2021-01-13:
I can change it to Berlin in the old GUI. There are all options available... Is this a bug?

Smanar, 2021-01-13:
It s an edit box, have you tried to edit the text ?

Maddinthebrain, 2021-01-14:
Ah, that works Thanks.

closed 2021-01-14 by Maddinthebrain