dresden-elektronik/deconz-rest-plugin #4184: Missing states 'opening' and 'closing' in IKEA Kadrilj and IKEA Fyrtur implementation

fribse, 2021-01-13:
## Describe the bug

The IKEA Fyrtur and IKEA Kadrilj blinds connected to deconz reports the states 'closed' when the curtains are completely closed, and changes to 'open' as soon as they move.
The do not report 'closing' or 'opening' during the move, which makes it impossible to stop them while moving (as you can't detect they are moving, and thus issue a 'stop' command instead of an 'open' or 'close'.
This funcionality is present in the ikea hub.

## Steps to reproduce the behavior

  1. Press 'close' on an open curtain.
  2. The state doesn't change until it is closed.
  
or

  1. Press 'open' on a closed curtain.
  2. The state immediately changes to 'open'

  If the problem can't be reproduced and is sporadic, please provide some details
  on how often and when the issue happens.

## Expected behavior

  1. Press 'close' on an open curtain.
  2. The status changes to 'closing'
  3. When the curtain is fully closed, the state changes to closed.

or

  1. Press 'open' on an closed curtain.
  2. The status changes to 'opening'
  3. When the curtain is fully opened, the state changes to 'open'.

## Screenshots

## Environment

 - Host system: Supervised HomeAssitant on PVE on a NUC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.07.01 / 12/8/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes - USB3 cable
 - Is there any other USB or serial devices connected to the host system? AEOTEC z-wave Gen-5

## deCONZ Logs

## Additional context

ebaauw, 2021-01-13:
> The state doesn't change until it is closed.

But it does, just not `state.open`.  The blind reports its current position, and the API sends notifications for `state.lift`.  I had to round that value to 5% to limit the amount of updates in HomeKit.

fribse, 2021-01-13:
I see, but I don't see that state updated in Home Assistant, so where do the problem come from?

Mimiix, 2021-01-13:
Ha reports the current % of being opened.

@kane610 I remember having some discussion on this before. Do you recall on what issue?

Kane610, 2021-01-14:
> Ha reports the current % of being opened.
> 
> @Kane610 I remember having some discussion on this before. Do you recall on what issue?

I think it's this one you're thinking of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757

But it's not really the same, the other issue was about direction and state not being right where as this is more of a feature request for to report current movement of Ikea covers

fribse, 2021-01-14:
Well, it's not the same to have a % of opening, HA has states that says opening / closing, but if this isn't reported, then automations can't trigger on it.
They are described here:
https://developers.home-assistant.io/docs/core/entity/cover/#states
And in the code here:
https://github.com/home-assistant/core/blob/1fc4284a29602168708036c69d8d9fdea25bbc65/homeassistant/const.py#L241

Kane610, 2021-01-14:
And I haven't implemented it in the deconz integration so there are multiple steps to take here to get it to work, if the covers report it at all.

fribse, 2021-01-14:
I see, ok, now it makes sense :-)

Mimiix, 2021-01-14:
@Kane610 If the thing reports it's state, you can make the state trigger by having a calculation. For example:

If cur_position != cur_position {
        IF cur_position < cur_position {
              state = closing 
         }
         IF cur_position > cur_position {
              state = opening
         }
}

Not sure if this is going to be a feature request if it needs to be done on deCONZ end.

closed 2021-03-09 by github-actions[bot]