dresden-elektronik/deconz-rest-plugin #4189: GamaBit Ltd. Window/Door Sensor

Kristian8606, 2021-01-13:
## Device

- Product name: Contact Switch.
- Manufacturer: GamaBit Ltd.
- Model identifier: GMB-HAS-WD-B01.
- Device type : Please remove all unrelated device types. 
- Sensor
 

## Screenshots
[image]
[image]
[image]

SwoopX, 2021-01-18:
That would be indeed good if you could test it upfront.

Kristian8606, 2021-01-18:
Not yet visible in Phoscon

SwoopX, 2021-01-18:
That's because deconz is not Phoscon. It's the REST API what matters here.

For visibility in Phoscon, you'd need to request it in their repository.

Kristian8606, 2021-01-18:
I tried `GET API request ip: 80/api/67CE474F7D/sensors` but I don't see it, i see all the other sensors i have.

SwoopX, 2021-01-18:
Have you re-paired it?

Kristian8606, 2021-01-18:
Yes I tried twice, the different I see is that there is now a line in the deconz GUI connected to two bulbs as well as a connection to the conbee

SwoopX, 2021-01-18:
How does the node look like in deconz GUI.

Kristian8606, 2021-01-18:
I tried a third time. I do the search through Phoscon.
[image]

Kristian8606, 2021-01-18:
How could I find it in a database to make sure it's there?

Kristian8606, 2021-01-18:
GMB-HAS-WD-B01 I noticed a discrepancy in the Model identifier
[image]

Kristian8606, 2021-01-18:
If this is a mistake I think I wrote it wrong GMB-HAS-WD-B01 instead GMB-HAS-DW-B01

Kristian8606, 2021-01-18:
I fixed it sorry.

SwoopX, 2021-01-18:
Ah, a classic ;)

So, does it work now?

Kristian8606, 2021-01-18:
I fixed it in the title of the post but I expect you to fix it in the code.

SwoopX, 2021-01-18:
You wanted to say that you'd appreciate if I could change it in the code, I assume.

Yes, I already did that.

Kristian8606, 2021-01-18:
Yes, is are great you :)

Kristian8606, 2021-01-18:
Looks like I'll have to wait for the next beta. I'm compiling your storage but I can't see this sensor via an API request.

SwoopX, 2021-01-18:
So, you're saying it's not working, even with the amendments?

Just to be sure, you cloned the right branch?

Kristian8606, 2021-01-18:
I follow the instructions in the description, the only difference is that I copy and clone your storage `https://github.com/SwoopX/deconz-rest-plugin.git`  is this the right way?

SwoopX, 2021-01-18:
No, it isn't. You must specifically clone and compile the branch I made the changes on:

```
git clone --branch=gamabit_oc https://github.com/SwoopX/deconz-rest-plugin.git gamabit
cd gamabit
```

closed 2021-02-22 by manup

Kristian8606, 2021-02-24:
The sensor is not yet exposed to the API.
I will try a second time to be sure and I will report.

[image]

Mimiix, 2021-02-24:
> The sensor is not yet exposed to the API.
> I will try a second time to be sure and I will report.
> 
> [image]

It seems not to be paired correctly.

Kristian8606, 2021-02-25:
Sorry the mistake is mine.
I added it again and now it works.