dresden-elektronik/deconz-rest-plugin #4191: deconz after update: Network was not found

hammermannman, 2021-01-14:
## Describe the bug
<Hi,
after the update to 26660700 my gateway shows only "Network was not found.."
Then I restart the VM because the stick is not connected after the update. After that the gateway has the old version and I can not control the devices.

## Steps to reproduce the behavior
click on update

## Screenshots

[image]

## Environment

 - Host system: unRAID
 - Running method: virtual maschine with hassio vm, deconz addon
 - Firmware version: 26580700
 - deCONZ version:  6.6.2
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: No

## deCONZ Logs
can not copy, nothing happend
[image]

Mimiix, 2021-01-14:
Hi,

As per documentation, it is strongly recommended to use a native system to update the stick 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

hammermannman, 2021-01-14:
> 
> 
> Hi,
> 
> As per documentation, it is strongly recommended to use a native system to update the stick
> 
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

thanks, that solved my problem!

closed 2021-01-14 by hammermannman