dresden-elektronik/deconz-rest-plugin #4192: [Device Support Request] Centralite Smart Plug Mini

luczk0wiak, 2021-01-14:
Product name: Smart Plug Mini 
Manufacturer: Centralite Systems
Model identifier: 4200-C
Device type : Plug

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2021-01-15:
The device should work out of the box. Do you miss anything?

luczk0wiak, 2021-01-15:
It wasn't pairing correctly, but it seems overnight to have worked itself out. Thought it wasn't supported.

closed 2021-01-15 by luczk0wiak