dresden-elektronik/deconz-rest-plugin #4193: Phoscon App: Network was not found...

DennisMTB, 2021-01-15:
Can someone give me a clue? (i'm out of options)

### Steps to reproduce the behavior

When i log in to http://192.168.2.2:81/WPA  i see the Gateway (Sometimes i can see 2 gateways (one fixed IP linux and one loopback 127.0.0.1 (see screenshot))
[image]

Logging to the gateway which shows the fixed ip gives the login credentials.
After login i see a quick webpage but after 1 second the webpage  show 's:  "Network was not found..." (www address shows:  "http://192.168.2.2:81/pwa/index.html?607460642000&gwid=0000000000000000#control/group/1")

- It worked before.
- After reinstalling RPi on a SSD it stopped working
- Working on this for about 2 weeks now (Positive is that i learned a lot ;-))
- Did twice a fresh install (after migrating to the SSB solution)
- Once i got the webpage working but the gateway in de deCONZ web app shows "not connected" (see screenshot)

[image]

- updated firmware Conbee II (on Windows 10) succesfully (mean's no errors)

### Expected behavior

Login normally in the Phoscon web app and see a connected gateway

### Screenshots

```
pi@raspberrypi:/dev $ ls -l ttyAMA0
crw-rw---- 1 root dialout 204, 64 Jan 15 11:43 ttyAMA0
```

```
pi@raspberrypi:/dev $ lsusb
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 039: ID 1cf1:0030 Dresden Elektronik
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
pi@raspberrypi:/dev $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz.service.d
           └─override.conf
   Active: active (running) since Thu 2021-01-14 17:02:00 CET; 18h ago
 Main PID: 378 (deCONZ)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/deconz.service
           └─378 /usr/bin/deCONZ --dbg-info=2 --dbg-zdp=1 --dbg-zcl=1 --db-aps=1 --dbg-error=2 --dbg-http=1 -platform mi

Jan 14 17:02:00 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
```

```
pi@raspberrypi:/dev $ sudo netstat -pat | grep LISTEN
tcp        0      0 0.0.0.0:6789            0.0.0.0:*               LISTEN      896/java
tcp        0      0 localhost:4711          0.0.0.0:*               LISTEN      839/pihole-FTL
tcp        0      0 localhost:27017         0.0.0.0:*               LISTEN      311/mongod
tcp        0      0 0.0.0.0:8843            0.0.0.0:*               LISTEN      896/java
tcp        0      0 localhost:27117         0.0.0.0:*               LISTEN      1139/bin/mongod
tcp        0      0 0.0.0.0:8880            0.0.0.0:*               LISTEN      896/java
tcp        0      0 0.0.0.0:http-alt        0.0.0.0:*               LISTEN      896/java
tcp        0      0 0.0.0.0:http            0.0.0.0:*               LISTEN      776/lighttpd
tcp        0      0 0.0.0.0:81              0.0.0.0:*               LISTEN      378/deCONZ
tcp        0      0 localhost:28017         0.0.0.0:*               LISTEN      311/mongod
tcp        0      0 0.0.0.0:82              0.0.0.0:*               LISTEN      378/deCONZ
tcp        0      0 raspberrypi.home:51828  0.0.0.0:*               LISTEN      693/python3.8
tcp        0      0 0.0.0.0:domain          0.0.0.0:*               LISTEN      839/pihole-FTL
[... 183 lines cut ...]
11:30:08:573 dev /dev/ttyAMA0
11:30:10:278 try to reconnect to network try=1
11:30:11:527 device state timeout ignored in state 2
11:30:14:528 device state timeout ignored in state 2
11:30:15:529 try to reconnect to network try=2
11:30:18:571 dev /dev/ttyAMA0
11:30:20:027 device state timeout ignored in state 2
11:30:20:528 try to reconnect to network try=3
11:30:25:028 device state timeout ignored in state 2
11:30:25:527 try to reconnect to network try=4
11:30:28:611 dev /dev/ttyAMA0
11:30:30:028 device state timeout ignored in state 2
11:30:30:527 try to reconnect to network try=5
11:30:35:028 device state timeout ignored in state 2
11:30:35:528 try to reconnect to network try=6
11:30:38:630 dev /dev/ttyAMA0
11:30:40:027 device state timeout ignored in state 2
11:30:40:528 try to reconnect to network try=7
11:30:45:027 device state timeout ignored in state 2
11:30:45:528 try to reconnect to network try=8
11:30:48:569 dev /dev/ttyAMA0
11:30:50:027 device state timeout (handled)
11:30:50:071 dev /dev/ttyAMA0
11:30:50:072 device disconnected reason: 1, index: 0
```

wouterbr, 2021-01-15:
I also get 'network not found'  I did'nt even get a change to add a light
systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-01-15 14:07:47 CET; 17min ago
 Main PID: 3546 (deCONZ)
   Memory: 8.8M
   CGroup: /system.slice/deconz.service
           └─3546 /usr/bin/deCONZ -platform minimal --http-port=80

Jan 15 14:07:47 DOMOTICZ4US systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Jan 15 14:07:48 DOMOTICZ4US deCONZ[3546]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Jan 15 14:07:48 DOMOTICZ4US deCONZ[3546]: libpng warning: iCCP: known incorrect sRGB profile
Jan 15 14:07:58 DOMOTICZ4US deCONZ[3546]: This plugin does not support propagateSizeHints()
Jan 15 14:07:58 DOMOTICZ4US deCONZ[3546]: This plugin does not support propagateSizeHints()

DennisMTB, 2021-01-15:
#3974
i've tried to "downgrade" firmware as stated in this issue. This wasn't a solution for me.

Mimiix, 2021-01-16:
Stick is not seen. However, can you provide APS level 2 debug?

DennisMTB, 2021-01-17:
Hi Mimiix, tnx for helping..

just want you to know: Yesterday i installed the Conbee 2 stick on Windows 10 and it was working fine (so i asume that the stick is ok)..

Back to Raspberry i noticed this:

pi@raspberrypi:/home $ sudo GCFFlasher_internal -l 
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path: /dev/ttyAMA0
Vendor: **0x0000**
Product:  **0x0000**
Serial;
Type: **RaspBee**

when i do dsmesg is see a USB disconnect message:

[   20.693992] capability: warning: jsvc uses 32-bit capabilities (legacy support in use)
[ 3602.367764] usb 1-1.4: USB disconnect, device number 3
[ 3602.677729] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[ 3602.816469] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 3602.816490] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3602.816507] usb 1-1.4: Product: ConBee II
[ 3602.816523] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 3602.816538] usb 1-1.4: SerialNumber: DE2403671
[ 3606.464353] usb 1-1.4: USB disconnect, device number 4
[ 3606.767785] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[ 3606.906416] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 3606.906425] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3606.906432] usb 1-1.4: Product: ConBee II
[ 3606.906439] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 3606.906445] usb 1-1.4: SerialNumber: DE2403671
[ 7208.914361] usb 1-1.4: USB disconnect, device number 5
[ 7209.212569] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[ 7209.351350] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 7209.351358] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7209.351365] usb 1-1.4: Product: ConBee II
[ 7209.351372] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 7209.351379] usb 1-1.4: SerialNumber: DE2403671
[ 7212.754864] usb 1-1.4: USB disconnect, device number 6
[ 7213.072617] usb 1-1.4: new full-speed USB device number 7 using xhci_hcd
[ 7213.211443] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 7213.211460] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7213.211475] usb 1-1.4: Product: ConBee II

Here is the  log with debug level aps =2:

`pi@raspberrypi:/boot $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/deconz.service.d
           └─override.conf
   Active: active (running) since Sun 2021-01-17 15:21:38 CET; 13min ago
 Main PID: 32254 (deCONZ)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/deconz.service
           └─32254 /usr/bin/deCONZ -platform minimal --dbg-info=1 --**dbg-aps=2** --dbg-zcl=1 --dbg-zdp=1 --http-port=81 --w

Jan 17 15:21:38 raspberrypi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.`

pi@raspberrypi:~ $ tail -f /home/deCONZ-error.log
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
15:21:38:572 HTTP Server listen on address 0.0.0.0, port: 81, root: /usr/share/deCONZ/webapp/
15:21:38:580 CTRL. 3.27.215:21:38:749 dev /dev/ttyAMA0
15:21:38:749 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
15:21:38:970 gw run mode: systemd/headless
15:21:38:970 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
15:21:38:970 sd-card cid: 035344535236344780ffffffff014b00
[... 692 lines cut ...]
14:35:33:490 device state timeout ignored in state 2
14:35:33:990 try to reconnect to network try=9
14:35:34:079 dev /dev/ttyAMA0
14:35:38:490 device state timeout ignored in state 2
14:35:38:991 try to reconnect to network try=10
14:35:43:491 device state timeout (handled)
14:35:43:492 device disconnected reason: 1, index: 0
14:35:43:991 reconnect network failed, try later
14:35:43:991 start reconnect to network
14:35:43:991 wait reconnect 15 seconds
14:35:44:080 dev /dev/ttyAMA0
14:35:44:991 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
14:35:44:991 device state timeout ignored in state 1
14:35:44:991 wait reconnect 14 seconds
14:35:45:991 wait reconnect 13 seconds
14:35:46:991 wait reconnect 12 seconds
14:35:47:990 wait reconnect 11 seconds
14:35:48:991 failed to reconnect to network try=1
14:35:48:991 wait reconnect 10 seconds
14:35:49:991 wait reconnect 9 seconds
14:35:50:991 wait reconnect 8 seconds
14:35:51:991 wait reconnect 7 seconds
14:35:52:991 wait reconnect 6 seconds
14:35:53:991 failed to reconnect to network try=2
14:35:53:991 wait reconnect 5 seconds

DennisMTB, 2021-01-17:
I'm sorry. ignore my message about the USB disconnect message doing dsmesg. 
I hadn''t rebooted. this was probably me trying yesterday ;-)

dsmesg is clean now:

[    2.064471] usb 1-1.4: new full-speed USB device number 3 using xhci_hcd
[    2.203614] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[    2.203652] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    2.203686] usb 1-1.4: Product: ConBee II
[    2.203714] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[    2.203744] usb 1-1.4: SerialNumber: DE240367`

Mimiix, 2021-01-17:
Are you using a extension cable already? If not, there's the answer.

DennisMTB, 2021-01-17:
Yes i have. I tested the cable in Windows to.

Mimiix, 2021-01-18:
Did it work without SSD?

DennisMTB, 2021-01-18:
Yes it did work without SSD.

( are there besides the "Sudo apt remove Deconz command" and rm deCONZ files and Folders, other activities i can do to do a fresh install of the deCONZ stick?)

DennisMTB, 2021-01-19:
What i don't understand is that the RPi regonizes the ConBee 2 usb stick ( dmesg & lsusb) but deCONZ Not (GCFFlasher_internal -l)

Mimiix, 2021-01-19:
The SSD causes interference anyway. So there's 2 things that could be it: Either interference or a permission thing. 

Are you using a extension cable on the SSD?

DennisMTB, 2021-01-19:
No the ssd is not connected with extension cable . This is my setup 
[image]

Mimiix, 2021-01-19:
Bingo, i think that's where the interference comes from. Can

DennisMTB, 2021-01-20:
Tnx.. i Will try this when my extension cable arrives .. i'll let you know..

DennisMTB, 2021-01-24:
Hi Mimiix 

It works!

Now i have a Ugreen 2,5" SSD to USB 3.0 case with cable AND extension cable for the Conbee 2.

Tnx 

Dennis

Mimiix, 2021-01-24:
Case closed?

Smanar, 2021-01-24:
So with the conbee an extension cable > not working
SSD on extension cable > it works ?

DennisMTB, 2021-01-24:
Yes. Now i notice that it's very sensitive. Without a ssd in usb 3, the ConBee 2 covers my house. With ssd in usb 3 they must be more than 1 meter apart..

DennisMTB, 2021-01-24:
Yes mimiix case closed and tnx again

Mimiix, 2021-01-24:
Case closed!

closed 2021-01-24 by Mimiix