dresden-elektronik/deconz-rest-plugin #4194: Cant get api reponse whens sending request at boot

Rania-M, 2021-01-15:
## Describe the bug
Hi everybody, i have an issue with the rest api, i am trying to request it using a code that send a request for new devices .
its working great when i launch it when the computer is on, but i need to request everything at boot (via a systemd service) and when i do that i get the "server replied : service unavailable" response, if annyone has an idea on how to fix this it would be great

## Steps to reproduce the behavior
- launch a script to request the api at boot

## Expected behavior
Getting the available lights ans sensors

## Screenshots
A screenshot of my systemd service
## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian 
 - Firmware version: 0x0000000
 - deCONZ version: 2.07.01
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No
- my code : qt and QNetworkAccessManager to send http requests 

## deCONZ Logs

## Additional context

Smanar, 2021-01-15:
I think this feature need to be set on your side, not on deconz one.
Try to delay your script, or make it waiting for deconz running.

Rania-M, 2021-01-15:
> I think this feature need to be set on your side, not on deconz one.
> Try to delay your script, or make it waiting for deconz running.
> 
> On my side, I m waiting for :
> 
> * the deconz REST server is online.
> * The config request contain "fwversion" and with a value, if the value = 0x000000, it mean deconz is not ready.

Thank you for your answer, my script dors wait for the deconz service to start, ans i even made it wait incident the code for a delay up  to 3 minutes.
As for the firmware, i actually showed the one displayed in deconz gui as it was a request of github issue format, but the deconz is indeed launched.
I have for proof that once the computer is on , i launch my script for the second time and it works, so it's rallye about the boot.
I forgot to add the screenshot for my service, here it is.

[image]

Rania-M, 2021-01-15:
> I think this feature need to be set on your side, not on deconz one.
> Try to delay your script, or make it waiting for deconz running.
>
> On my side, I m waiting for :
>
> * the deconz REST server is [...]

Can i ask you how do you wait and Knowles that deconz is online, do you rely on deconz service with that ?

Rania-M, 2021-01-15:
I upgraded the wait time to 5 minutes, and its working.
I Wonder if  there is a way though to Know that the conbee is online so we dont have to make an arbitrer waiting time.
Thank you

closed 2021-01-15 by Rania-M

Mimiix, 2021-01-15:
@Rania-M as I've told you on discord, this "bug report" didn't belong here. I'm happy you've got your answer, but otherwise I would've closed it now anyway.