dresden-elektronik/deconz-rest-plugin #4198: Xiaomi Aqara D1 1-gang (no neutral wire) - model: QBKG21LM

isorin, 2021-01-16:
## Device

- Product name: Xiaomi Aqara D1 1-gang (no neutral wire)
- Manufacturer: Xiaomi 
- Model identifier: lumi.switch.b1lacn02
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch

## Screenshots

Clusters of the node:
[image]

Basic cluster's attributes:
[image]

Node Info:
[image]

RealKanashii, 2021-01-29:
Same device here. If I can help count on me.

RealKanashii, 2021-01-29:
Deconz found it as only 1 device (light, no switch). Phoscon shows 4 different lights. 
The device can be on/off with phoscon using the first light discovered ( 12 in my case ).
Phoscon:
[image]
Deconz:
[image]

RealKanashii, 2021-01-29:
@isorin I discovered something. Just try the lights from phoscon. One of them will work. Then rename this light and you will see the name changed in Deconz. From now you can swtich on/off the light from "node-red" or HA. The device is not fully supported and will need to Deconz devs to add it  to the list but at least you can operate it.

isorin, 2021-01-29:
Hi, thanks for the advice, I am using 3 of these switches for more than a month now, they work for me in HA (mostly, with some annoying delays sometimes).
I tried to add support for them myself but the code around the light discovery in Deconz is a mess.
As it stands it doesn't offer access to the "model identifier" during the discovery phase and makes adding new devices basically impossible. 
Unless someone refactors it, redesigns the way it works and calls "addLightNode" with the proper info. I see only trouble ahead for deconz development.
I think I will move away from Phoscon / Deconz soon depending how the testing with Zigbee2mqtt will fare.

RealKanashii, 2021-01-29:
I see a noticeable delay only the first time you send a command. Later it works relatively fast with only a small delay. They are constantly adding support for different devices, I'm also waiting for a Tuya Siren which is in development in deconz. There're hundreds of devices and some of them have different versions ( neutral, no neutral, 2016 production, 2018 production). For now I keep my hope in deconz as I always got a nice answer and support in their channels. 
This switch seems a modified version from another xiaomi single switch so maybe they can add the code easily.

isorin, 2021-01-30:
I have 1, 2 and 3-gang Aqara D1 model 2019/2020 wall switches. They are all "no-neutral" version.
The 3-gang is discovered properly and works more stable (less variable delays in commands).
The "annoying delay" is < 1 second and only happens sometimes but when the lights are controlled by motion it is annoying.
I also ordered a Yagusmart (w/ neutral) Zigbee switch to test.
If I find more time I'll give the code another go. I'm using my own compiled version of deconz-rest-plugin. At the time when I started using these switches they were not supported at all in the official version.

Chreece, 2021-02-27:
Is this already resolved?

RealKanashii, 2021-03-06:
No. In fact, with the last beta update the switch is working worst, not being able to control it (althoug still appearing as 3 different switches.

SwoopX, 2021-04-19:
Anyone having this device would be able to compile the plugin as described here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4199#issuecomment-821901386 to test it?

closed 2021-04-25 by manup