dresden-elektronik/deconz-rest-plugin #4199: Xiaomi Aqara D1 2-gang (no neutral wire) - model: QBKG22LM

isorin, 2021-01-16:
## Device

- Product name: Xiaomi Aqara D1 2-gang (no neutral wire)
- Manufacturer: Xiaomi 
- Model identifier: lumi.switch.b2lacn02
- Device type : Please remove all unrelated device types. 
  - Light
  - Switch

## Screenshots

Clusters of the node:
[image]

Basic cluster's attributes:
[image]

Node Info:
[image]

Mimiix, 2021-02-01:
We are missing screenshots from all clusters.

esseki, 2021-02-01:
Here are the screenshots for the missing clusters 👍 

Identity Cluster
[image]

Power configuration cluster
[image]

Device Temperature Configuration cluster
[image]

Time cluster
[image]

OTAU cluster
[image]

Mimiix, 2021-02-01:
@esseki Looks good to me :)

SwoopX, 2021-02-01:
It unfortunately doesn't since the attributes haven't been read. The required clusters are: power configuration and binary output from **both** endpoints.

esseki, 2021-02-01:
Another attempt :)

Power configuration cluster (after clicking on "read")
[image]

Binary output cluster (after clicking on "read") for endpoint 02
[image]

Binary output cluster (after clicking on "read") for endpoint 03
[image]

Do you need anything else ?

SwoopX, 2021-02-01:
Hm, that's a mains powered device, right? Rubbish to expose the power configuration cluster then...

I'm a bit puzzled about the other cluster, seems something's odd, but on the deconz side probaly. Regardless, that should suffice for now, thanks.

isorin, 2021-02-01:
An update of the "Clusters of the node" - 
[image]

I don't know why some time ago when I took the 1st screenshot it only discovered 5 endpoints.
(It might be the updated deconz-rest-plugin)

If you still need screenshots of the clusters, I can provide them.
For the 2-gang or 1-gang (they look identical anyway).

I looked at the code in de_web_plugin.cpp / addLightNode() method, unfortunatelly the "model identifier" which is instrumental in adding support for these light switches is not available. "addLightNode" should be called at a time when modelId is available.
I added some debugging in the code and the modelId becomes available shortly after "addLightNode()" is called.
Unfortunately, this requires some refactoring and implementation changes.

SwoopX, 2021-02-01:
Yeah, that's gonna be a slight challenge, but I already have an idea how to overcome that with comparable low effort. Dirty, but it should work.

Now that we also got the Analog input cluster in the latest screenshot as well, that's of interest on top. I assume the values reflect the power currently consumed and change rapidly (when the lights are on)?

isorin, 2021-02-01:
From what I know these switches don't have power monitoring, so there will be nothing reported there.
[image]

This is the request for the 1-gang version:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4198

I couldn't find anything to differentiate between the 1-gang and 2-gang versions apart from the "model identifier"

esseki, 2021-02-01:
> Hm, that's a mains powered device, right? Rubbish to expose the power configuration cluster then..
I confirm this is a wired switch

Here is a screenshot for the Analog Input (Basic) cluster for the 2-gang version. As @isorin it looks pretty identical to the 1-gang version
[image]

isorin, 2021-02-01:
Sorry, I think my last post was a bit confusing as I mixed two different things together:

1) The "Meter interface" endpoint shouldn't report data as the switches don't have power monitoring capabilities.
The "Analog Cluster" snapshot was from a 2-gang version but it's exactly the same for the 1-gang as well.

2) The link I posted to the 1-gang device request was for @SwoopX - to be used as context and comparison (in general, not only for the "Meter interface") between the 1-gang and 2-gang versions as there are very little differences between them and, from this point of view, I think adding them to the existing code base might not be as trivial or easy. I look forward to his idea and implementation.

On the wired aspect, all "Aqara D1 wall switches" (apart from the "Aqara D1 wireless") are wired switches with the "no-neutral" version being an "end device" and the "neutral" version being a "router" device.

esseki, 2021-02-03:
Something else I would like to add about those wired switch is that like many Xiaomi/Aquara wired switch the switch buttons can be decoupled from the relay.

Decoupling buttons seems to be working just fine in the current state.
[image]
[image]

When setting the correct values in the "Xiaomi Diconnect 1 or 2" attributes I can hear that the relay is not triggered anymore when the switch button is pressed.

Thought it was worth to be mentioned to make sure that this feature will not be forgotten or disappear when you guys will be working on adding support. As this is a feature that can be useful to some.

neuroverflow, 2021-02-04:
> Something else I would like to add about this wired switch is that like many Xiaomi/Aquara wired switches the switch buttons can be decoupled from the relay.
> When setting the correct values in the "Xiaomi Diconnect 1 or 2" attributes I can hear that the relay is not triggered anymore when the switch button is pressed.
> 
> Thought it was worth being mentioned to make sure that this feature will not be forgotten or disappear when you guys will be working on adding support. As this is a feature that can be useful to some.

This would be amazing Esseki! Thank you for raising this point as it would solve many of my problems if the D1 become compatible and I could decouple some buttons from the relay ...

esseki, 2021-02-04:
@neuroverflow : the good news is that the decoupling already works with the current state of Deconz. You can test it. There is nothing to add. Did just mention it in this thread to make sure it won't be removed or forgotten.

Actually controlling the relays already works and decoupling the button too. The big missing part with those switch is the fact that no ZHAswitch resource is created when you add the switch to your network (you can have a look at the ticket that i opened #4316) making the decoupling pretty useless at this point.

neuroverflow, 2021-02-04:
I noticed that, I guess ... I'm using HA (2021.2.0) with deconz plugin (6.6.4) and I cannot detect the D1 in Phoscon, I tried as a light or as a switch. But later on when I go in deConz I see new entities. By the way I'm talking about the D1 1-gang no neutral wire.
I suppose it's the non-technical description of this issue :) For now I keep it installed in my office and wait to see compatibility is improving before ordering them for the entire apartment.

neuroverflow, 2021-02-10:
Hi! Any news regarding this kind of devices integration? I see that this new product is getting more and more populate and is apparently supported by zigbee2mqtt. Thank you for your work!

Chreece, 2021-02-28:
Can we please have an update of the request? How is it going?

Mimiix, 2021-03-01:
@SwoopX What do we need to make this happen?

olunia131, 2021-03-03:
Would be great to get it properly integrated. Got 2 of these and for now unable to use with Homeseer (deConz--JowiHue--Homeseer).

esseki, 2021-03-22:
Hi, any update about the support for this switch @SwoopX ?
There seem to be quite a bit of demand from our community

SwoopX, 2021-03-27:
So, what needs to be done now for the device? From the code, i'd assume everything's created as expected.

esseki, 2021-04-05:
Hi @SwoopX ,

I took my time to test carefully before replying.
The main issue with this wired switch has not been fixed yet unfortunately. As I reported it initially in this post : #4316 

I have tested with the latest stable release : Deconz 2.10.04 / 14/03/2021 - Conbee II with firmware 26680700.

Let me try to explain : when adding this switch into the Zigbee network, 2 new light resources are created in the API, they show as smart plug in Phoscon. These are the 2 relays from the switch as you can see in @isorin 's screenshot here : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4199#issuecomment-770812269 Everything is fine here.

The problem is that on this same screenshot you can see 3 switches clusters (1 for the left button, 1 for the right button, the third one is for when the two buttons are clicked at the same time) but there is no corresponding sensor created for these 3 buttons. In the API when calling the GET /sensors/ I can not find the resource corresponding to those 3 buttons. 
Basically this switch works like this one that is already supported by Deconz : QBKG12LM. Except that it supports  "click", "hold" and "release".

Finally this switch support the decoupled mode as I explained here : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4199#issuecomment-772776194
But this seems to be working fine.  

So really the only issue is about the sensor resource in the API.

Thank you in advance

SwoopX, 2021-04-05:
@esseki Ah, ok. That definitively helps, thanks. I'm busy the rest of the week but I can then follow up. Would you be able to compile an updated version of the plugin to test once ready?

esseki, 2021-04-06:
@SwoopX yes i have a testing setup. I never compiled a version myself but i should find a way how to do it. Tell me when you have something to test. Thank you again.

esseki, 2021-04-17:
Hi @SwoopX do you have any update that I can help with test ?

SwoopX, 2021-04-17:
@esseki The instructions how to compile are described in the readme of this repo. However, you'd need to compile my version, so step 1 for you would be

`git clone --branch=xiaomid1 https://github.com/swoopx/deconz-rest-plugin.git`

The code should now allow to have a switch created and enable us to get the relevant data from the button presses. I'd require a debug log from deconz (info + info_L2) where you push all buttons in all combinations once you have created the switch + the Xiaomi special attribute reporting. Something around 15 mins should collect everything necessary to continue.

esseki, 2021-04-18:
Thanks a lot will get this done soon and share the result.

esseki, 2021-04-18:
Hi @SwoopX 

here is the log file that you asked for.
Hopefully you will find in there all the information you need. This is the first time that I'm doing this exercise, so in case you miss anything juts let me know and I will redo it.

Here are some info about what this log file contains : 

- i made it from a fresh test setup with only two nodes : a Conbee II and the Aqara D1 switch 
- you should not find in this log file the moment when I have added the D1 switch into the network. Deconz has been restarted to capture the logs after the D1 switch had been added to the network.
- i have pressed the buttons from the switch more or less in this order : **button left** : 1 click then 2 clicks then 3 clicks (I don't think this one is supported) then long click then release ; **button right** : 1 click then 2 clicks then 3 clicks (I don't think this one is supported) then long click then release ; **both buttons (left + right)** : 1 click then 2 clicks then 3 clicks (I don't think this one is supported) then long click then release. While doing so I might have tried several times the same action in case I was under the impression that I failed it (mainly in case of double ou triple clicks)
- i have a little doubt if the clicks can be properly captured (mainly simple clicks) since this switch can work in decoupled mode (see my previous comment about this https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4199#issuecomment-772776194) but I did not decouple it before capturing the logs. I would still expect an event to be sent by the switch even if it was connected to the relay in case of single click but if you don't find those events it might mean that I should decouple the relays ans start again the capture.

Let me know if you need anything else, I would be happy to help.

Thank you again for your help

[debug5.txt.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6331355/debug5.txt.zip)

SwoopX, 2021-04-18:
Thanks. I've taken already a short glimpse and it looks like everything's in there. Will have a closer look later and make amendments as necessary.

SwoopX, 2021-04-18:
So, data looks a bit weird but sometimes that is related to Xiaomi's interesting interpretation of the standards. However, I guess we can already go into finetuning.

You'd need to find the file `button_maps.json` on your system and add the following lines to get some button events emitted:

```
        "xiaomiSwitchMap": {
            "vendor": "Xiaomi",
            "doc": "Xiaomi Aqara D1 2-gang QBKG22LM",
            "modelids": ["lumi.switch.b2lacn02"],
            "map": [
                [1, "0x04", "ONOFF", "ATTRIBUTE_REPORT", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x04", "ONOFF", "ATTRIBUTE_REPORT", "1", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x04", "ONOFF", "ATTRIBUTE_REPORT", "2", "S_BUTTON_1", "S_BUTTON_ACTION_TREBLE_PRESS", "Triple press"],
                [1, "0x05", "ONOFF", "ATTRIBUTE_REPORT", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x05", "ONOFF", "ATTRIBUTE_REPORT", "1", "S_BUTTON_2", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x05", "ONOFF", "ATTRIBUTE_REPORT", "2", "S_BUTTON_2", "S_BUTTON_ACTION_TREBLE_PRESS", "Triple press"],
                [1, "0x06", "ONOFF", "ATTRIBUTE_REPORT", "0", "S_BUTTON_3", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x06", "ONOFF", "ATTRIBUTE_REPORT", "1", "S_BUTTON_3", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x06", "ONOFF", "ATTRIBUTE_REPORT", "2", "S_BUTTON_3", "S_BUTTON_ACTION_TREBLE_PRESS", "Triple press"]
            ]
        },
```

I'm not too sure if that's a 100% adequate, but you can amend the last 2 values as required to get it right. Verify that with line beginning with `[INFO] - ` in the debug log. Once this is sorted/confirmed, it should be ready to go.

esseki, 2021-04-19:
Thanks a lot for your quick response.
I will test this tonight and report back.

SwoopX, 2021-04-19:
Thanks for having it checked and amended! There's actually one additional thing you could lend me a helping hand, which is confirmation on the Xiaomi special attribute reporting.

Every 5-6 minutes, the device sends the following:

```
15:08:42:558 0x00158D0005472AA2 extract Xiaomi special attribute 0xFF01
15:08:42:561 	64 on/off 0
15:08:42:562 	65 on/off 0
15:08:42:563 	6e unknown 0 (0x00)
15:08:42:564 	6f unknown 0 (0x00)
15:08:42:565 	01 battery 3300 (0x0CE4)
15:08:42:567 	03 Device temperature 16 °C
15:08:42:568 	05 RSSI dB (?) 42 (0x002A)
15:08:42:569 	08 unknown 13852 (0x361C)
15:08:42:570 	0a Parent NWK 0 (0x0000)
15:08:42:571 	99 unknown 0 (0x00000000)
15:08:42:572 	9b unknown 0 (0x0000)
```

I'd assume both on/off attributes (64 and 65) represent the state of the left/right switch. If you could help me in finding out which endpoint belongs to which attribute by pressing a turning a side to on and wait for the corresponding report to come in, I could add that to the code as well. 👍

esseki, 2021-04-20:
Hey no problem will have a look at it tonight.

esseki, 2021-04-20:
Hi @SwoopX ,

so I have tested tonight and I confirm that both on/off attributes (64 and 65) represent the state of the left/right switch : 
- 64 : left button
- 65 : right button
- 0 : switch is off
- 1 : switch is on

Let me know if you need anything else.

SwoopX, 2021-04-21:
Great. Then I'd just need to know which light endpoint (02 or 03) is left/right.

SwoopX, 2021-04-21:
Ah, seems like this is not even required... So all required changes (as far as I can tell) are now in the branch. Feel free to give it another test.

closed 2021-04-25 by manup

7leman7, 2021-08-17:
Just added Aqara D1 double-rocker no neutral and can see it in deConz, but no Xiaomi Disconnect 1 / 2 parameters seen to switch on decoupled mode. What am I doing wrong here?

[image]