dresden-elektronik/deconz-rest-plugin #4201: ZigBee Security: Change Network Key

bcutter, 2021-01-17:
## Describe the question or issue you are having
 While "permit join" as one major security setting is disabled per default, another way to improve security is to change the network key. I found several keys in deCONZ (Network Key, PAN ID TC Link Key).

1) Which value is the network key? (assume: network key)
2) How is the default network key used on a new device created? Randomly, based on MAC address, ...?
3 Steps to change? Assume:
   - 1) Enter new key,
   - 2) Save,
   - 3) Leave/Rejoin network,
   - 4) Repair devices (if already paired before, that's why network key change should be performed before)
   - 5) Done?

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

Smanar, 2021-01-17:
It's called "network key".
The TC link key (trust center) is always the same.

I don't see security problem if someone else know your network key. what is afraid you ? You think someone will stole your key, come at less than 5m from your house, to turn on the light when you are sleeping ?

ebaauw, 2021-01-17:
Afaik the coordinator creates the network key when it creates a new Zigbee network.  I think this is done in de RaspBee/ConBee firmware, but might also be in the deCONZ core.  Both are closed source, so I cannot check.

In my experience, the network key and pan id are chosen at random.  The extended pan id is the mac address of the coordinator.  This is the same for all Zigbee gateways I've seen.

Given the right amount of resources, no consumer-grade Zigbee network is fully safe from attacks.  There's plenty of proof-of-concept hacks described on the Internet, using drones with specialised radio hardware for extended range.  I (naively?) think the potential gain some random hacker hopes to get from hacking my network won't justify the resources needed to perform such a hack, but governance-backed or criminal organisations will be able to compromise your Zigbee network if they wanted to.

The payload of Zigbee messages is encrypted by the network key.  The network key exchanged with a device on pairing.  This exchange is encrypted by the TC Link key, `ZigbeeAlliance09` for ZHA.  ZLL uses a secret key, which, of course, can be found on the Internet.  If one of the routers on your network is willing to facilitate pairing new devices, an attacker could obtain the network key simply by pairing a device.  In fact, this is what we do to sniff the traffic of a new gateway.

Then there's touchlink.  This traffic is encrypted by a secret key, which again, of course, can be found on the Internet.  An attacker can use it to "steal" a device from the network, get it to join another network, and get it to upgrade its firmware to a rogue version providing a backdoor (e.g. always accepting pair requests).  When the user resets and repairs the light, the attacker now has access to your network.  As a more recent security measure, Hue lights only accept touchlink requests within 30 minutes of powering on.  Of course if the attacker can arrange a power outage, that won't help much.

I understand professional Zigbee networks don't allow pairing nor touchlink at all.  Instead each device needs to be commissioned, manually storing the network key on the device.  Of course, this procedure is way too complicated for consumers.  As always, security is a trade-off with usability.

I'm not sure if the Zigbee standard allows for changing the network key without re-pairing all devices.  Even if it does, I'm pretty sure, no consumer-grade device has implemented that part of the standard.

bcutter, 2021-01-17:
Thank you very much @ebaauw. You covered almost all current attack scenarios I came over doing internet research.

Touchlink is not supported by deCONZ powered adapters by the way.

The article my initial question was based on is https://www.zigbee2mqtt.io/how_tos/how_to_secure_network.html, asking myself how to achieve this on the ConBee 2.

Even ZigBee 3.0 standard seems to improve security (e. g. by pairing with a physical secret like QR codes), overall security is not that great. **Summary:** in case someone wants to get in, it's just a matter of resources, but it's possible in almost every condition.

So assuming the network key is generated on a random basis (indications exists, proof/_confirmation by Dresden Elektronik officials_ don't until now), I think there's not much security added to my ZigBee network by manually changing the network key.

ebaauw, 2021-01-17:
> So assuming the network key is generated on a random basis (indications exists, proof/confirmation by Dresden Elektronik officials don't until now)

@manup, can you comfirm, please?

> I think there's not much security added to my ZigBee network by manually changing the network key.

My conclusion as well.  Not using the default z2m key on the other hand seems a no-brainer.  Of course, you'd better address this before pairing your devices...

bcutter, 2021-01-20:
A short confirmation would be great. Thank you.

manup, 2021-01-21:
>@manup, can you comfirm, please?

That's correct a random network key is generated at first boot of the firmware and stored in NVRAM.

Further a random key is generated when the configuration is reset via REST-API in the `resetConfiguration()` method.
Although I'm not happy with the code here, a better implementation with proper random number generator is on the todo list.

closed 2021-03-05 by github-actions[bot]

pfefferCode, 2021-12-25:
> That's correct a random network key is generated at first boot of the firmware and stored in NVRAM.
>
> Further a random key is generated when the configuration is reset via REST-API in the [...]

In deconz/edit network settings/ there is a field "network key". It shows 0x01030507090b0d... which looks not to be any kind of random.
See also the manual: https://www.dresden-elektronik.de/funk/software/deconz.html?file=files/dresden-elektronik/content/downloads/dokumente/funktechnik/deCONZ-BHB-en.pdf
[image]

a) Is this key used? 
b) If so: How can I change it? - If I enter a new key and press "save", it seems not to have any effect: I still can switch the lights which were connected before changing the key. This even holds after a complete reboot of the raspberry pi.

I am using Homeassistant with the deCONZ-addon.

Smanar, 2021-12-25:
You disconnect before changing the key ?
Perhaps a message in logs ?

pfefferCode, 2021-12-25:
no, I did not disconnect before changing the key.
Disconnecting, changing the key an joining again, renders the devices out of work.
Disconnecting again, changing the key back, makes the devices reachable again. So this works :-)
Thanks for the hint!

Then, 
(1) The information that the key is generated randomly is not true - at least for the HomeAssistant-add-on.
(2) Please add an error message when trying to change the network key while connected

Smanar, 2021-12-25:
>Please add an error message when trying to change the network key while connected

It s the normal procedure, for all settings, like to change the channel.

>The information that the key is generated randomly is not true - at least for the HomeAssistant-add-on.

IDK what you want to say, but this key is random, else we would all have the same ?

pfefferCode, 2021-12-25:
> It s the normal procedure, for all settings, like to change the channel [...]
yes, for all settings in the specific dialog. Just adding an information box which appears after clicking on "save" would be very helpful.

> > The information that the key is generated randomly is not true - at least for the HomeAssistant-add-on.
> IDK what you want to say, but this key is random, else we would all have the same ?
You do not want to tell me that the key "1 3 5 7 9 11 13 15 00 2 4 6 8 10 12 14" is a random key? Do you have a different one? Maybe that is an issue specifically on HomeAssistant on RaspberryPI?

Smanar, 2021-12-26:
You have not exaclty the same ^^
And no, mine is 0x33616623256636383237303330316565, so totally different.

pfefferCode, 2021-12-26:
Thank you for
(1) opening a feature request in the forum
(2) the explanation regarding the network key. In fact, yes, I tried z2m in between. So I guess, z2m changed the network key to their (strange) default, and after switching back, deCONZ read it from the conbee II stick (instead of using the old one stored somewhere on the computer). That seems to me to be a plausible explanation for the phenomenon that I did not have a random network key. I am sorry that I bothered you.

Smanar, 2021-12-27:
Lol, np ^^