dresden-elektronik/deconz-rest-plugin #4203: Ubisys S1-R Metering functions

tgoerlich, 2021-01-17:
## Describe the question or issue you are having

Hi all,
I successfully conneted the Ubisys S1-R rail switch, however I seem to miss the metering functions that come with it. I cannot see any sensors in the phoscon web app, also not in other connected applications („Hue Essentials“ iOS App, iobroker deconz adapter).
In the deconz application itself I see the sensors, so they must be somewhere! Am I missing something? Can anyone confirm if the metering functions of the S1-R are implemented fully?

## Screenshots

[image]

## Environment

Host system: Raspberry Pi
Running method: Raspbian
Firmware version: 26520700
deCONZ version: 2.05.74/25.2.2020
Device: RaspBee II
Do you use an USB extension cable: no
Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Thanks

tgoerlich, 2021-01-17:
Additional info:
In Deconz, I read power measurement values that look plausible. So, the more precise question is: how can the metering sensor be made visible through the API?
Thanks, Thomas

ebaauw, 2021-01-17:
Should already be exposed as a ZHAConsumption `/sensors` resource for cluster 0x0702 and a ZHAPower `/sensors` resource for 0x0B04.

tgoerlich, 2021-01-17:
There is no ZHAConsumption or ZHAPower in my `/sensors` list.
According to

tgoerlich, 2021-01-17:
Ok, Problem solved. It seems that the S1-R is only recognised as a sensor when there is current load during the pairing process. So, switch on a radiator, go downstairs, press reset for 10secs, pair again... and here we are. I have no idea why this is so and if this is a bug or a feature. This behaviour is not mentioned in the documentation, however.

closed 2021-01-17 by tgoerlich