dresden-elektronik/deconz-rest-plugin #4206: Ubisys S1-R sensor functions not shown in REST api under certain conditions

tgoerlich, 2021-01-17:
## Describe the bug

Hi,
See also #4203 for detailed description.

## Steps to reproduce the behavior

- Install Ubisys S1-R, connect it to power, but do not connect consumer load to it.
- Pair it with deconz
- the device is visible as “light” in the API, but not as sensor
- sensor is visible in DeConz, correct power usage is shown
- Connect a load to the S1-R
- Reset the S1-R (press reset button for 10 sec) and pair again
- Sensors now visible in the REST API 

## Expected behavior

Should be shown in the api regardless of load during pairing process

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26520700
 - deCONZ version: 2.05.74/25.2.2020
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Thx

ebaauw, 2021-01-17:
Did endpoint 0x04 and the 0x0702 and 0x0B04 cluster show in the GUI when pairing the first time (when the resources weren't created)?  If deCONZ hasn't read the _Simple Descriptor_ for that endpoint, the `/sensors` resources wouldn't be created.  Generating load probably causes the S1-R to send a notification, which causes deCONZ to add the endpoint and cluster anyways.

Did the API create the ZHASwitch resource for combined endpoints 0x02 and 0x03.

tgoerlich, 2021-01-17:
Sorry i cannot check that anymore. But i can tell that I was able to see actual power values within the deconz GUI without seeing the Sensor in the API.

closed 2021-03-05 by github-actions[bot]