dresden-elektronik/deconz-rest-plugin #4207: RaspBee II Firmware upgrade fails with RC=3

gpoul, 2021-01-17:
Firmware upgrade failing with RC=3; I followed the installation instructions as listed on the homepage and in the docker deCONZ documentation; the RaspBee II works with deCONZ to control the lights via the "unofficial" docker image, but the firmware upgrade fails in the docker container implementation and without docker as well; provided debug logs below without docker as that will be easier to debug and reproduce. - There is nothing else connected to the Raspberry Pi apart from power.

## Screenshots
```
pi@raspberrypi:~ $ sudo GCFFlasher_internal -d /dev/ttyAMA0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF -x 3
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
16:23:27:823 using firmware file: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF 
16:23:27:904 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
16:23:27:904 dev /dev/ttyAMA0
16:23:27:904 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
Reboot device /dev/ttyAMA0 (RaspBee)
16:23:27:905 query bootloader v1 ID after 0 ms
16:23:29:407 query bootloader v1 ID after 1501 ms
16:23:29:907 query deCONZ firmware version
16:23:31:910 uart reset failed, try RaspBee reset
action: reset device RaspBee
wiringPi 2.52 initialized
16:23:32:114 query bootloader v1 ID after 0 ms
16:23:32:601 RX 14 bytes ASCII ??? after 487 ms
16:23:32:601 RX c0818203000300780f9f765bfdc0 after 487 ms
16:23:32:601 SLIP RX frame length: 10
16:23:32:601 check magic: 0x81 cmd: 0x82 after 487 ms
Bootloader version 0x00030003, app crc: 0x769F0F78
16:23:32:602 GCF_ResetDeviceDone
16:23:32:604 bootloader v3 update firmware request size: 164106, addr: 0x00005000, type: 1, crc-32: 0xAAAAAAAA
16:23:32:604 TX c081030a8102000050000001aaaaaaaaf6fbc0
16:23:32:612 SLIP RX frame length: 3
pi@raspberrypi:~ $ echo $?
3
pi@raspberrypi:~ $ sudo GCFFlasher_internal -l
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
```

## Environment
```
pi@raspberrypi:~ $ lsb_release  -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
pi@raspberrypi:~ $ cat /etc/debian_version 
10.7
```

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian
 - Firmware version: 26610700
 - deCONZ version: 2.07.01-raspbian-buster-stable
 - Device: RaspBee II

SwoopX, 2021-01-17:
You try to flash the wrong firnware

Mimiix, 2021-01-18:
Yep. Agreeing with Swoop. You are flashing the ConbeeII firmware instead of the RaspbeeII firmware.

gpoul, 2021-01-18:
Let me retest the flashing on the weekend; that totally makes sense and I missed it as I'm still confused by the different names of these two products although they're pretty well named ;-)

what confuses me when I followed the guides is that the log I get clearly states I should use what I tried to flash...

```
15:36:11:759 Device firmware version 0x26610700 RaspBee II
15:36:13:633 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
15:36:13:633 GW firmware version: 0x26610700
```

Is that ok or is there some misconfiguration on my end? I thought I needed to flash what the logs print out. :-/

Mimiix, 2021-01-18:
The log said it _found_ firmware. Not what you _needed_ to flash. It isn't pretty tho.

Mimiix, 2021-01-18:
I've forwarded this to Manuel, let's see if it's a bug and then you just caught us one ^^

manup, 2021-01-18:
Good catch, I've checked, the test for device selection fails here.
Will be fixed soon.

gpoul, 2021-01-24:
I now tried flashing a RaspBeeII firmware file, but that wasn't more successful it seems. Any ideas on what might be off? Again, I also tested this with the docker container as well.

```
pi@raspberrypi:~ $ sudo GCFFlasher_internal -d /dev/ttyAMA0 -f /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF -x 3
GCFFlasher V3_17 (c) dresden elektronik ingenieurtechnik gmbh
17:41:04:710 using firmware file: /usr/share/deCONZ/firmware/deCONZ_RaspBeeII_0x26680700.bin.GCF 
17:41:04:793 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
17:41:04:794 dev /dev/ttyAMA0
17:41:04:794 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
Reboot device /dev/ttyAMA0 (RaspBee)
17:41:04:796 query bootloader v1 ID after 0 ms
17:41:06:298 query bootloader v1 ID after 1501 ms
17:41:06:799 query deCONZ firmware version
17:41:06:807 SLIP RX frame length: 9
deCONZ firmware version 26610700
17:41:06:807 send watchdog reset 2 seconds
17:41:06:807 TX c00b02000c0005002602000000baffc0
17:41:06:816 set watchdog ttl status: 0x00
17:41:06:819 wait reboot: 2100 ms
17:41:08:921 query bootloader v1 ID after 0 ms
17:41:10:970 query bootloader v1 ID after 2050 ms
17:41:12:472 query bootloader v3 ID after 3551 ms
17:41:12:472 TX c081027dffc0
17:41:12:488 RX 28 bytes ASCII ??? after 3568 ms
17:41:12:488 RX c0818203000300780f9f765bfdc0c0818203000300780f9f765bfdc0 after 3568 ms
17:41:12:489 SLIP RX frame length: 10
17:41:12:489 check magic: 0x81 cmd: 0x82 after 3568 ms
Bootloader version 0x00030003, app crc: 0x769F0F78
17:41:12:490 GCF_ResetDeviceDone
17:41:12:491 bootloader v3 update firmware request size: 164373, addr: 0x00005000, type: 31, crc-32: 0xAAAAAAAA
17:41:12:491 TX c0810315820200005000001faaaaaaaaccfbc0
17:41:12:499 SLIP RX frame length: 10
pi@raspberrypi:~ $ echo $?
3
```

ahoenle, 2021-02-08:
Hey @gpoul I can confirm your issue. I have a RaspBee II with FW version 0x26610700 and the output I got was identical to what you already reported.

I downloaded all available RaspBee II firmwares from http://deconz.dresden-elektronik.de/deconz-firmware and just tried one after the other.
For unknown reasons, after a couple of failed attempts, I was able to flash version 0x26680700.

```
08:43:21:358 dev /dev/ttyAMA0
08:43:21:359 GW firmware version: 0x26680700
```

Afterwards I tried the latest version 0x26690700 and ran into the old issues. Again, after a couple of retries it worked and I'm now on  0x26690700.

```
08:52:33:359 dev /dev/ttyAMA0
08:52:33:360 GW firmware version: 0x26690700
```

What I found very confusing is that `GCFFlasher_internal -l` reports the device as `RaspBee` and not `RaspBee II`. This made me wonder if I should try flashing RaspBee firmwares instead.

closed 2021-03-09 by github-actions[bot]