dresden-elektronik/deconz-rest-plugin #4218: IKEA Tradfri remote stops working

andersthomas, 2021-01-18:
## Describe the bug

IKEA TRÅDFRI Remote suddenly stopped working. Pressing buttons does not trigger the configured action. The remote is still fully visible in the UI. Disconnected and reconnecting works, but either doesn't fix the problem or makes the problem move to another Tradfri remote.

## Steps to reproduce the behavior

Don't know if/how the problem can be reproduced as it occurred out of the blue within a working setup. Interestingly, though, the problem moved from one Tradfri remote to the other after disconnect/reconnect (pairing) of the first remote.

1. First Tradfri remote stopped working
2. Delete switch in Phoscon app
3. Reconnect first remote to bridge
4. Give remote the same name as before
5. First remote inherited the old settings (configured actions) and starts to work again
6. Immediately, second remote stopped working
7. Redo steps 1-4 for the second remote
8. Second remote inherited the old settings (configured actions), but does NOT start to work again

## Expected behavior

All remotes should work as configured.

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian 10.4
 - Firmware version: 26660700
 - deCONZ version: 2.09.00 / 23.12.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

Please advise on suggested log levels for this type of problem.

## Additional context

- Two IKEA Tradfri remotes
- 12 connected lights
- some more Zibgee hardware purchased, but not yet connected to isolate the problem

Mimiix, 2021-01-18:
Hi! Are you able to provide some logs?

To enable logging. Open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

manup, 2021-01-18:
Which Firmware version is shown for the remotes?
I guess these are the newer ones, iirc bindings are a bit different here as the remote doesn't have the default setup anymore as in older versions. I only have the old one here for testing, will get the new version soon.

andersthomas, 2021-01-18:
My bridge is running in headless mode (" /usr/bin/deCON  -platform minimal --http-port=80") without the deCONZ UI. Am I correct to assume that your debug suggestions translate into appending "--dbg-info=2 --dbg-error=2 --dbg-aps=2" to the command-line? How long should it run with these options? 

The Tradfri firmware is 2.3.014. Is this "old" or "new"? Interestingly, for the non-working remote, the firmware versions is not shown in the Phoscon App.

andersthomas, 2021-01-18:
Attached please find the output from starting deCONZ with the additional debug options, pressing buttons on the non-working remote and shutting deCONZ down again.

[deconz-debug.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5832367/deconz-debug.txt)

Mimiix, 2021-01-18:
Two things:

- stick isn't seen. So that's why it doesn't work
- your database seems corrupt as it had double columns.

andersthomas, 2021-01-18:
What steps do you suggest to fix the database corruption? It has never been touched manually yet.

Mimiix, 2021-01-19:
In your case i wouldn't. I'd say just start from scratch. Why? Because if you don't know what your doing, it is a mess to clean up.

No clue why it happened though.

andersthomas, 2021-01-19:
Why start from scratch? I've switched to an earlier backup and at least the DB corruption message is gone. However, the Phoscon app shows all lights as offline and I cannot control them. Attached please find the debug log. What next?
[deconz-debug-20210119.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5838712/deconz-debug-20210119.txt)

andersthomas, 2021-01-21:
I switched back to the latest backup and I'm back to the situation as originally described in the ticket where just one Tradfri remote is not working. I'm interested to explore manual ways to fix the database corruption if there are useful hints or documentation. I can still start from scratch as a last resort.

Shall I open a separate ticket for the backup/restore issue os is this a known problem?

manup, 2021-01-21:
The database isn't corrupted, the code just goes through the schema migration versions.
I don't see the ConBee as device in the logs (like /dev/ttyACM0).

One thing I'd highly recommend is to update the firmware to the latest version 0x26680700 to mitigate problems with end-device handling.

Unfortunately I still haven't the newer version of the remote here for testing, my old one is on version 1.2.214.

andersthomas, 2021-01-22:
I've updated the deconz package on Raspbian to 2.09.01, but the firmware still is 26660700. Am I supposed to update it manually to 26680700?

andersthomas, 2021-01-22:
I successfully performed the manual update, but the problem persists.

felschr, 2021-02-15:
Have you been able to resolve the issue? I've just updated deconz from `2.05.82` to `2.09.03` and afterwards all my IKEA remotes were disconnected. Then I updated the ConBee II firmware as well to `26680700`.
The remotes still aren't working. I tried adding them again but I can't get past the step to hold the main button for about 10 seconds. It doesn't work and just times out. I've tried it multiple times now.

andersthomas, 2021-02-20:
I've also updated to DeCONZ 2.09.03 and ConBee II firmware 26680700, but the problems persist. I've deleted all key assignments of the remote for the connected light group and created new assignments of the remote for an On/Off switch - and it works for this purpose, so the remote is NOT broken! The original light group now isn't even switchable with the Phoscon app anymore. How to do further troubleshooting?

andersthomas, 2021-02-25:
Could you please provide guidance how to fix/troubleshoot my non-working light group?

andersthomas, 2021-03-07:
I deleted the affected lights and light group via Phoscon App and tried to add them from scratch. Resetted the light, light starts blinking, but the Phoscon App doesn't find it via "Add new light". I've tried multiple times, but no success. Any suggestions how to fix this mess? Is "delete" process through the Phoscon App really a full delete from deCONZ?

Mimiix, 2021-03-08:
It should be. Can you check what the rest api says?

Also, did you reboot once after removing them ?

andersthomas, 2021-03-09:
I rebooted, deleted and reconnected almost everything (including the Tradfri remotes), but exactly one light (Paulmann) still refuses to get connected again even though it's clearly in connection mode after the reset procedure. How to find out what's wrong? What exact REST API call should I check?

andersthomas, 2021-04-02:
The issue still persists. Any suggestion how to further troubleshoot/fix?

andersthomas, 2021-04-25:
The issue is not solved yet. Any hints for troubleshooting still very much appreciated!

Mimiix, 2021-04-30:
Are you on the latest versions? Perhaps you can share some newer logs?

Gorole, 2021-05-21:
Hi, i have same problem (Only IKEA bulbs and the remote )

sysofwan, 2021-06-11:
+1 Having the same issue. Light reconnect does not work either. All tradfri lights and switches stopped working.

closed 2021-07-14 by github-actions[bot]