dresden-elektronik/deconz-rest-plugin #4222: All devices become unavailable every minute.

guilly08, 2021-01-18:
## Describe the bug
All of my devices become unavailable every minute. I've narrowed it down to hardware, however I'm not sure what could be causing the issue since the hardware is fully supported.

**Working Hardware:**
- Toshiba Laptop 8GB of RAM
- O/S : Ubuntu 20.04 LTS (Desktop) and Windows 10

**Not working Hardware:**
- Raspberry Pi 3 Model B Plus Rev 1.3

- Custom NAS (i3 / 8GB / Ubuntu 20.04 LTS)

## Steps to reproduce the behavior
Problem is consistent, however as mentioned seems to be hardware specific. The only reason I'm opening a bug is because the hardware is fully supported for deCONZ.

## Expected behavior
Device should remain connected and not drop in / out every minute. I don't believe this is an HA issue since the devices become unresponsive in Phoscon as well. Also, it's fully functioning on the same HA instance with the working hardware mentioned above.

## Screenshots
[image]

## Environment

**deConz Versions**
FW version: 26660700
Build version: 2.05.81 / 9/14/2020
Product: ConbeeII

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu )
 - Firmware version: (26660700)
 - deCONZ version: (2.05.81)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? If so: no

## deCONZ Logs
[debug2.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5830070/debug2.log)

## Additional context
Very small zigbee network. A couple of Sengle smart bulbs and ikea motion sensor. I currently only have 1 sengled bulb plugged paired. The zigbee hub is on the same floor to rule out any interference as the issue.

The debug log is with deconz running on the pi. If it helps I can provide logs of deconz running on my ubuntu instance as well.

Mimiix, 2021-01-18:
Hi

For some reason, it keeps disconnecting the firmware. Those systems with not working hardware, what OS is that? Is that HassOS or Docker based?

Are you able to provide some APS logs?

Thanks!

guilly08, 2021-01-18:
Hey @Mimiix,

I was troubleshooting this with you on discord a week ago. The OS in the logs is raspbian and I'm running deconz headless.

The logs I provided had aps enabled. See below for switches I ran. I may not of let it run long enough though ?
 
`/usr/bin/deCONZ --platform minimal --dbg-info=2 --dbg-error=2 --dbg-aps=2`

manup, 2021-01-18:
In the problematic cases deCONZ runs nativ, not in an virtual environment (Docker, VM, ...)?

On native installations usually such connection problems indicate USB problems like interference with other periphery, or too weak power supply on Raspberry Pi. For the interference the USB extension cable should help but you already have this.

It might also be worth a try to check if the recent firmware 0x26680700 and deCONZ version helps to mitigate the issue.

guilly08, 2021-01-18:
Thanks,

I'll test this evening with the latest firmware and using my Ubuntu NAS to eliminate the power draw as an issue.

Is the power draw lower when running deCONZ as an Add-On in HA ? Reason I ask is in my previous deployment I was running HA on the pi in question with deCONZ as an Add-ON with zero issues. I upgraded my HA instance to a KVM as I've expanded the # of devices in my household and wanted a more stable env.

guilly08, 2021-01-19:
Here are the latest logs on ubuntu 20.04 LTS using a USB Extension cable. Same behavior so I don't think power is the issue.

[debug_ubuntu.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5832835/debug_ubuntu.log)

Mimiix, 2021-01-19:
This looks way better :)! 

@manup What do you think of the latest log?

manup, 2021-01-19:
The basic RF operation in the log looks fine, requests are send and responses received.

But at least the firmware version looks to be the older one, 0x26660700. Can you please try the version 0x26680700 as suggested. Firmware update is described in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

Also with which deCONZ version was the log made?

guilly08, 2021-01-19:
Here are latest logs with latest firmware and **2.09.01** of deCONZ.

[debug_ubuntu_0x26680700.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5835533/debug_ubuntu_0x26680700.log)

The devices are still dropping in and out... there's also a 2-3 second lag when I turn the device on / off through phoscon and HA.

[image]

Mimiix, 2021-01-19:
@guilly08 for debugging purposes: please stick with phoscon first. After that, expand to HA. Just to make sure the issue isn't with HA.

guilly08, 2021-01-19:
Meaning remove the deCONZ integration from HA ? In any case, I've done this in the pass and phoscon demonstrates the same behavior.

Light action becomes unresponsive for a couple seconds
Sensors drop in and out of status

I can provide more debugs without HA integration if that helps though.

Mimiix, 2021-01-19:
No, the testing you do i mean. Your screenshots are from HA. 

About the delay: Can you show a screenshots of the deCONZ gui? I want to see the lines/LQI.

guilly08, 2021-01-19:
ok no problem,

[image]

Mimiix, 2021-01-19:
The one with the hex name seems to be not paired correctly or is a unsupported device.

That probably is the remote, isn't it?

guilly08, 2021-01-19:
I suspect it's my ikea sensor that I have paired in the passed.... I repaired it, here's the latest GUI.

[image]

Mimiix, 2021-01-19:
So which device is getting unavailable? Is that with _all_ devices? Or just with the light?

guilly08, 2021-01-19:
All devices

manup, 2021-01-19:
Are the devices in the Phoscon App shown as unavailable as well?
The logs look alright, no signs of unavailable state there.

If they are shown as available within the Phoscon App and REST-API, it could be that this is the Websockets push events issue we have seen the other day (the bug fix is already on the todo list).

guilly08, 2021-01-20:
It's difficult to say in phoscon if they become unavailable. I'm inclined to say no, the only thing I've noticed is a lag when turning the device on and the brightness bar updating. Whenever a device is turned on from phoscon there's a 2-3 second delay for the brightness bar to update.

Is there a bug# about the websocket bug ?

guilly08, 2021-01-20:
I decided to try zigbee2mqtt instead of phoscon and things have been stable. So I suspect it could be the websocket issue.

If you'd like me to test further please let me know or else I believe this issue can be closed until the fix is introduced.

manup, 2021-01-20:
The related Websocket issue is https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3707#issuecomment-753947400

I think once this is figured out in upcoming releases, it might be worth a try to see this solves the problems here as well.

guilly08, 2021-01-20:
Great,

I'll keep an eye out and hopefully that fixes it!

closed 2021-01-20 by guilly08