dresden-elektronik/deconz-rest-plugin #4225: Colour Temperature Change not working with  Ikea TRÅDFRI

tactmaster, 2021-01-18:
## Describe the bug

Set up 
Ikea TRÅDFRI bulb E14 opal 400lm or E27 WS opal 1000lm 
and TRÅDFRI remote control
not controlling colour temperature via buttons

## Steps to reproduce the behavior

[image]

< and > buttons don't work. Long press/ short press....repeated pressing
This happen every time I set it up.

## Expected behavior
The colour temperature should get warmer or colder depending on buttons.

## Screenshots
## Environment

 - Host system: Raspberry Pi 3B
 - Running method:  Home Assistent deCONZ Add-on 6.6.2  ( Hassio 2021.01.4)
 - deCONZ version: (2.7.1)
 - Device:   ConBee II 
 - Do you use an USB extension cable:   no
 - Is there any other USB or serial devices connected to the host system? Telldus Tellstick/ BT CSR 4.0

## deCONZ Logs
[image]

Light 
```
        "9": {
            "colorcapabilities": 0,
            "ctmax": 65279,
            "ctmin": 0,
            "etag": "9bcb381dada4fb422d3aa093366ba225",
            "hascolor": true,
            "lastannounced": "2021-01-18T17:24:03Z",
            "lastseen": "2021-01-18T18:07Z",
            "manufacturername": "IKEA of Sweden",
            "modelid": "TRADFRI bulb E14 WS opal 400lm",
            "name": "Side Light",
            "state": {
                "alert": "none",
                "bri": 25,
                "colormode": "ct",
                "ct": 454,
                "on": true,
                "reachable": true
            },
            "swversion": "2.3.050",
            "type": "Color temperature light",
            "uniqueid": "00:0b:57:ff:fe:37:1e:5d-01"
        }

```
## Additional context

tactmaster, 2021-02-24:
Still a problem. I overriden the setting using Hue Essentials app to make it possible to chnage colour temp from a ikea remote.

closed 2021-03-27 by github-actions[bot]

ebaauw, 2021-03-27:
Hm, the Phoson screen says long press, but the log shows short press events.  And the “discard too fast event”, see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4627#issuecomment-808229188.

Are you sure the remote is in the right mode and sending long press events?  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3756#issuecomment-808619291.