dresden-elektronik/deconz-rest-plugin #4232: Deconz trigger 1002 event on smartthings button on version 2.09.01

MartinTerp, 2021-01-19:
## Describe the bug
After change to version 2.09.01, my Smartthings button GP-U999SJVLEGA sometimes sends the 1002 event (according to deconz) causing my lights to turn on/off occasionally, atm it seems like every hour

## Steps to reproduce the behavior
Update to 2.09.01, have a Smartthings button, wait 

## Expected behavior
It dosnt send 1002 event unless button press

## Screenshots
This is the only log from the event time
[image]

## Environment
Version 2.09.01 / 12/23/2020
Firmware 26680700

 - Host system: PC
 - Running method:  Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.09.01 / 12/23/2020
 - Device: ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs
See screenshot

## Additional context

Smanar, 2021-01-19:
Hello, you haven't the log when it send the command ? There is written "[INFO] - Button 1002 ...."

Because, your log is about cluster 0x0500, not used for buttonevent.

MartinTerp, 2021-01-20:
Hi

i have enabled the following debug info:
```
      - "DEBUG_INFO=1"
      - "DEBUG_APS=1"
      - "DEBUG_ZCL=1"
      - "DEBUG_ZDP=1"
      - "DEBUG_OTAU=1"
```

And the only log I got from that time where I can see the event trigger:

```
16:29:16:810 [INFO] - No button handler for: button endpoint: 0x01 cluster: IAS_ZONE (0x0500) command: 0x0A payload[0]: 002
16:29:16:810 ZCL attribute report 0x286D9700010827B9 for cluster: 0x0500, ep: 0x01, frame control: 0x08, mfcode: 0x0000
16:29:16:811 Write poll cluster check-in interval for 0x286d9700010827b9
16:29:16:811 write attribute of 0x286D9700010827B9 ep: 0x01 cluster: 0x0020: 0x0000
```
Not sure that the first line is related

MartinTerp, 2021-01-20:
Hi

Here is what happend that second.

```
17:29:53:143 ZDP skip fetch 0x86BD7FFFE503FA5, diff last seen: 626 seconds [4]
17:29:53:303 ZDP skip fetch 0xD0CF5EFFFE6EC90B, diff last seen: 626 seconds [4]
17:29:53:305 Mgmt_Lqi_req zdpSeq: 193 to 0x000B3CFFFEF77E25 start index 15
17:29:53:305 APS-DATA.request id: 234, addrmode: 0x03, addr: 0x000b3cfffef77e25, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
17:29:53:327 APS-DATA.confirm id: 234, status: 0x00 SUCCESS
17:29:53:328 APS-DATA.confirm request id: 234 -> confirmed, timeout 1611160193
17:29:53:344 APS-DATA.indication srcAddr: 0xf083, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 255, rssi: -38
17:29:53:345 APS-DATA.indication request id: 234 -> finished
17:29:53:346 APS-DATA.request id: 234 erase from queue
17:29:53:346 ZDP status = 0x00 -> SUCCESS
17:29:53:347 ZDP Mgmt_Lqi_rsp zdpSeq: 193 from 0x000B3CFFFEF77E25 total: 16, startIndex: 15, listCount: 1
17:29:53:347 * neighbor: 0x086BD7FFFE05F7F4 (0xEDAF), LQI: 241, relation: 0x02 rxOnWHenIdle: 1
17:29:53:562 APS-DATA.indication srcAddr: 0xafd3, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -55
17:29:53:564 APS-DATA.request id: 238, addrmode: 0x02, addr: 0xafd3, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
17:29:53:565 [INFO] - No button handler for: button endpoint: 0x01 cluster: IAS_ZONE (0x0500) command: 0x0A payload[0]: 002
17:29:53:566 ZCL attribute report 0x286D9700010827B9 for cluster: 0x0500, ep: 0x01, frame control: 0x08, mfcode: 0x0000
17:29:53:566 APS-DATA.request id: 239, addrmode: 0x02, addr: 0xafd3, profile: 0x0104, cluster: 0x0500, ep: 0x01 -> 0x01 queue: 2 len: 5 tx.options 0x00
17:29:53:598 APS-DATA.confirm id: 238, status: 0x00 SUCCESS
17:29:53:599 APS-DATA.confirm request id: 238 -> erase from queue
17:29:53:602 APS-DATA.confirm id: 239, status: 0x00 SUCCESS
17:29:53:602 APS-DATA.confirm request id: 239 -> erase from queue
17:29:53:623 aps request id: 238 finished, erase from queue
17:29:53:703 aps request id: 239 finished, erase from queue
```

Smanar, 2021-01-20:
Do you know how to log websocket request ? Using your third app or a plugin browser ?

To be sure it s deconz that send the notification.
Because I m sorry, but on your log again, deconz haven't try to send websocket request to change a buttonstate.

alaghbar, 2021-01-22:
I have the same issue after latest docker container update. Routinely triggers event 1002 and on another button, event 1004 without physically pressing any button

MartinTerp, 2021-01-26:
> Do you know how to log websocket request ? Using your third app or a plugin browser ?
> 
> To be sure it s deconz that send the notification.
> Because I m sorry, but on your log again, deconz haven't try to send websocket request to change a buttonstate.

Is the event from NodeRed enough? else provide info on webrequest log.

[image]

Smanar, 2021-01-26:
Nope it s fine.
A last test, you can done, just to be sure the websocket is send for the buttonevent change and not another thing.

You problem concern only the 1002 value ? so can you change this value, using the remote (idk wich one feature it support), and check the value during the bug.

Just to check if it s a new event 1002, or just a repeat of the last one. (the value is memorised in the API till it change, you have always the last value visible)

SwoopX, 2021-01-26:
You might want to try something out here. In deconz GUI, select the IAS Zone cluster, scroll down to the attributes and double click on the value for attribute 0x0002. Make sure the device is awake by pushing the button and then read the reporting configuration (must be done within 2 seconds). That should give you a `NOT_FOUND` as below.

The code has a passage to configure 3600 as max interval, which would correspond with your experience. If anything is configured, try disabling the reporting configuration by setting min interval to 0 and max interval to 65535, then press write config (again, make sure the device is awake).

[image]

kyletpugh, 2021-01-26:
It appears that this device sends a battery or status indication message on a regular interval. This is just sent as the previous event that was triggered for some reason. If no event has been triggered yet then you get this (I removed the ids but they were all unique):
```
{
    "event_type": "deconz_event",
    "data": {
        "id": "removed_id_1",
        "unique_id": "removed_unique_id_1",
        "event": null,
        "device_id": "device_id_1"
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-26T19:29:02.952226+00:00",
    "context": {
        "id": "removed_device_id_2",
        "parent_id": null,
        "user_id": null
    }
}
```
This message appears to be sent about every hour.
```

MartinTerp, 2021-01-27:
> Nope it s fine, thx.
> A last test, you can done, just to be sure the websocket is send for the buttonevent change and not another thing.
> 
> You problem concern only the 1002 value ? so can you change this value, using the remote (idk wich one feature it support), and check the value during the bug.
> 
> Just to check if it s a new event 1002, or just a repeat of the last one. (the value is memorised in the API till it change, you have always the last value visible)
> 
> And no problem before the 2.09.01 version ?

Hi, yes its happend like, 5min after I patched to 2.09.01, and I blamed my girlfriend for playing with the lights :D 
I pushed the button, read the details, waited until it send a event itself, and its not the same event as the manual on (etag is different) and yes, its only 1002 it sends

MartinTerp, 2021-01-27:
> You might want to try something out here. In deconz GUI, select the IAS Zone cluster, scroll down to the attributes and double click on the value for attribute 0x0002. Make sure the device is awake [...]

> You might want to try something out here. In deconz GUI, select the IAS Zone cluster, scroll down to the attributes and double click on the value for attribute 0x0002. Make sure the device is awake [...]

Hi

I dont have switches in my node list, but the switch also has a temperature sensor which I can see in the gui, can I use that or am i missing something regarding switches in node list?

alaghbar, 2021-01-27:
> Nope it s fine, thx.
> A last test, you can done, just to be sure the websocket is send for the buttonevent change and not another thing.
>
> You problem concern only the 1002 value ? so can you change [...]

It also started here few minutes after it was updated to 2.09.01. It auto generates 1002 and sometimes 1004 events without pushing a button. I have 2 of these buttons and it happens for both. I had to disable all automations related to SmartThings buttons. Thanks

Smanar, 2021-01-27:
I m checking the change in version beetween v2.8.0-beta and v2.9.1-beta
I just see :
- A new reporting setting  in binding.cpp for "Samjin" device https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ba05ea1ab65a0cfa1da01e417fb3a11c850c73ae
- A special part to "process only once" Samjin device with the model id = multi https://github.com/dresden-elektronik/deconz-rest-plugin/commit/ba05ea1ab65a0cfa1da01e417fb3a11c850c73ae
- A big part on IAS, but idk the impact the changes can have

kyletpugh, 2021-02-12:
So the changes in the beta channel did not prevent this. It seems that there are two events being sent by these buttons that end up firing a deconz_event with the previous event data, a periodic (looks like hourly) temperature update, and what I think is a device is going to sleep event five minutes after last being pressed. Both of these appear to be re-firing the previous real event, if there was one, when they should only be updating the state of the device. I will be able to gather deconz logs soon, I hope.

kyletpugh, 2021-02-13:
Just so the information is here. See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4227#issuecomment-778503973 for more information. I will also get logs for you when I am able to access the server directly. This is on beta-channel, version 2.09.03 / 12/23/2020.

kyletpugh, 2021-02-24:
Attached the log. The event is fired when these temperature reads appear:
`06:31:36:192 Force read attributes for ZHATemperature SensorNode Kyle's Room Desk Temperature`
`06:31:41:193 Force read attributes for ZHASwitch SensorNode Dining Room Button`
But this is also firing the event (this is the same device as the temperature sensor from above) caused by motion(?):
```06:31:31:193 Force read attributes for ZHASwitch SensorNode Kyle's Room Desk Button
06:31:31:199 APS-DATA.indication from child 0xD540
06:31:31:200 Node data 0xb0ce181403374589 profileId: 0x0104, clusterId: 0x0008
06:31:31:200 0xB0CE181403374589: update ZCL value 0x01/0x0008/0x0000 after 0 s
06:31:31:201 ZCL attribute report 0xB0CE181403374589 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
06:31:31:201 	payload: 00002042
06:31:31:541 Poll APS request to 0xB0CE181403396234 cluster: 0x0006 dropped, values are fresh enough
06:31:31:776 Node data 0x286d970001136cb2 profileId: 0x0104, clusterId: 0x0402
06:31:31:777 0x286D970001136CB2: update ZCL value 0x01/0x0402/0x0000 after 0 s
06:31:31:777 enqueue event state/temperature for /sensors/12
06:31:31:777 enqueue event state/lastupdated for /sensors/12
06:31:31:778 [INFO] - No button map for: motion, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0402, command: 0x0A, payload: 0000296C07, zclSeq: 57
06:31:31:778 ZCL attribute report 0x286D970001136CB2 for cluster: 0x0402, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
06:31:31:778 	payload: 0000296c07
06:31:31:779 0x286D970001136CB2 (motion) create binding for attribute reporting of cluster 0x0001 on endpoint 0x01
06:31:31:779 queue binding task for 0x286D970001136CB2, cluster 0x0001
06:31:31:779 0x286D970001136CB2 (motion) create binding for attribute reporting of cluster 0x0500 on endpoint 0x01
06:31:31:779 queue binding task for 0x286D970001136CB2, cluster 0x0500
06:31:31:779 [INFO_L2] - Set config/pending flags R_PENDING_ENROLL_RESPONSE | R_PENDING_WRITE_CIE_ADDRESS (0x0000000000000030) in 0x0000000000000000 --> 0x0000000000000030
06:31:31:780 read attributes of 0x286D970001136CB2 cluster: 0x0500: [ 
06:31:31:780 0x0000 
06:31:31:780 0x0010 
06:31:31:780 ]
06:31:31:781 add task 4569 type 19 to 0x286D970001136CB2 cluster 0x0500 req.id 30
06:31:31:781 0x286D970001136CB2 (motion) create binding for attribute reporting of cluster 0x0001 on endpoint 0x01
06:31:31:781 discard double entry in binding queue (size: 8) for for 0x286D970001136CB2, cluster 0x0001```

SwoopX, 2021-02-25:
Thanks for providing the logs. Already taken a brief look at it and assume I got a rough idea on what's going on here. Will explain later in more detail.

In the meantime, could you provide another log which contains ALL feasible/recognized button presses in short consecutive order? I understand it can deliver short, double press and hold. Thanks!

kyletpugh, 2021-02-25:
I went through each type of press twice, it looks like they don't show up correctly. It is /sensors/3, the desk button/temp.

MartinTerp, 2021-03-19:
still an issue.

Mimiix, 2021-03-19:
@MartinTerp as long as nobody provides the info swoop requested, this can't be solved...

bitbyt3r, 2021-03-19:
It looks like each of the possible events are already in the log that @kyletpugh provided. If other data is needed I'm also seeing this issue and can try to get logs.

MartinTerp, 2021-03-19:
So i should just close the issue with "buy other button" ? 

I have provided the logs i have regarding this issue, if others have more logs i hope they will add it aswell

kyletpugh, 2021-03-19:
Yeah, I provided what was requested. I can always provide more but I have several intertwined problems, it's super frustrating because half my house is not working...

Mimiix, 2021-03-19:
Swoop asked:

_In the meantime, could you provide another log which contains ALL feasible/recognized button presses in short consecutive order? I understand it can deliver short, double press and hold. Thanks!_

kyletpugh, 2021-03-19:
I provided that in the log, that is what the log is. I pressed the switch buttons on every possible way, all 3, lol. The problem occurred multiple times with the same and other buttons during this time. What else is actually needed?

himem-sys, 2021-04-03:
I'm having the same issue, my smartthings buttons are triggering phantom events approx every hour ever since I upgraded from Deconz 2.09.00 to 2.10.04.    It freaked me out at first because my garage door opened by itself, then all the lights turned off an hour later!  Can I assist with logs or debug info to help get this sorted?  Thanks.

RichJeanes, 2021-04-17:
Can some please post precise instructions for obtaining the logs needed to fix this? I'm happy to provide them. This bug has rendered all of my buttons useless and has significantly hampered my home for months now.

RichJeanes, 2021-04-20:
@SwoopX I don't want to be a pest, but do you have any more info on this issue for us? @kyletpugh says they provided what you asked for in their logs. If that doesn't contain what you need, please let us know so we can help get this issue fixed.

maurcol, 2021-04-23:
Also having the same issue, If instructions are provided on what type of logs are needed or how to get them I also can provide them to have more data if that helps

Mimiix, 2021-04-23:
I asked swoop. He mentioned: What hapens if you set `websocketnotifyall` to `false`  (wiki https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#response-fields)

SwoopX, 2021-04-23:
We probably have already mentioned that to various people, but just not here. In the various logs we've received here or on Discord, there's no duplicate events on zigbee level. This would pretty much leave it to the websocket events.

By default, all key/value items are send whenever there's an update to **ANY** of them. However, you can change that behavior by the parameter Mimiix already mentioned above, please see here https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#parameters_4

After having that changed, only the actually changing values are sent. Presumably, this solves the observed behavior. We've suggested this before at least on Discord, but unfortunately never got any update on this back. So, if anybody could try this out and let us know, it should bring us one step further.

maurcol, 2021-04-23:
Thanks for your replies. I'd gladly try but at least now I see my knowledge is limited. Don't know where the changes have to be made. How I access the place to do the changes?
I use Home Assistant with Deconz, if you provide me with a short instruction I could give it a try and reply on the outcome.

P.s. I tried to google info from Mimiix reply and yours(SwoopX) but couldn't figure it out so far where to look  for the place to input the changes.

Smanar, 2021-04-23:
> By default, all key/value items are send whenever there's an update to **ANY** of them. However, you can change that behavior by the parameter Mimiix already mentioned above, please see here https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#parameters_4

Si it mean the first notification is the buttonevent, and the second one ? what is the changed field that trigger a second notification ?

RichJeanes, 2021-04-23:
@SwoopX Was a change made to the websocket events or configuration that caused this to start happening? I can't say I remember what the event data actually looked like before this issue cropped up, but I didn't change my handling of these events which means the Home Assistant events didn't used to contain the buttonevent unless that is what triggered the HA event.

@maurcol Configuration changes need to be made via the backing web API.
1. From HA Supervisor, go to your deCONZ add-on configuration and set a Host port fro the "API backend" (I just used the default 40850). You can now use the hostname/IP of your HA server and the port set in the deCONZ config as your target URL.
1. Now you need to interact with the web API. If you aren't familiar with this kind of interaction already, you can use Node-RED on your HA server to do it.
    Add the necessary nodes: inject => www-request => debug
[image]
1. [DOCS](https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/configuration/#acquire-api-key) First we need to get an API key to authenticate with the gateway. Configure your inject node like this:
[image]
Set the www-request node Method option to "- set by msg.method -".
[image]
Deploy the changes.
[image]
Go to the deCONZ add-on, Phoscon web UI, Gateway settings, Advanced, and click on "Authenticate App".
[image]
Now go back to Node-RED and activate your inject node by clicking the grey square on its left.
[image]
Now switch your right pane to the debug console in the top right:
[image]
You should see a message there with JSON that looks like "{"username":"xxxxxxxxxx"}" Save this username. It is your API key that allows you to authenticate to the deCONZ gateway. Anywhere you see "\<apikey\>", replace it with your key.
1. Now you can use the API docs to do anything you want. For example: When you see "GET /api/\<apikey\>/config" in the docs, set msg.method to "GET" and msg.url to "http://\<server\>:40850/api/\<apikey\>/config". If you need to supply parameters, set msg.payload to whatever JSON is called for, otherwise just leave it blank.
    Deploy the changes, then inject. You'll see the results in the debug console. If it's a particularly long response, I suggest copy/pasting it into a text editor for easier reading, preferably one with syntax highlighting (two good options are [Sublime Text](https://www.sublimetext.com/) and [Notepad++](https://notepad-plus-plus.org/))
1. To make the configuration modification that has been suggested here, this is what your inject node should look like:
[image]
    And you should get a response like this:
[image]

RichJeanes, 2021-04-23:
Unfortunately that configuration change did not solve the issue. I am still getting events on the Home Assistant bus that report a buttonevent when no button was pressed.

maurcol, 2021-04-23:
Thanks for the great instructions, I'm familar with node-red so this was great. I will try and see what happens

maurcol, 2021-04-23:
For now I follow the instructions and got the response as you provided on the screen shot. I now have to wait if there will be a phantom trigger. 
I set up a notification to my phone to be notify if that happens and what code has been send. So now is matter of waiting. Actually another thing which I don't know if it is related to this issue and perhaps @RichJeanes can be also interested. I only have 2 smartthings buttons so it was not such a big deal when all this started but a week after all this started happening one of them die. Not sure if it's entirely dead, I chnage the battery and tried to put it again in paring mode but it won't react.

RichJeanes, 2021-04-23:
I am still getting ghost events on the HA bus that include button press events.
[image]
[image]

Here is an event from an actual button press:
[image]

I have also noticed that when a button has not been interacted with for a while (not sure exactly how long), it sends two events the next time it is pressed.

maurcol, 2021-04-23:
I can confirm, I also got a phantom trigger. now. However didn't catch the debug info. Will post it as soon as I get it

kyletpugh, 2021-04-24:
So I finally got the button that was having an issue pairing to join the network after updating deCONZ and deleting the node from the GUI. This button appears to have no issues now, after over an hour with no phantom events. No extra events for this button but still getting the extra presses for the other buttons. Appears still two immediate and five minutes after last activity plus randomly about each hour for the others. I wonder if there is something in the config now that is preventing them all from working like this one. I am going to delete them, go around and re-pair them tomorrow to see if that works. I will let you know if this solves the issue...

Using v2.11.1-beta 3/14/2021
Firmware: 26660700

Smanar, 2021-04-24:
One change it have between the working and the not working version, are the reporting values
I realy don't think it s the guilty but the test is fast

ATM you need to have as reporting setting on the POWER_CONFIGURATION_CLUSTER, attribute 0x0021
```
            rq.minInterval = 30;
            rq.maxInterval = 21600;
            rq.reportableChange8bit = 10;
```

Try with olders values
```
            rq.minInterval = 300;
            rq.maxInterval = 2700;
            rq.reportableChange8bit = 1;
```

chadwikx, 2021-04-24:
I went from 6.6.2 to 6.8.0 last night. I can also confirm the phantom triggers... Unfortunately I have about 15 smartthings buttons. 

Hoping my snapshot downgrade was enough to fix it for now.

I might spin up an test environment to attempt to get some logs soon.

rob1303-hub, 2021-04-27:
This is what occurs for me:

 - Phantom trigger type is always the same as last real trigger (ie if last real press was 1002 then phantom triggers from then on are 1002; if I then press 1004 future phantom triggers will be 1004).
 - Phantom trigger - always exactly 60 minutes and 40.7 seconds after last press/ last phantom trigger.
 - Phantom triggers do not align with 'last seen' (which occur more frequent than 60m40.7 seconds) 
 - Phantom triggers do not align with temperature updates (which occur more frequent than 60m40.7 seconds) 

I have made the websocketnotify API change and this did not resolve the issue

My system:

Phoscon Version: 2.10.04/ 3/14/2021
Firmware: 26680700
HA Add-on version: 6.8.0

Not sure if they are useful but attached screenshot of REST API settings for the button and temperature entity, and also screenshot of logbook of automations I set up in HA to capture phantom triggers.

Let me know if you need any further information.

[image]

[image]

kyletpugh, 2021-04-27:
I have different behavior with my buttons. I have three types of lights, three types of buttons/switches, and motion sensors. All the lights and motion sensors are working. The buttons/switches all have various strange issues that can be ignored but this one cannot since it causes events to recur. I agree that I am getting a phantom trigger every hour plus some seconds for every button, I suspect this is the battery level update. I am getting a five minute last seen from the first press phantom event for some of the buttons. I am getting a phantom event that always to be for temperature every twenty minutes, I think, on some of the buttons. These are overlapping groups, some are routed through Samsung bulbs and some are routed through Hue bulbs. The other bulbs are only end devices. All the buttons switches seen to be connected through a router except one, and that one button appears to not have any issues... So I am super confused....

rob1303-hub, 2021-05-09:
> automations I set up in HA to capture phantom triggers

Counter up to phantom triggers 300 now.

Unfortunately this issue has made my Smarthings buttons unusable.

Is any solution possible/likely?

kyletpugh, 2021-05-09:
So, I've discovered that my previously paired devices are working differently than unpaired and re-paired devices after some recent version update. The still paired from before devices behavior is still the same. The re-paired devices behavior is worse, they now disconnect whenever they send an update and go offline until the battery is removed and replaced. The wake event is causing a double press again...  I'm beginning to think this is a problem with devices that are actually multiple devices. It would be so much easier if these devices were actually differentiated with sub-devices or something so that only the correct device is being updated.

RichJeanes, 2021-05-10:
We're now at almost four months of absolutely no progress on this issue and barely any feedback. I think I'm giving up on the deCONZ add-on and I'm going to start over with the ZHA integration. You can check if your devices are compatible with a different Zigbee integration here: https://zigbee.blakadder.com/index.html

Smanar, 2021-05-10:
Just by curiosity, one of you have tried to set the same reporting than before the issue ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4232#issuecomment-826045321

rob1303-hub, 2021-05-11:
Smanar

Happy to try the change - but not sure where/ how I make this change?

In the Phoscon view I can only see the Temperature Sensor for the buttons - the buttons themselves do not show as nodes.  The Temperature sensor blue dot pulses when then button is pressed, and there is an deconz_event in HA.

If I go to POWER_CONFIGURATION_CLUSTER, attribute 0x0021 for the Temperature Sensor node then it states it is read only and I am not able to amend.

Also even if I try to read the current Reporting Config - I get 'Reading config failed'

See screenshot.
[image]

Smanar, 2021-05-12:
Ha yes sorry, you probably need to wake up the device in same time you make the request, bit yes, it s exactly this one.

rob1303-hub, 2021-05-13:
@Smanar 

Managed to change the attribute (by writing config as I pressed button).

Unfortunately - no change to the behaviour of the button.

Still getting phantom every hour.
[image]

iceninejon, 2021-06-01:
Just wanted to echo that I too am still having this issue. It’s really frustrating that this hasn’t been addressed.

dkjerkegaard, 2021-06-08:
Just bought a Smartthings button only to find out that it doesn't work on deconz. Great...

alaghbar, 2021-06-08:
Yes, that was the reason I moved to zigbee2mqtt and I'm glad I did. All devices including Smartthings button work perfectly there

RichJeanes, 2021-06-09:
Yup, I moved all five of mine (and the rest of my Zigbee network) to Home Assistant's ZHA integration a month ago and haven't had any issues with them.

JamieP83, 2021-06-09:
> Yup, I moved all five of mine (and the rest of my Zigbee network) to Home Assistant's ZHA integration a month ago and haven't had any issues with them.

Hey
How long did it take?
Any steps/guide to follow? 
Any downsides?

closed 2021-06-12 by manup

RichJeanes, 2021-06-18:
@JamieP83 It didn't take long, but I only have about a dozen Zigbee devices. I didn't use any guides, just uninstalled the deConz add-on, added the ZHA integration, and started resetting devices. They just pop-up in the interface and you can rename them. It's pretty smooth. Also since most automations are based on entity name, as long as you keep those the same, you shouldn't have to modify anything to keep them working, I have noticed lately that my devices have been taking several seconds to respond (motion and contact activation, button presses, etc.), however I don't know if that's ZHA's fault or has to do with devices I've added since switching. If this PR actually fixes the buttons, I may switch back to test its performance with the additional devices.

rob1303-hub, 2021-08-21:
For those using the deConz add-in within Home Assistant - the update to version 6.10.0 of the add-in on 20 August 2021 includes the fix to this issue.

After updating the add-on it did not work for me initially - I was still received ghost triggers.

However, after deleting the buttons in deConz and then re-adding them I can confirm this has resolved this issue for me.

Thank you.