dresden-elektronik/deconz-rest-plugin #4233: Heiman Warning Device (Siren) lost connection to zigbee, and won't rejoin

fribse, 2021-01-19:
## Describe the bug

Following a breakdown in the zigbee network (upgraded OS, and rebooted the server, and a lot of devices wouldn't come back online), I'm trying to readd the last couple of devices, two of them are Heiman Sirens.
One was just to be readded, the other one seems to put up a tease.
I can see a 'ghost' device, like this in deconz:
[image]
I suspect it's the missing sirene.
I can't see the mac address shown in the rest interface.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Intel NUC i5
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: (26660700)
 - deCONZ version: (2.07.01)
 - Device: ConBee II
 - Do you use an USB extension cable: yes - USB3
 - Is there any other USB or serial devices connected to the host system? Aeotec Z-wave stick

## deCONZ Logs

## Additional context

Mimiix, 2021-01-19:
Can you _at least_ add the device ID / Manufacturer ID?

Smanar, 2021-01-19:
And re-inclusion don't work ? It s not a battery powered device ?
When you make the re-inclusion (I hope the node is not grayed at this step ?) , can you try to re-send the command that ask for " XXXX Descriptor" (the checkbox on your capture)

fribse, 2021-01-19:
@Mimiix Sorry, I will try and do better, I thought I had put everything in it.
I've takes some shots of the one that is included, it has the same data:

[image]
[image]

@Smanar It is a battery backed up device, so it has both battery and 230V.
I've tried different steps, the node that looked odd doesn't change when I try to readd it.
I also tried all the menu items that appeared when I right clicked on it (just to be sure I didn't miss anything).
I've now deleted that node by right clicking on the deconz view, and select delete node.

But it still doesn't show up. It has been a few hours now, so I will retry the adding again.
I'm not sure how to get the logs out from the home assistant add-on, I'll work a bit on that when I try to readd it.

Smanar, 2021-01-19:
If you have the GUI, will be easy, just go in deconz/ help / log and select "info" "info_l2" and perhaps "error"

fribse, 2021-01-20:
@Smanar Ahaa, I'm learning all the time. Not sure if the port is a USB3, and I really don't want to unplug it in the current situation with the zigbee having crashed twice in as many days, which is the reason for all these troubles.

I set up adding again, and I got this logfile:

```
08:15:01:564 enqueue event config/localtime for /config/
08:15:01:565 poll node d0:cf:5e:ff:fe:cb:79:3e-01
08:15:01:566 Poll light node Glass Case
08:15:02:454 poll node d0:cf:5e:ff:fe:15:85:54-01
08:15:02:456 Poll light node Dinnertable 2
08:15:02:563 enqueue event config/localtime for /config/
08:15:03:064 Daylight now: dawn, status: 130, daylight: 0, dark: 0
08:15:03:271 poll node 00:0d:6f:ff:fe:1f:09:54-01
08:15:03:272 Poll light node Dinnertable 1
08:15:03:563 enqueue event config/localtime for /config/
08:15:04:110 poll node 00:0b:3c:ff:fe:f2:7f:e3-01
08:15:04:111 Poll light node Lian bedside
08:15:04:563 enqueue event config/localtime for /config/
08:15:04:564 poll node 84:2e:14:ff:fe:16:1e:1c-01
08:15:04:565 Poll light node IKEA-hub
08:15:04:565 Idle timer triggered
08:15:04:565 Force read attributes for ZHAConsumption SensorNode Consumption 69
08:15:04:566 binding for attribute reporting of ep: 0x01 cluster 0x0702 seems to be active
08:15:04:566 Force binding of attribute reporting for node Consumption 69
08:15:05:564 enqueue event config/localtime for /config/
08:15:05:881 poll node ec:1b:bd:ff:fe:30:bc:ea-01
08:15:05:882 Poll light node Bedroom Range extender
08:15:06:564 enqueue event config/localtime for /config/
08:15:06:721 poll node ec:1b:bd:ff:fe:b5:06:95-01
08:15:07:243 poll node cc:cc:cc:ff:fe:b2:e9:cb-01
08:15:07:244 Poll light node Nathalie Range extender
08:15:07:564 enqueue event config/localtime for /config/
08:15:07:848 Node data 0xd0cf5efffe2e4d3d profileId: 0x0104, clusterId: 0x0300
08:15:07:849 enqueue event attr/lastseen for /lights/21
08:15:07:850 0xD0CF5EFFFE2E4D3D: update ZCL value 0x01/0x0300/0x0008 after 0 s
08:15:07:851 ZCL attribute report 0xD0CF5EFFFE2E4D3D for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:15:07:851 	payload: 08003001
08:15:07:854 Websocket 172.30.32.1:39134 send message: {"attr":{"colorcapabilities":0,"ctmax":65535,"ctmin":0,"id":"21","lastannounced":"2021-01-18T13:15:06Z","lastseen":"2021-01-20T07:15Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 980lm","name":"Wallpaper lamp","swversion":"2.3.007","type":"Color temperature light","uniqueid":"d0:cf:5e:ff:fe:2e:4d:3d-01"},"e":"changed","id":"21","r":"lights","t":"event","uniqueid":"d0:cf:5e:ff:fe:2e:4d:3d-01"} (ret = 423)
08:15:07:855 Websocket 127.0.0.1:41560 send message: {"attr":{"colorcapabilities":0,"ctmax":65535,"ctmin":0,"id":"21","lastannounced":"2021-01-18T13:15:06Z","lastseen":"2021-01-20T07:15Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI bulb E27 WS opal 980lm","name":"Wallpaper lamp","swversion":"2.3.007","type":"Color temperature light","uniqueid":"d0:cf:5e:ff:fe:2e:4d:3d-01"},"e":"changed","id":"21","r":"lights","t":"event","uniqueid":"d0:cf:5e:ff:fe:2e:4d:3d-01"} (ret = 423)
08:15:08:064 poll node 14:b4:57:ff:fe:fd:49:f1-01
08:15:08:563 enqueue event config/localtime for /config/
08:15:08:589 poll node d0:cf:5e:ff:fe:ea:23:05-01
08:15:09:117 poll node d0:cf:5e:ff:fe:ce:33:13-01
08:15:09:118 Poll light node Livingroom Range Extender
08:15:09:564 enqueue event config/localtime for /config/
08:15:09:565 poll node 00:0b:3c:ff:fe:f2:88:e0-01
08:15:09:566 Poll light node Kenneths Bedside
08:15:09:566 Idle timer triggered
08:15:09:567 Force read attributes for ZHAThermostat SensorNode Thermostat - livingroom
08:15:09:567 binding for attribute reporting of ep: 0x01 cluster 0x0201 seems to be active
08:15:09:568 Force binding of attribute reporting for node Thermostat - livingroom
08:15:10:563 enqueue event config/localtime for /config/
08:15:10:882 poll node 00:0b:57:ff:fe:ec:75:0f-01
08:15:10:883 Poll light node Gardentable
08:15:11:564 enqueue event config/localtime for /config/
08:15:11:721 poll node 00:0d:6f:00:15:39:d8:97-01
08:15:11:722 Poll light node Siren Livingroom
08:15:11:846 Node data 0xd0cf5efffe2e4d3d profileId: 0x0104, clusterId: 0x0300
08:15:11:847 0xD0CF5EFFFE2E4D3D: update ZCL value 0x01/0x0300/0x0007 after 0 s
08:15:11:848 ZCL attribute report 0xD0CF5EFFFE2E4D3D for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
08:15:11:850 	payload: 0700217a01
08:15:12:563 enqueue event config/localtime for /config/
08:15:12:564 poll node 00:15:8d:00:04:5c:b3:18-01-0402
08:15:12:565 Poll ZHATemperature sensor node Refridgerator
08:15:13:063 Daylight now: dawn, status: 130, daylight: 0, dark: 0
08:15:13:456 poll node 00:15:8d:00:04:5c:b3:18-01-0405
08:15:13:457 Poll ZHAHumidity sensor node Refridgerator
08:15:13:564 enqueue event config/localtime for /config/
08:15:13:980 Node data 0x00158d0003c3fb67 profileId: 0x0104, clusterId: 0x0201
08:15:13:981 0x00158D0003C3FB67: update ZCL value 0x01/0x0201/0x4003 after 0 s
[... 269 lines cut ...]
08:15:58:267 Poll light node DVD bookcase
08:15:58:564 enqueue event config/localtime for /config/
08:15:59:105 poll node 90:fd:9f:ff:fe:16:7c:44-01
08:15:59:106 Poll light node Drinks cabinet
08:15:59:563 send permit join, duration: 0
08:15:59:564 enqueue event config/localtime for /config/
08:15:59:565 poll node d0:cf:5e:ff:fe:2e:4d:3d-01
08:15:59:565 Poll light node Wallpaper lamp
08:15:59:566 Idle timer triggered
[... cut ...]
08:15:59:569 Force read attributes for ZHAPresence SensorNode Motion Sensor - Kitchen
08:15:59:570 don't create binding for attribute reporting of sensor Motion Sensor - Kitchen
08:15:59:570 Force binding of attribute reporting for node Motion Sensor - Kitchen
08:16:00:041 Poll APS request to 0xD0CF5EFFFE2E4D3D cluster: 0x0006 dropped, values are fresh enough
08:16:00:053 Search sensors done
08:16:00:563 enqueue event config/localtime for /config/
08:16:01:139 poll node 00:0b:57:ff:fe:8a:7e:a6-01
08:16:01:140 Poll light node Trio 2
```

SwoopX, 2021-01-20:
There isn't any communication from your device in there whatsoever. Have you woken it up after having started the sensor search? You do that by pushing the physical (reset) button once..

fribse, 2021-01-20:
Yes, it has a pinhole that I press for 5s, and then it flashes green rapidly. I don't see any other buttons on the siren.

Smanar, 2021-01-20:
IDK for zigbee model but for Zwave

> Reset to factory default
> -Long press Net_Button at least 10 seconds in the product.
> 
>  Inclusion
> Click [Add] icon in Z-Wave Controller.-Press the Net_Button 3 times within 1.5s, Green LED is Blinking 3 times within 1 .5second.-If add Process is successful, Green LED will turn off.

fribse, 2021-01-23:
Yes, that's the only guide I can find as well (for z-wave), but that's not the way to do it for the zigbee device.
I don't have the included guide any more, but I've ordered a new one, I hope there is one included there, it's very very odd that it doesn't react at all. Could there be some 'leftover' in the deconz that then makes it so it won't attach again?

Smanar, 2021-01-23:
Yep it s possible, but you will see the device announce on log, even if something is not good and make the device leave just after.

SwoopX, 2021-01-23:
@fribse Can you also please share a screenshot of the node info panel? Just to be sure, we should check that everythings fine there as well. A screenshot where you can see all the available device clusters would also be nice. Thanks!

fribse, 2021-01-25:
@SwoopX node info panel of what? I can't get the thermostat in, so I'm a bit confused, sorry for being a dumbass??
Is it just the deconz overview you want, like this?
[image]

Smanar, 2021-01-25:
I think it s the siren info panel.
Something like that https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/master/img/dev_integration_1.png

fribse, 2021-01-27:
This is the working siren:
[image]

fribse, 2021-01-28:
Ok, I've had it unplugged for a while now, and the battery finally wore out.
I tried plugging it to power, and join it to the network, and lo and behold, it did so. I now have a light, and a sensor (battery) for it, weeheee.
Darned that was weird. I guess I can close it now, as I got it working, but it still doesn't clear up what happened, but it could of course be internally in the siren, or some of the routers that already 'knew it' somehow, and that's cleared up now??

SwoopX, 2021-01-30:
Hm, indeed a bit weird but good that it works now. I'd put that into `device goes banana section`. Also have one candidate, which is typically rock solid,but hangs every few months while reporting not being connected to the network. Interestingly, it still was in any case.

However, resetting it is always a pain since it's an in–wall socket and therefore requires to have the fuse popped. On top of that, it's the socket which powers my router and smart home, so no joy...

fribse, 2021-01-30:
> I'd put that into `device goes banana section`

HA HA HA HA.

closed 2021-01-30 by fribse

thogens, 2022-09-08:
Hi all!

I just saw this bug and I occasionally face the exact same issue with the same device!
I found out that by pushing the reset pin hole repetitive and like crazy (preferably without power connected), the device will blink red at one point and only then it is able to reconnect again.
It takes some time, but it is much faster than waiting a week until the battery drains...
Hope that helps someone!
Thomas

thogens, 2023-02-08:
Great! I think those devices are very good repeaters, so it is worth using them :-).

evanwon, 2024-08-07:
Thank you @thogens, this totally worked for me too. I wish I knew *how* exactly it worked, but after poking that little button for like five minutes straight, it finally blinked red and I was about to immediately discover the device again.