dresden-elektronik/deconz-rest-plugin #4237: Not able to pair any device

JustNZ, 2021-01-19:
I've switch from a raspbee2 to a conbee2 on my raspberry pi 4 with hassio installed.
I'm using the official deconz addon to controll the conbee2.

The conbee2 is plugged in on the 2.0 USB Port.

## Describe the bug

When i search for any device on the phoscon webinterface it simply doesn't find anything.
The only thing i can see is that the log reports some errors.

## Steps to reproduce the behavior

  1. Plugged conbee2 in 
  2. configured deconz hassio addon
  3. start the addon
  4. search for devices

## Expected behavior

Find devices

## Screenshots
On the screenshot i loaded an backup and tried if this is working.
But some devices dont even get a connection or get a timeout.

[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - The Raspberry is using an SSD as Booting Method. The SSD is connected on USB 3.0 Port

## deCONZ Logs

[17:38:59] INFO: Successfully send discovery information to Home Assistant.
17:39:04:835 dev /dev/ttyAMA0
17:39:04:837 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
17:39:04:837 GW firmware version: 0x26660700
17:39:04:838 GW firmware version is up to date: 0x26660700
17:39:16:287 binding for cluster 0x0006 of 0x7CB03EAA0A04F08A exists (verified by reporting)
17:39:16:288 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x7CB03EAA0A04F08A (seems to be active)
17:39:16:289 binding for cluster 0x0008 of 0x7CB03EAA0A04F08A exists (verified by reporting)
17:39:16:290 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x7CB03EAA0A04F08A (seems to be active)
17:39:17:671 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:18:623 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:39:21:451 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:39:25:232 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:26:039 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:39:26:287 binding for cluster 0x0006 of 0x84182600000F679E exists (verified by reporting)
17:39:26:288 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x84182600000F679E (seems to be active)
17:39:28:815 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:39:33:647 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:35:465 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:39:38:317 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:39:42:153 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:43:973 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:39:46:858 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:39:48:802 Current channel 15
17:39:48:810 Device TTL 3090 s flags: 0x7
17:39:49:671 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:51:470 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:39:54:355 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:39:57:178 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:39:58:955 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:40:01:804 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:40:04:568 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:40:06:367 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:40:09:155 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:40:13:061 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:40:14:857 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:40:17:666 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:40:20:569 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:40:22:339 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task
17:40:25:158 0x7CB03EAA0A0A7273 error APSDE-DATA.confirm: 0xE1 on task
17:40:28:083 0x7CB03EAA0A09640B error APSDE-DATA.confirm: 0xE1 on task
17:40:29:857 0x7CB03EAA00B06572 error APSDE-DATA.confirm: 0xE1 on task

## Additional context

In the past i used a raspbee2 and it was running for about 3 Months. But i had temperature issues and system crashes.
So i decided to switch to an conbee2.

Mimiix, 2021-01-19:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

Pretty easy answer: Youahve interference issues.

JustNZ, 2021-01-19:
Oh ok, could that be solved with a USB extension?

Mimiix, 2021-01-19:
Yes.

Just read about the interference:) The answer is there.

JustNZ, 2021-01-19:
Alright thanks a lot

closed 2021-01-19 by JustNZ