dresden-elektronik/deconz-rest-plugin #4238: Xiaomi Mi Smart Plug not fully usable (only switch feature offered)

bcutter, 2021-01-19:
## Describe the question or issue you are having

I paired a *Xiaomi Mi Smart Plug* using Phoscon with category "lights". In my smart home automation system only one entity (switch) is offered, I miss all the others for the power monitoring this device fully supports.

According to https://phoscon.de/de/conbee2/compatible the device is supported.

Do I need to pair it by another way? Did not pop up by trying to add as switch.
**How to get access to the other entities?**

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.07.01
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? - NO

## deCONZ Logs

## Additional context

bcutter, 2021-01-19:
Update: Grabbing the model code **ZNCZ04LM** from https://www.zigbee2mqtt.io/information/supported_devices.html#xiaomi and looking for this on https://phoscon.de/de/conbee2/compatible I can not find this device.

So... it´s not supported yet, really?

That would be a no-go - 3rd zigbee device to be integrated using Phoscon/deCONZ - and would likely force me to switch to ZigBee2MQTT :-( :-( :-(

bcutter, 2021-01-19:
Update 2: Could find it on https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-smart-plugs--in-wall-switches--power-meters

[image]

Will digg through https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2583, fingers crossed...

tonyawad88, 2021-02-05:
@bcutter I am using Home Assistant with Phoscon/Conbee II and when I pair it as a "sensor" in Phoscon app, I get 3 entities in Home assistant, 1 switch, 1 current power consumption and 1 historical power consumption. Remove it from Phoscon and re-add it using the Sensor => Other.
[image]

bcutter, 2021-02-07:
@tonyawad88 That´s the same result for me. BUT:
- You can start pairing on the sensors page, but the plug will always be listed on the lights page.
- I tried pairing a second Mi Smart Plug. I only get one switch entity in Home Assistant:
[image]

So my issues still are:
1) Only switch entity offered by deCONZ to Home Assistant.
2) With a lot of luck (pairing, waiting, checking, deleting, pairing again, hope and pray, ...) I get 3 entities: switch, current and total power consumption.
3) Total power consumption does not calculate (always 0.001 kWh) and it´s not constantly available:
[image]
4) I still miss all the other entities and sensors which are offered when using another ZigBee Gateway and ZigBee2MQTT (device temperature, overpower, link quality).
5) After deCONZ being not available (restart e. g.) the plug needs to be "awaken" by pressing the button, otherwise it will be unavailable in the ZigBee network.

So that´s HELL OF A PAIN and I´m not sure why it is that horrible. As a friend of mine runs this device just fine with other ZigBee equipment I tend to fingerpoint to the deCONZ part.

framesetter99, 2021-02-07:
Worked for me. All in all I probably needed 3 pairing attempts to get it right.
I paired it as a Light, waited, checked the sensors with the rest API.
If I got a consumption and power sensor I was good.
If not I removed the "light" via the Phoscon GUI + the sensors via rest API and started again.
Did it step by step with two devices.
Once I was able to see values with the rest API I added it in Home Assistant with the Deconz Integration.

[image]

bcutter, 2021-02-07:
After another hour of pairing / deleting / repeat I ended up in a somewhat stable total consumption entity which is acutally usable, so at least issues #1 and #3 are gone:
[image]

I think reading the basic cluster in deCONZ after startint sensor search in Phoscon made some difference (or reading the simple descriptos, don´t know exactly).

As you can see the attributes still show no values. @framesetter99 do you see actual values there (check current and total consumption entities)?

[image]

bcutter, 2021-02-13:
Update: Need to revert my word from related to issue 3: it fell back to normal behaviour, so no stable data delivery and values aren´t updated at all.

[image]

Additionally hope in latest v2.9.3 which should fix the issue 6 were not satisfied (see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4139#issuecomment-778615728) so currently I need to sum all my experiences by simply stating **"this device is not usable with Dresden Electronic hardware and software"**.

Will try to swap these plugs, just horrible experience. Same plugs with another ZigBee gateway using ZigBee2MQTT and everything´s working fine. Too bad.

bcutter, 2021-03-07:
I gave up. Great hardware but not usable with ConBee 2/deCONZ in my case.

closed 2021-03-07 by bcutter