dresden-elektronik/deconz-rest-plugin #4239: ConBee Support for Thread Protocol

Veldkornet, 2021-01-20:
## Feature request type

Add support for the Thread protocol to the ConBee / RaspBee devices

## Description

Since the launch of the Apple HomePod Mini supporting Thread, consumer products have now become available from both [Nanoleaf](https://nanoleaf.me/en-EU/integration-hub/thread/) as well as [Eve](https://www.evehome.com/en/blog/homekit-over-thread), with more devices on the way.
Currently, only the HomePod Mini is the only consumer border router available. 
If the ConBee supported Thread, it could be used in combination with the [OpenThread border router](https://hub.docker.com/r/openthread/otbr) docker container, the setup of which is briefly described [here](https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/connected-home-over-ip-sneak-peek-of-early-development-on-nrf52840#mcetoc_1eo2h5j243) in an article about CHIP, which makes use of Thread. 
This in turn would hopefully make it possible to be used in HomeAssistant and other similar applications for example.

## Considered alternatives

- Wait for another manufacturer to create a device which does support this
- - Purchase a [Nordic nRF52840](https://www.nordicsemi.com/Software-and-tools/Development-Kits/nRF52840-Dongle) development kit instead

## Additional context

It's my understanding from [this ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2057#issuecomment-716873042) post that the ConBee could be updated to support Thread.

Kristian8606, 2021-01-21:
Бих искал да  инсталирам фърмуер който да превърне моят Conbee 2 в граничен рутер. Моля нека екипът на Deconz сподели своите мисли.

Mimiix, 2021-01-21:
Passed it onto Manup. Awaiting his response.

manup, 2021-01-21:
I think it should be possible with the OpenThread R21 hardware support, the Microchip R21 is the radio used ConBee II and RaspBee II. Currently we don't have plans to create and maintain a firmware for Thread support on our own in *near future*, since the focus is on Zigbee and there aren't many documented Thread products out there yet.

This might change in future if Thread becomes more popular. I really hope Thread becomes more adoption, as one common firmware stack like OpenThread for multiple vendors is a huge step forward.

https://github.com/openthread/openthread/tree/master/third_party/microchip

scobby, 2021-05-11:
> Nice, good to know there is activity in that direction. I did a little research here too and found that Thread alone wouldn't cut it in future, on top of it sits the CHIP protocol, which luckily also uses Zigbee APS (or parts of it) and ZCL layers. In theory I can think of lower layer Thread support in deCONZ while ZCL handling could be shared with current code.

That would be great! And for sure more orders for conbee sticks :)

gesendet

Kristian8606, 2021-08-29:
I built the binaries and got to this step,  https://github.com/openthread/ot-samr21#flash-binaries
 Can anyone help me flash these files on conbee 2

ln-12, 2021-11-12:
Hey, are there any updates on this? I read an [article at the german news site heise](https://www.heise.de/news/Smart-Home-Standard-Matter-soll-bislang-inkompatible-Komponenten-verbinden-6237042.html) which states that there are 60 devices which are currently being certified and that Matter (successor of CHIP) will launch in summer 2022.

reinisb, 2021-11-14:
Hey! Same question on the progress. I'm in the process of ordering a bunch of new generation [eve](https://www.evehome.com/en/thread) devices that all support thread. I'd love to use my conbee ii + raspberry pi as a thread border router!

ebaauw, 2021-11-14:
Eve have announced that their Thread-enabled devices will support Matter, see https://www.evehome.com/en/matter?mtm_campaign=Matter%20Video&mtm_source=newsletter, but currently they run HomeKit over Thread.  I doubt they’ll currently work with any non-Apple border router.

reinisb, 2021-11-14:
> but currently they run HomeKit over Thread. I doubt they’ll currently work with any non-Apple border router.
I did not know that it is possible for a device to run Thread and at the same time have a limitation baked in defining that it can only work with Homekit border router (e.g. HomeKit Mini or Apple TV 4k 2gen). I had understood that a thread-enabled device works with any border router.

ebaauw, 2021-11-14:
They run HomeKit (HAP) using Bluetooth and Thread as underlying transports.  It's HomeKit, i.e. Apple, so anything non-standard is possible, I'm afraid.  I don't know the details, but I think the Eve devices only turn on their Thread capability when they see an Apple border router (over Bluetooth).  I only have Eve (Thermo, Energy, Weather, and Door Sensor) and Apple (HomePod mini, latest Apple TV) Thread devices, so I cannot test whether they play nice with other Thread devices.  It's probably not so much Thread itself as the HomeKit application-level protocol on top of that.  Matter should hopefully get rid of this nonsense, as it also standardised the application-level protocols.

Kristian8606, 2021-11-14:
https://9to5mac.com/2021/08/19/nanoleaf-homekit-lights-get-thread-router-support-software-update/

ebaauw, 2021-11-14:
Cool, another item on my list for when I'll have won the lottery.

ajdezigns, 2022-01-06:
### Feature request type
Add support for Matter (formerly Project CHIP)

URL: https://github.com/project-chip/connectedhomeip

### Description
Matter (formerly Project Connected Home over IP, or Project CHIP) is a new Working Group within the Connectivity Standards Alliance (CSA, formerly Zigbee Alliance). This Working Group plans to develop and promote the adoption of a new, royalty-free connectivity standard to increase compatibility among smart home products, with security as a fundamental design tenet.

The goal of the Matter project is to simplify development for manufacturers and increase compatibility for consumers. The project is built around a shared belief that smart home devices should be secure, reliable, and seamless to use. By building upon Internet Protocol (IP), the project aims to enable communication across smart home devices, mobile apps, and cloud services and to define a specific set of IP-based networking technologies for device certification.

gloomytree, 2022-01-21:
I'd like to add that the Home Assistant Yellow project goes into this direction. They include a Zigbee & Thread Gateway on their Yellow. I think this will give Thread another push forward. I'd love to see compatibility on the Conbee II, as I have just ordered my upgrade from the Raspbee 1 =)

stefanuytterhoeven, 2022-01-26:
Really waiting die this! More and more than enabled devices appear...

stefanuytterhoeven, 2022-02-16:
still no news about thread support? Or do i need to buy the home assistant yellow?

fcastilloec, 2022-02-16:
I'm looking into buying a Zigbee dongle/hub right now but given that Matter is right around the corner, I want to make sure that whatever I buy will work with Matter too. It doesn't have to work right now, any promise that will work is good enough.
The Conbee II seems like the best option for me IF there's a path to update to Matter. Home Assistant yellow is not an option for me because I already own a Raspberry Pi 4 and don't want to replace it this soon. This is the same sentiment behind the Conbee II, I don't want to have to replace it soon once Matter is ubiquitous.

manup, 2022-02-17:
Just made a test with https://github.com/openthread/ot-samr21 compilation is smooth. I very much like the design of the OpenThread stack as it allows to let it even run a very minimal version on the MCU of the ConBee II (even ConBee I could be ported) letting the bulk work to the host application.

However after flashing the binary nothing worked, which makes sense.
Next steps are to adapt the linker script since ConBee II bootloader expects the application code on a certain address as well as to enable the watchdog handling in the OpenThread stack otherwise it would kill itself after a few seconds.

To be frank I'm yet not too deep in the Matter/Thread topic and currently only have brief knowledge about it.
But I think with OpenThread Border Router application a Matter controller might be able to connect to it. As soon as the firmware is available it would be cool if someone could make a few tests with it, like connecting with an existing Thread network.

manup, 2022-02-17:
And please keep in mind a ConBee II can only run Thread or Zigbee but not both at the same time.

Kristian8606, 2022-02-17:
Is it ready to work as a border router?  I built a thread net of 10 nanoleaf bulbs and one homepod.  I own 3 raspberries and would like to buy two more conbee2 to build border routers for more network stability, as my wife sometimes disconnects the HomePod from the power supply, and the network becomes unmanageable.

Kristian8606, 2022-02-17:
Is there an easy way to flash the openThread firmware of conbee 2?  The openThread samr21 instructions require additional flash hardware that I don't have.

ajdezigns, 2022-02-17:
> Is it ready to work as a border router? I built a thread net of 10 nanoleaf bulbs and one homepod. I own 3 raspberries and would like to buy two more conbee2 to build border routers for more network [...]

How do you control the bulbs from home assistant? Or do you?

Mimiix, 2022-02-17:
Can we please stay on topic here? @Kristian8606 @ajdezigns

Kristian8606, 2022-02-17:
The topic is for openThread, and my question is, is there an easy way to flash the ConBee 2 firmware to work as a border router.  I would like to test ConBee2 on my existing network. 

@manup  Is there an easy way to flash the ConBee2 firmware without special tools, for example using raspberry pi?

austwhite, 2023-04-06:
I came here looking for the same thing as @Kristian8606  but it looks like no one is responding to this for over a year.
Is there an official Conbee firmware to upgrade it to be an OpenThread Border Router, or is it still experimental?    

As far as I can see, this is quite relevant to this particular Github thread as the title is ConBee Support for Thread Protocol

Smanar, 2023-04-06:
https://github.com/dresden-elektronik/gcfflasher

If I m right with it you can flash the conbee "without special tools"