dresden-elektronik/deconz-rest-plugin #4246: Migration to new server fails partly

mhoogenbosch, 2021-01-21:
## Describe the question or issue you are having
I have tried migrating to a new server. I have been using a Pi2 (with the plain image for a PI) for a while but bought a Pi4. Installed docker and installed the docker container. I created a backup, restored it on the new host, pulled the USB stick and added it to the docker host. Checked if it was still ttyACM0 which is linked to the container and it was.

## Screenshots
It also finds it:
[image]

## Environment

 - Host system: Raspberry Pi 4 running with docker
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: 2.09.01
 - Device: ConBee II 
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? Yes; USB3 SSD hard disk from which everything runs. There is no SD card present in the system.

## deCONZ Logs
The log shows:
```
13:47:45:811 [INFO] - No button map for: SML001 endpoint: 0x02 cluster: 0x0400 command: 0x0A payload[0]: 000
13:47:45:812 ZCL attribute report 0x00178801020271C5 for cluster: 0x0400, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:47:46:122 0x588E81FFFEDE9005 error APSDE-DATA.confirm: 0xE1 on task
13:47:49:578 [INFO] - No button map for: SML001 endpoint: 0x02 cluster: 0x0402 command: 0x0A payload[0]: 000
13:47:49:579 ZCL attribute report 0x0017880104B6B2B9 for cluster: 0x0402, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
13:47:50:623 [INFO] - No button map for: SML001 endpoint: 0x02 cluster: 0x0406 command: 0x0A payload[0]: 000
13:47:50:624 ZCL attribute report 0x00178801020271C5 for cluster: 0x0406, ep: 0x02, frame control: 0x18, mfcode: 0x0000 
13:47:51:212 0x00178801020271C5 error APSDE-DATA.confirm: 0xE1 on task
13:47:51:439 0x588E81FFFED4AE9A error APSDE-DATA.confirm: 0xE1 on task
13:47:51:793 Current channel 25
13:47:51:798 CTRL got nwk update id 0
13:47:51:802 Device TTL 2378 s flags: 0x7
13:47:53:158 0x0017880104B6B2B9 error APSDE-DATA.confirm: 0xE1 on task
13:47:55:116 0x588E81FFFEFE94C1 error APSDE-DATA.confirm: 0xE1 on task
13:47:56:888 0x588E81FFFEF0C82D error APSDE-DATA.confirm: 0xE1 on task
13:47:58:331 0x680AE2FFFE6ADB4F error APSDE-DATA.confirm: 0xE1 on task
13:47:59:113 ZCL attribute report 0x588E81FFFEDE9005 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
13:47:59:310 0x680AE2FFFEFDC53E error APSDE-DATA.confirm: 0xE1 on task
13:48:00:406 0x588E81FFFEC81325 error APSDE-DATA.confirm: 0xE1 on task
13:48:01:630 0x588E81FFFEC9D51A error APSDE-DATA.confirm: 0xE1 on task

```
[image]
[image]

## Additional context
Using IKEA routers for most routing, added a few powersockes from blitzwolf a few days ago and am planning to remove the IKEA ones. I have a lot of routers across the whole house. Everything worked just fine when connected to the Pi2. 
Unfortunately I cannot connect the Stick back to the Pi2 anymore, because for some reason deconz doesn't start on it anymore.

Mimiix, 2021-01-21:
14:58:29:765 0x588E81FFFEDE817F error APSDE-DATA.confirm: 0xE1 on task,

You have interference issues :)

mhoogenbosch, 2021-01-21:
Wauw, did not expect that. Already connected it to USB2 but now added the extension cord and everything came back to life! Thanks guys.

closed 2021-01-21 by mhoogenbosch