dresden-elektronik/deconz-rest-plugin #4248: Lidl Livarno Lux Leuchten

Schnuecks, 2021-01-21:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light

## Screenshots

ebaauw, 2021-01-21:
Looks like they're already exposed by the REST API Plugin?  If not, please provide all screenshots, including, for each of the lights, the node with endpoints and clusters, and the _Node Info_ panel.

We map the weird names for the other Livarno lights to "LIDL Livarno Lux" as manufacturer name and the product number as model id (e.g. HG06106C for the RGB E27 light bulb), so please provide those as well (typically on the box or in the user manual).

Schnuecks, 2021-01-22:
First thank you for your response, here are the modelnr of the LED-Stimmungslicht 14148906L and the model nr LED-Lichtleiste is 14149505L / black version 14149506L / white version.

yes it seems that it is exposed and working with the api. i have a question that leads me here.

on my openhab installation the lights raises an error message, so i put it here.
```
2021-01-21 16:49:44.881 [INFO ] [z.internal.handler.LightThingHandler] - Could not determine effect light type for thing deconz:extendedcolorlight:b2252869:842e14fffe24b96801, please request adding support on GitHub.
2021-01-21 16:49:44.884 [INFO ] [z.internal.handler.LightThingHandler] - Could not determine effect light type for thing deconz:extendedcolorlight:b2252869:588e81fffeecb2be01, please request adding support on GitHub.
2021-01-21 16:49:45.221 [INFO ] [z.internal.handler.LightThingHandler] - Could not determine effect light type for thing deconz:extendedcolorlight:b2252869:842e14fffe22fe3701, please request adding support on GitHub.
```
The LED light bar does not switch off completely, but remains on at around 10% and continues to light. All other commands such as colors and brightness work normally, they just don't go off completely. I also tried various apps.

ebaauw, 2021-01-22:
> here are the modelnr of the LED-Stimmungslicht 14148906L and the model nr LED-Lichtleiste is 14149505L / black version 14149506L / white version.

Does the black version of the Lichtleiste have a different _Manufacturer Name_ than the white one?  Otherwise we cannot distinguish them.

> on my openhab installation the lights raises an error message

Best file that with openhab (or the deCONZ/openhab integration).

> And another problem i have is the LED light bar does not switch off completely, but remains on at around 10% and continues to light.

Hoe are trying to switch off the light bar?  What API command are you sending?

Could you check in the GUI whether it reacts to an _Off_ command?  And to a _Move to Level (with On/Off)_, setting level to `0` or `1`?  Could you please include  the _Node Descriptor_ screenshot of the light bar?  Particularly interested in the _Manufacturer Code_.

Schnuecks, 2021-01-22:
> Does the black version of the Lichtleiste have a different _Manufacturer Name_ than the white one? Otherwise we cannot distinguish them.

No it's the same

> Best file that with openhab (or the deCONZ/openhab integration).

they mentioned it should report it here ;)
https://community.openhab.org/t/deconz-which-thing-type-is-needed-for-lidl-melinera-led-wire/109462/44?u=J-N-K
> 
> > And another problem i have is the LED light bar does not switch off completely, but remains on at around 10% and continues to light.
> 
> Hoe are trying to switch off the light bar? What API command are you sending?

I don't know which command openhab or HueEssentials is sending but both have the same result so that it doesn't go off completly. But i think they do the right thing, because of below.

> Could you check in the GUI whether it reacts to an _Off_ command? And to a _Move to Level (with On/Off)_, setting level to `0` or `1`? Could you please include the _Node Descriptor_ screenshot of the light bar? Particularly interested in the _Manufacturer Code_.

i have toogle power on / off not completly off, same with move to level.

here the node info of the lightbar
[image]

here a small video first On / Off second level 0 to 255 and return
i put it on my space because it's too big for github, 14 days valid

<a href="https://wolke.frickelboxx.de/f/25a91928a114487086ab/" target="_blank">Video Lighbar</a>

best regards

ebaauw, 2021-01-22:
The link doesn't work.

If the light doesn't react to an _Off_ command the issue is with the light firmware.  Nothing deCONZ nor the REST API Plugin can do.  Never seen this before, though.  You might check _Toggle_ or _Off with Effect_, but these are, I think optional, and I've seen plenty of lights not supporting these.  As there are not supporting _Move to Level (with On/Off)_.

Did you try the LIDL app with their Silvercrest gateway?  Is that able to turn off the light?.

Schnuecks, 2021-01-22:
Hello,

sorry that it not worked, I put it on Google Drive

https://drive.google.com/file/d/1HGxOfZN0sgvLOzal9CTnOJNfC_LlENvW/view?usp=drivesdk

I don't have the gateway right now because  I would try it without it.

I could give it a try and order one.

Schnuecks, 2021-01-22:
So, how can we solve the message from openhab with the effect feature?

Is this feature available on the lamp or is openhab mentioning that that kind of lamp has this feature and can't use it?

I just like to understand.

Heiner1967, 2021-01-25:
Hi, 
I got such Livarno Lux Stimmungsleuchte to work with conbee II and Phoscon, but the related remote i do not get at all to work.
For me the trick to get it  was to apply the learning directly beside the conbee, and not beside another  zigbee device which shall work as range extender - this only after initial registration seems to work.
But it would be great to get the remote to work as well.

Mimiix, 2021-01-25:
@Heiner1967 Please open a seperate issue for that remote containing all screenshots :)

Heiner1967, 2021-02-24:
let me add this ping to avoid this get closed unsolved.

closed 2021-03-27 by github-actions[bot]

Hl2run, 2021-03-27:
Please add support for the light bar. Thanks :)

Einstein2150, 2022-03-25:
there is a new model of the light on the market --> [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5900](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5900)